Ysurac/openmptcprouter #978: Aggregation speed is subpar

marcschaller, 2020-04-20:
## Expected Behavior
I have two WAN, with speed `S1` for the first and speed `S2` for the second. OMR should give the local network a connection speed at least better than `S1` or `S2`, and capped at `MAX(S1 + S2)`.

## Current Behavior
The internet speed does not exceed the lowest of both connections, and is sometimes even lower.

## Possible Solution
I tried the workarounds of the related issues (#689, #381, #648):
- Disabling ShadowSocks: no effet
- Disabling ShadowSocks encryption: no effect
- Disabling `fast-opn`: no effect

## Steps to Reproduce the Problem
  1. Have everything in working order (all is green in the network state)
  2. Try to download something
  3. Wait longer than it has to be 

## Context (Environment)
- `wan1` is ADSL
- `wan2` is 4G

Here are some speed tests with `omr-test-speed`:
```bash
root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 77.6M    0     0  1313k      0  2:13:03  0:01:00  2:12:03 1540k^C
```
```bash
root@OpenMPTCProuter:~# omr-test-speed wan2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  217M    0     0  3662k      0  0:47:43  0:01:00  0:46:43 2131k^C
```
```bash
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 71.1M    0     0  1197k      0  2:25:52  0:01:00  2:24:52 1222k^C
```

I also tried to download a file directly on the VPS:
```bash
root@vps:~# wget http://www.ovh.net/files/100Mio.dat
--2020-04-20 09:16:21--  http://www.ovh.net/files/100Mio.dat
Résolution de www.ovh.net (www.ovh.net)… 213.186.33.6
Connexion à www.ovh.net (www.ovh.net)|213.186.33.6|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 104857600 (100M) [application/octet-stream]
Sauvegarde en : « 100Mio.dat »

100Mio.dat          100%[===================>] 100,00M  35,5MB/s    ds 2,8s
```
So no bottleneck here.

## Specifications

  - OpenMPTCProuter version: `0.54.7`
  - OpenMPTCProuter VPS version: `0.1015 4.19.104-mptcp`
  - OpenMPTCProuter platform: RPI4
[image]

Ysurac, 2020-04-20:
You should define master as the Freebox. You can try ECF and BLEST MPTCP sheduler and some other TCP congestion control in Network->MPTCP.

I also see the VPS is at firstheberg, there is know problems host load problems on them for now.
Can you check the result of "dmesg" on the VPS ? Check also System Log in State menu of the router.

marcschaller, 2020-04-20:
On the VPS, `dmesg` gives a lot of the following line:
```bash
[499586.555890] Shorewall:net-fw:DROP:IN=eth0 OUT= MAC=00:2d:75:d1:56:85:00:1e:f7:f6:1b:00:08:00 SRC=YYY.YYY.YYY.YYY DST=XXX.XXX.XXX.XXX LEN=40 TOS=0x00 PREC=0x00 TTL=243 ID=19430 PROTO=TCP SPT=45283 DPT=7352 WINDOW=1024 RES=0x00 SYN URGP=0
```
(`DST` contains the VPS IP, and `SRC` some random IP)

---

I'm sorry, how should I define a master? Can't seem to find the option :(

---

And here are the system logs, perhaps there is something wrong here. It's overflowing with these lines:
```log
Mon Apr 20 12:10:58 2020 kern.err kernel: [337593.955836] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1500000 (-12)
Mon Apr 20 12:10:58 2020 kern.err kernel: [337593.964406] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1500000)
Mon Apr 20 12:10:58 2020 kern.err kernel: [337593.975138] cpufreq: __target_index: Failed to change cpu frequency: -22
```
And sometimes these:
```log
Mon Apr 20 12:11:02 2020 kern.err kernel: [337597.074354] xhci_hcd 0000:01:00.0: Ring expansion failed
```

Ysurac, 2020-04-20:
Log on the VPS is about shorewall, the firewall used on the VPS. So it's when someone try to access to a port of the server (they are all closed by default, only port 65000 to 65535 are open for OpenMPTCprouter).

You can define a master in Network->MPTCP. You should set in System->OpenMPTCProuter and advanced parameters tab the select master to none.

No real problems in systems logs, at least nothing that should impact speed.

marcschaller, 2020-04-20:
I played a little bit with TCP congestion controls, and it did some results!
Here are the results of my experiments (I let `omr-test-speed` running for 1 minute on each test):

```
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 71.1M    0     0  1197k      0  2:25:52  0:01:00  2:24:52 1222k^C
```

| Congestion Control| Average speed| 
| --- |---:|
|`bbr` (default)|`1197k`|
|`reno`|`3197k`|
|`cubic`| `3386k`|
| `lia`| `2158k`|
|`olia`|`2247k`|
|`wvegas`|`713k`|
|`balia`|`914k`|
|`mctcpdesync`|`3733k`|

So `mctcpdesync`, `cubic` and `reno` are the leaders here (with a preference for `mctcpdesync`).
So, let's check if `Multipath TCP scheduler` has any impact with the selected congestion control:

| Multipath TCP scheduler| Average speed| 
| --- |---:|
|Default|`3733k`|
|`BLEST`|`3780k`|
|`cubic`| `3534k`|

Not a much of an impact. I'll stick with default settings.

Finally, a graph showing that both the WANs are responding the best they can during the tests:
[image]

Anyway, I think that's it. Changing the congestion control did the trick for me. Thanks @Ysurac ! I'll close this issue if it's alright with you.

closed 2020-04-20 by marcschaller