Ysurac/openmptcprouter #979: Shorewall -> deprecated action.drop

gnosoz, 2020-04-20:
# Issue / error
I noticed a small error on the shorewall configuration that affects both vps and router:
Checking /usr/share/shorewall/deprecated/action.Drop for chain Drop...
   WARNING: "You are using the deprecated Drop default action. Please see http://www.shorewall.net/Actions.html

Shorewall, also with this error, is working like a charm so is not any concern or major problem. 

## Steps to Reproduce the Problem
 sudo shorewall check

## Possible Solution
I am going to check and play a bit with the various actions to see what the new version requires to remove this hint as I am unsure how long the deprecated will remain valid.

Ysurac, 2020-04-20:
It's fixed in develop server script now.

gnosoz, 2020-04-21:
grteat news I mark it as closed then

closed 2020-04-21 by gnosoz