Ysurac/openmptcprouter #980: Shadowsocks exit-code, invalid header, ERROR: crypto: stream: repeat IV detected

cron410, 2020-04-21:
## Expected Behavior
Shadowsocks services running on VPS normally

## Current Behavior

```
root@vps:~# systemctl status shadowsocks-libev-server@config
● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:ss-server(1)
root@vps:~# systemctl restart shadowsocks-libev-server@config
root@vps:~# systemctl restart shadowsocks-libev-server@config2
root@vps:~# systemctl status shadowsocks-libev-server@config
● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-04-21 11:10:59 EDT; 1min 7s ago
     Docs: man:ss-server(1)
  Process: 19571 ExecStart=/usr/bin/ss-server -c /etc/shadowsocks-libev/config.json (code=exited, status=255)
 Main PID: 19571 (code=exited, status=255)

Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 INFO: using tcp fast open
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 INFO: UDP relay enabled
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 INFO: enable TCP no-delay
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 INFO: initializing ciphers... chacha20
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 INFO: tcp server listening at [::0]:65101
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 INFO: tcp port reuse enabled
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 ERROR: bind: Address already in use
Apr 21 11:10:59 router ss-server[19571]:  2020-04-21 11:10:59 ERROR: failed to bind address
Apr 21 11:10:59 router systemd[1]: shadowsocks-libev-server@config.service: Main process exited, code=exited, status=255/n/a
Apr 21 11:10:59 router systemd[1]: shadowsocks-libev-server@config.service: Failed with result 'exit-code'.
root@vps:~# systemctl status shadowsocks-libev-server@config2
● shadowsocks-libev-server@config2.service - Shadowsocks-Libev Custom Server Service for config2
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-04-21 11:11:07 EDT; 1min 3s ago
     Docs: man:ss-server(1)
  Process: 19574 ExecStart=/usr/bin/ss-server -c /etc/shadowsocks-libev/config2.json (code=exited, status=255)
 Main PID: 19574 (code=exited, status=255)

Apr 21 11:11:07 router systemd[1]: Started Shadowsocks-Libev Custom Server Service for config2.
Apr 21 11:11:07 router ss-server[19574]:  2020-04-21 11:11:07 ERROR: Invalid config path.
Apr 21 11:11:07 router systemd[1]: shadowsocks-libev-server@config2.service: Main process exited, code=exited, status=255/n/a
Apr 21 11:11:07 router systemd[1]: shadowsocks-libev-server@config2.service: Failed with result 'exit-code'.

root@router:~# ps aux | grep shadow
root     18582  0.0  0.1  21132  2640 ?        Ss   09:15   0:00 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json
root     18587  0.4  2.2  92564 45044 ?        Ss   09:15   0:35 ss-server --manager-address 127.0.0.1:8839 -f /root/.shadowsocks/.shadowsocks_65101.pid -c /root/.shadowsocks/.shadowsocks_65101.conf --acl /etc/shadowsocks-libev/local.acl -t 600 -u --fast-open --reuse-port --no-delay --mptcp -6 --plugin /usr/local/bin/v2ray-plugin --plugin-opts server -s 0.0.0.0
root     24265  0.0  0.0  14428  1012 pts/0    S+   11:25   0:00 grep --color=auto shadow
root@router:~# killall ss-server

root@router:~# systemctl restart shadowsocks-libev-server@config
root@router:~# systemctl status shadowsocks-libev-server@config
● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-21 11:25:59 EDT; 4s ago
     Docs: man:ss-server(1)
 Main PID: 24574 (ss-server)
   CGroup: /system.slice/system-shadowsocks\x2dlibev\x2dserver.slice/shadowsocks-libev-server@config.service
           └─24574 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json

Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: tcp port reuse enabled
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: tcp server listening at 0.0.0.0:65101
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: tcp port reuse enabled
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: udp server listening at [::0]:65101
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: udp port reuse enabled
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: udp server listening at 0.0.0.0:65101
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: udp port reuse enabled
Apr 21 11:25:59 router ss-server[24574]:  2020-04-21 11:25:59 INFO: running from root user
Apr 21 11:26:03 router ss-server[24574]:  2020-04-21 11:26:03 ERROR: [udp] invalid header with addr type 213
Apr 21 11:26:03 router ss-server[24574]:  2020-04-21 11:26:03 ERROR: [udp] invalid header with addr type 234

root@router:~# systemctl status shadowsocks-libev-server@config
● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-21 11:25:59 EDT; 22min ago
     Docs: man:ss-server(1)
 Main PID: 24574 (ss-server)
   CGroup: /system.slice/system-shadowsocks\x2dlibev\x2dserver.slice/shadowsocks-libev-server@config.service
           └─24574 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json

Apr 21 11:48:20 router ss-server[24574]:  2020-04-21 11:48:20 ERROR: crypto: stream: repeat IV detected
Apr 21 11:48:20 router ss-server[24574]:  2020-04-21 11:48:20 ERROR: failed to handshake with 173.153.91.6: authentication error
Apr 21 11:48:28 router ss-server[24574]:  2020-04-21 11:48:28 ERROR: crypto: stream: repeat IV detected
Apr 21 11:48:28 router ss-server[24574]:  2020-04-21 11:48:28 ERROR: failed to handshake with 173.153.91.6: authentication error
Apr 21 11:48:32 router ss-server[24574]:  2020-04-21 11:48:32 ERROR: crypto: stream: repeat IV detected
Apr 21 11:48:32 router ss-server[24574]:  2020-04-21 11:48:32 ERROR: failed to handshake with 173.153.91.6: authentication error
Apr 21 11:48:35 router ss-server[24574]:  2020-04-21 11:48:35 ERROR: crypto: stream: repeat IV detected
Apr 21 11:48:35 router ss-server[24574]:  2020-04-21 11:48:35 ERROR: failed to handshake with 173.153.91.6: authentication error
Apr 21 11:48:40 router ss-server[24574]:  2020-04-21 11:48:40 ERROR: crypto: stream: repeat IV detected
Apr 21 11:48:40 router ss-server[24574]:  2020-04-21 11:48:40 ERROR: failed to handshake with 173.153.91.6: authentication error

```

## Possible Solution

## Steps to Reproduce the Problem

  1. Wait until Shadowsocks dies on server
  2. Restart services, look at status.

## Context (Environment)
 I have two VPS, one in Softlayer, the other on a private server owned by friends.

## Specifications

  - OpenMPTCProuter version: Version 0.54.7
  - OpenMPTCProuter VPS version: Version 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-04-21:
shadowsocks-libev-server is not used anymore. shadowsocks-libev-manager is now used, it's what you see in your ```ps aux```.

cron410, 2020-04-21:
These services?

```
root@vps:~# systemctl status shadowsocks-libev-manager@config
● shadowsocks-libev-manager@config.service - Shadowsocks-Libev Custom Manager Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-manager@.service; indirect; vendor preset: enabled)
   Active: inactive (dead)
root@vps:~# systemctl status shadowsocks-libev-manager@config2
● shadowsocks-libev-manager@config2.service - Shadowsocks-Libev Custom Manager Service for config2
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-manager@.service; indirect; vendor preset: enabled)
   Active: inactive (dead)
```

Ysurac, 2020-04-21:
This use ```shadowsocks-libev-manager@manager``` service.

cron410, 2020-04-21:
```
~# systemctl status shadowsocks-libev-manager@manager
● shadowsocks-libev-manager@manager.service - Shadowsocks-Libev Custom Manager Service for manager
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-manager@.service; indirect; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-21 11:50:32 EDT; 2h 36min ago
 Main PID: 1379 (ss-manager)
   CGroup: /system.slice/system-shadowsocks\x2dlibev\x2dmanager.slice/shadowsocks-libev-manager@manager.service
           └─1379 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json

Apr 21 13:27:16 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:19 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:43 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:43 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:44 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:44 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:44 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:27:44 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:28:01 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
Apr 21 13:28:09 mptcp3 ss-server[1383]: getpeername: Transport endpoint is not connected
```
```
:~# systemctl status shadowsocks-libev-manager@manager
● shadowsocks-libev-manager@manager.service - Shadowsocks-Libev Custom Manager Service for manager
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-manager@.service; indirect; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-21 14:27:58 EDT; 2min 12s ago
 Main PID: 11091 (ss-manager)
   CGroup: /system.slice/system-shadowsocks\x2dlibev\x2dmanager.slice/shadowsocks-libev-manager@manager.service
           ├─11091 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json
           └─11094 ss-server --manager-address 127.0.0.1:8839 -f /root/.shadowsocks/.shadowsocks_65101.pid -c /root/.shadowsocks/.shadowsocks_

Apr 21 14:27:58 mptcp3 ss-server[11094]: tcp port reuse enabled
Apr 21 14:27:58 mptcp3 ss-server[11094]: udp server listening at 0.0.0.0:65101
Apr 21 14:27:58 mptcp3 ss-server[11094]: udp port reuse enabled
Apr 21 14:27:58 mptcp3 ss-server[11094]: running from root user
Apr 21 14:29:11 mptcp3 ss-server[11094]: getpeername: Transport endpoint is not connected
Apr 21 14:29:12 mptcp3 ss-server[11094]: getpeername: Transport endpoint is not connected
Apr 21 14:29:13 mptcp3 ss-server[11094]: getpeername: Transport endpoint is not connected
Apr 21 14:29:14 mptcp3 ss-server[11094]: getpeername: Transport endpoint is not connected
Apr 21 14:29:49 mptcp3 ss-server[11094]: getpeername: Transport endpoint is not connected
Apr 21 14:29:59 mptcp3 ss-server[11094]: remote recv: Connection reset by peer
```

After restarting GloryTUN-UDP and shadowsocks-manager I still cannot get a stable connection.  Getting `user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)` in the OpenWRT System Log every few minutes.

Ysurac, 2020-04-21:
Check that there is not a firewall blocking VPS API port (65500).

cron410, 2020-04-21:
There is none. My VPS is connected directly to the internet with no NAT or any other firewall. Only Shorewall as it comes with OpenMPTCP VPS script. I only change the logging level in `/etc/shorewall/policy` by deleting `info` entries.

Before:
[image]

After:
[image]

Ysurac, 2020-04-22:
Check that omr-admin service is running

cron410, 2020-04-22:
I saw the server has a zombie process so I tracked it down to OMR-Admin

```
:~# ps -ef | grep [d]efunct
root     30835   809  0 07:37 ?        00:00:00 [sh] <defunct>
```
```
:~# ps aux | grep 809
root       809  2.5  3.7 1113768 77740 ?       Ssl  01:06  10:09 python3 /usr/local/bin/omr-admin.py
```
but OMR-admin service seems to be running fine. I'll restart it and see if that resolves the issue.
```
:~# systemctl status omr-admin.service
● omr-admin.service - OMR-Admin
   Loaded: loaded (/lib/systemd/system/omr-admin.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-04-22 01:06:42 EDT; 6h ago
 Main PID: 809 (python3)
   CGroup: /system.slice/omr-admin.service
           └─809 python3 /usr/local/bin/omr-admin.py

Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... glorytun
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... dsvpn
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... iperf3
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... openvpn
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... mlvpn
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... mptcp
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config... ipv6
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    get server IPv4
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    get hostname
Apr 22 07:33:08 mptcp3 omr-admin.py[809]: DEBUG:    Get config: done
```

```
:~# systemctl status glorytun-udp@tun0.service
● glorytun-udp@tun0.service - Glorytun UDP on tun0
   Loaded: loaded (/lib/systemd/system/glorytun-udp@.service; indirect; vendor preset: enabled)
   Active: active (running) since Wed 2020-04-22 07:31:28 EDT; 4min 34s ago
  Process: 19987 ExecStartPost=/etc/glorytun-udp/post.sh /etc/glorytun-udp/tun0 (code=exited, status=2)
 Main PID: 19986 (glorytun)
   CGroup: /system.slice/system-glorytun\x2dudp.slice/glorytun-udp@tun0.service
           └─19986 glorytun bind 0.0.0.0 65001 keyfile /etc/glorytun-udp/tun0.key dev gt-udp-tun0 chacha persist

Apr 22 07:31:28 mptcp3 systemd[1]: Starting Glorytun UDP on tun0...
Apr 22 07:31:28 mptcp3 glorytun-udp-run[19986]: running on device gt-udp-tun0 as pid 19986
Apr 22 07:31:28 mptcp3 post.sh[19987]: RTNETLINK answers: File exists
Apr 22 07:31:28 mptcp3 systemd[1]: Started Glorytun UDP on tun0.
```

```
:~# systemctl status glorytun-tcp@tun0.service
● glorytun-tcp@tun0.service - Glorytun TCP on tun0
   Loaded: loaded (/lib/systemd/system/glorytun-tcp@.service; indirect; vendor preset: enabled)
   Active: active (running) since Wed 2020-04-22 01:06:42 EDT; 6h ago
  Process: 628 ExecStartPost=/etc/glorytun-tcp/post.sh /etc/glorytun-tcp/tun0 (code=exited, status=0/SUCCESS)
 Main PID: 627 (glorytun-tcp)
   CGroup: /system.slice/system-glorytun\x2dtcp.slice/glorytun-tcp@tun0.service
           └─627 glorytun-tcp listener keyfile /etc/glorytun-tcp/tun0.key dev gt-tun0 port 65001 mptcp chacha20 retry count -1 const 5000000 t

Apr 22 01:06:42 mptcp3 systemd[1]: Starting Glorytun TCP on tun0...
Apr 22 01:06:42 mptcp3 post.sh[628]: Cannot find device "gt-tun0"
Apr 22 01:06:42 mptcp3 systemd[1]: Started Glorytun TCP on tun0.

root@mptcp3:~# systemctl restart glorytun-tcp@tun0.service
root@mptcp3:~# systemctl restart glorytun-udp@tun0.service
root@mptcp3:~# systemctl status omr
omr-admin.service      omr.service            omr6in4@user0.service
root@mptcp3:~# systemctl status omr
omr-admin.service      omr.service            omr6in4@user0.service
root@mptcp3:~# systemctl status omr
omr-admin.service      omr.service            omr6in4@user0.service
[... 42 lines cut ...]
Wed Apr 22 07:29:33 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:29:41 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:29:46 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:29:51 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:03 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:08 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:09 2020 daemon.err uhttpd[5371]: luci: accepted login on /admin/system/openmptcprouter/status for root from 10.1.1.91
Wed Apr 22 07:30:13 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:13 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:21 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:29 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:34 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:39 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:44 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:52 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:30:57 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
[... cut ...]
Wed Apr 22 07:31:19 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:31:27 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:31:31 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:31:32 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
Wed Apr 22 07:31:37 2020 daemon.err /usr/bin/ss-local[8706]: remote_recv_cb_recv: Connection reset by peer
```

Ysurac, 2020-04-22:
It's an update on the VPS or a fresh install ? No package was installed manually (like shadowsocks) ? No change in encryption manually ?

cron410, 2020-04-22:
No I haven't changed anything. Just installed the VPS script. Is there a way to reset the encryption keys on both of my VPS? Maybe because they are different keys it could be causing issues at the router?

Ysurac, 2020-04-22:
If you validate the wizard again this should sync the keys.

cron410, 2020-04-22:
How do I validate the wizard? Save and Apply button?

cron410, 2020-04-22:
I deleted the GloryTUN key from the VPS tun0.key files, deleted the VPS profiles from the router, save/apply, added them again, verified the VPN keys were automatically placed back in the tun0.key files for TCP and UDP VPNs. GloryTUN TCP VPN is up, and still showing Direct Output status on the VPS.

[image]

Ysurac, 2020-04-22:
For the public ip address, it get it via a check to ```http://ip.openmptcprouter.com```.
Can you get your IP via this address ? What do you have in State->system log ? No DNS issue ?

cron410, 2020-04-22:
Visiting ip.mptcprouter.com returns the IP of the VPS correctly. 
Here's my router system logs with WAN IP sanitized. https://pastebin.com/YUNnKZrd

Ysurac, 2020-04-22:
There is really a problem with shadowsocks, this seems to be not connected.
Can you check /var/log/daemon.log on the VPS if there is anything about shadowsocks ?
Can you also check on the VPS the configuration in /etc/shadowsocks-libev/manager.json and if encryption is the same as the one on the router ?

cron410, 2020-04-22:
On the router I have chacha20-ietf-poly1305 in http://10.1.1.1/cgi-bin/luci/admin/services/shadowsocks-libev/servers which is the same as listed in /etc/shadowsocks-libev/manager.json

/var/log/daemon.log does not exist on VPS or Router.

Ysurac, 2020-04-22:
What is the VPS distribution ? Ubuntu ?

cron410, 2020-04-22:
```
:~# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.4 LTS"
```

Ysurac, 2020-04-22:
ok, it's why there is no /var/log/daemon.log. The log should be somewhere in /var/log/
I will make some test on ubuntu, long time since no test...

cron410, 2020-04-23:
Are you looking for the output of OMR-admin.py? I can see it with 
`journalctl -f -n 0`  and every few minutes I see the following:
```
Apr 23 14:23:05 mptcp3 omr-admin.py[820]: DEBUG:    Get status...
Apr 23 14:23:05 mptcp3 omr-admin.py[820]: DEBUG:    Get status: done
```

Ysurac, 2020-04-23:
ok, so omr-admin.py is running at least.
Are you using obfuscation on shadowsocks ?

cron410, 2020-04-23:
I installed the script with ` OPENVPN=no DSVPN=yes OBFS=yes VPS_DOMAIN=`  with VPS_DOMAIN set to a DNS A record with my VPS IP. The router Obfuscation has been disabled since I rebuilt the router a few days ago. I'll try enabling it in the router and see if that helps.

cron410, 2020-04-23:
I changed settings to this. 
[image]

Ysurac, 2020-04-23:
This should help only in China.
All is working if you set only one VPS ?
You should also try to run script without any options.

cron410, 2020-04-23:
This entire ticket has been with 1 VPS, the other removed from router config. I ran the script with `OBFS=no` and disabled the setting in the router. I'll report back after the tunnels have some time to run.

Ysurac, 2020-04-23:
Try without any options.

cron410, 2020-04-30:
Very stable this morning! I reinstalled without any options last week as you said. I have GloryTUN UDP selected in Advanced Options on the Router and the only VPS in Router config is the 2CPU/2GB RAM VPS in Softlayer. 

[image]

It's been a very busy week for me and I finally have some time to test today so I plan to reinstall the VPS on my friend's server and reconfigure the VM for 2CPU/4GB RAM. It's also being moved to a dedicated SSD from my old laptop I installed a few days ago in the server.

closed 2020-09-14 by github-actions[bot]