Ysurac/openmptcprouter #982: Use VPS dns A record instead IP

magicsurfer, 2020-04-22:
## Expected Behavior
I would like to use an A record pointed to the public IP of the VPS instead IPv4 address.

## Current Behavior
I setupped my VPS and i created a A record pointed to the public IP of the VPS. I tried to set the DNS name of the VPS (for expample vps0001.mydomain.com) instead the IP on the "Setting Wizard" page; i don't receive any warning but the tunnel does not come up. On the VPS side the daemon log i see the first part of the connection but stuck on "Get config: done". If i modify the DNS to the public IP of the VPS all work's perfect.

There is anyway to use this configuration? maybe are very useful if i change in the future my VPS and i not need to reconfigure my router. 

Thank you so much for your work, best regard's Luca.

## Specifications
  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64 virtual machine

Ysurac, 2020-04-22:
The problem is:  should I use IPv4 or IPv6 when I resolve the domain ?

servernoise, 2020-04-22:
Check if AAAA record is available, else use legacy IP from A record.

Ysurac, 2020-04-22:
But if AAAA is available then it will not work if connections are IPv4 only.
Also if VPS is not the initial one, the VPS key will also change, so wizard will still need to be modified.

magicsurfer, 2020-04-22:
Dear Ysurac,
thank you so much for your reply.
1. I use only IPv4
2. the record AAAA are not defined (i try this)
However i agree with you, when i define new VPS the script generate new key and i need to reconfigure my Router.
Thank you so much, best regard's luca

Ysurac, 2020-04-22:
I know for you, but I try to make something that can work for everybody :)
In fact I have several choices:
* I can add a "resolve domain to ip" via the wizard and add IP to every VPN (so new IP is take into account via a wizard new validation only but it's easy)
* I can add a "resolve domain to ip" via VPN script (changes are take into account via VPN script restart, this may be automatic but is not as easy to add)

I don't know what is the best. I think I need to add a function that check if IP defined for connections are IPv6 or IPv4 only then I can resolve to A or AAAA (all domains I use for VPS resolve to both).

closed 2020-09-14 by github-actions[bot]