Ysurac/openmptcprouter #984: Not a bug - Personal record (using rpi4)

kevin39, 2020-04-23:
Hi,
Not a bug, just a proof it working very great :)

My environment : 
- 2x 4G 
- 1x DSL
- OpenMTCPRouter on Rpi4
- TCP Scheduler : BLEST
- Congestion Control : Cubic

I just got a personal record, using speedtest.net : 
[image]

And peak saved by OpenMTCPRouter :
[image]

Thank you @Ysurac for your works. Before OpenMTCP I was depend of my unstable DSL connection and a slow Satellite connection. Thanks to your project I can work more effeencitly and during Covid19 it changed my life :)

gwarf, 2020-04-23:
You are in France, right? What ISP are you using for 4G? I'm currently using 3 ADSL (around 10-12Mbps each), didn't yet considered moving to some 4G.

kevin39, 2020-04-23:
Bouygues (my best Wan) and Free
(Free is also my DSL provider with its AliceADSL offer)

closed 2020-09-14 by github-actions[bot]