Ysurac/openmptcprouter #985: block-mount missing - Bonding going very well - 310/80 mbps down/up on rpi4 - Super

salchris, 2020-04-23:
## Expected Behavior

I though that being derived from openwrt i could find block-mount package.

What do you suggest to solve?

openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d

BTW.. Raspi4 and openmptcprouter rocks! Thank you.
I get 310/80 mbps from two 4g+ lines... , 1 eth interface
Thank you

Ysurac, 2020-04-23:
Should be available in package list.

salchris, 2020-04-23:
root@OpenMPTCProuter:~# opkg list block*
blockdev - 2.35.1-1 - The blockdev program is the command-line interface to call block device ioctls.
root@OpenMPTCProuter:~# 

I tried on rpi4 and wrt3200 and there's no package, while in standard openwrt it's there.

Thank you

salchris, 2020-04-23:
also.. blockd is missing

Ysurac, 2020-04-23:
True, no idea why it's not in package list. I will try to add them in next release.

salchris, 2020-04-23:
Thank you very much!

closed 2020-09-14 by github-actions[bot]