Ysurac/openmptcprouter #986: No Server ping response after 1 second

hello--world, 2020-04-24:
## Expected Behavior

[image]

## Current Behavior

### openmptcprouter gui detail
[image]
### ping gateway
[image]
### Huawei E8372 (192.168.12.1)
[image]

### system log
```
...
Fri Apr 24 01:20:50 2020 daemon.notice netifd: wan4 (32435): udhcpc: sending select for 192.168.11.100
Fri Apr 24 01:20:51 2020 daemon.notice netifd: wan4 (32435): udhcpc: lease of 192.168.11.100 obtained, lease time 86400
Fri Apr 24 01:20:51 2020 daemon.notice netifd: Interface 'wan4' is now up
Fri Apr 24 01:20:53 2020 user.notice post-tracking-post-tracking: eth2 switched off because gateway down and ping from 192.168.11.100 error (223.6.6.6,223.5.5.5)
Fri Apr 24 01:20:53 2020 user.notice post-tracking-post-tracking: eth1 switched off because gateway down and ping from 192.168.12.100 error (223.5.5.5,114.114.114.114)
Fri Apr 24 01:20:53 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.11.1 dev eth2
Fri Apr 24 01:20:53 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.12.1 dev eth1
Fri Apr 24 01:20:53 2020 user.notice post-tracking-post-tracking: wan2 down. Replace default route by 192.168.13.1 dev eth3
Fri Apr 24 01:20:53 2020 user.notice post-tracking-post-tracking: wan1 down. Replace default route by 192.168.13.1 dev eth3
Fri Apr 24 01:20:56 2020 user.notice post-tracking-post-tracking: Set server vps (118.89.219.10) route via 192.168.13.1 metric 5
...

```

### route
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.13.1    0.0.0.0         UG    0      0        0 eth3
default         192.168.13.1    0.0.0.0         UG    5      0        0 eth3
10.255.251.1    *               255.255.255.255 UH    0      0        0 tun0
118.89.219.10   192.168.13.1    255.255.255.255 UGH   5      0        0 eth3
127.0.0.0       *               255.0.0.0       U     1      0        0 lo
192.168.11.0    *               255.255.255.0   U     6      0        0 eth4
192.168.12.0    *               255.255.255.0   U     4      0        0 eth2
192.168.13.0    *               255.255.255.0   U     5      0        0 eth3
192.168.14.0    *               255.255.255.0   U     3      0        0 eth1
```

## Possible Solution

## Steps to Reproduce the Problem

  1. Start normal use

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  `openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d` 
  - OpenMPTCProuter VPS version: `OpenMPTCProuter VPS 0.1008` 
  - OpenMPTCProuter platform: `RPI4 + 4 x Huawei E8372`

Ysurac, 2020-04-24:
VPS need to be updated.
On your interface one need to be set as master (in Network->MPTCP)
The no server ping response after 1 second is only a warning, this mean the command ```ping -B -I wan1 vpsip``` doesn't give an answer in less then 1 second. If it's only sometimes, this mean some packets are lost else ICMP ping can be blocked/ignored somewhere on the path.

hello--world, 2020-04-30:
Thank you.
VPS has been updated.
It's working normally now.

closed 2020-04-30 by hello--world