Ysurac/openmptcprouter #987: Not all interface are aggregated

matfil92, 2020-04-25:
Hi,
I successfully using the openmptcp router to aggregate three connection, 2 connection is with ethernet and the third is with USB Tethering. My openmptcp hardware is raspberry PI 4B with 4GB of ram. Today I've added the fourth connection via WiFi using the wifi adapter integrated into raspberry. The wifi connects successfully but that connection is not aggregated with the other. 
I can confirm that the WiFi connection is configured and connected correctly because if I force with OMB-Bypass my computer lan IP to use only the WiFi connection I can browse internet only with that connection also because whoismyisp and speed test give me that I'm using the wifi connection of Postemobile provider. 
If I made speedtest when I've forced with OMR-Bypass to use only the wifi postemobile I can reach 20mb in download.

I attach here a speed test that I've launched with all 4 aggregation, as you can see the PosteMobile is not used and I don't understand why. 

Do you have any ideas?
 
[https://ibb.co/xSbrz6d](url)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d
  - Kernel 4.19.108
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1015 >
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1

gnosoz, 2020-04-25:
the configuration link is not working, i can't see your setup.

Ysurac, 2020-04-25:
Using wifi for a LTE connection is often not a good idea, many packets are lost.
All is green in OpenMPTCProuter status page ?

matfil92, 2020-04-25:
> Using wifi for a LTE connection is often not a good idea, many packets are lost.
> All is green in OpenMPTCProuter status page ?

@Ysurac Yes all is green in OpenMPTCPRouter status page. The ping of postemobile is nearly 50ms

@gnosoz what specific setup page you want see?

gnosoz, 2020-04-25:
i would use iptraf to monitor a request to monitor the serious connections
usage and see whether is using them all correctly.

https://openwrt.org/packages/pkgdata/iptraf-ng

this will tell you packets sent/ received/ errors in real time for all
network cards and therefore you can see how the scheduler+congestion
protocols are behaving.

From past checks, if all is green and have no logs signalling issues with
the postemobile vlan,  the problem might reside  in the
congestion+scheduler.

matfil92, 2020-04-25:
@gnosoz  Thanks for your fast reply. Could you kindly tell me how can I use iptraf-ng and how to see the congestion+scheduler problem? If you can kindly said me step by step each command it would very appreciated. 

I attach the status page here --> https://ibb.co/xzwsMfR

gnosoz, 2020-04-25:
Are you familiare with ssh?

if so you need to following steps:
  -  login ti the router ssh
  - opkg update && opkg install iptraf-ng
  - iptraf-ng
  - open browser and start an heavy tcp based speed test
  - monitor the lines in the iptraf page

matfil92, 2020-04-25:
@gnosoz 
if I select only the wwan (wireless interface) from iptraf and start a speedtest this is result. How can I read it?  https://ibb.co/7k6FRt3

I don't know if this could help but if I run test of MPTCP Support Check I have different result from the interface that are working and from interface than are not aggregate but I'm not able to read the result 
from interface that aggregation NOT work https://ibb.co/5rs87Nj 

from interface that aggregation work https://ibb.co/NxqZvhB

gnosoz, 2020-04-25:
delete this picture, it has you ip in clear

matfil92, 2020-04-25:
@gnosoz  I've mask all my ip. 
Could you help me on how to read it if there is congestion?

gnosoz, 2020-04-25:
sry for the confusion I don't remember if iptraf gives you all interfaces
together in one screen I can check later.

what you want to see is something like this which in the picture is done
via the program bmon, if memory serves me right also iptraf does it...

https://www.flickr.com/photos/xmodulo/16234265932/

matfil92, 2020-04-25:
@gnosoz  I'll attach two pict more

https://ibb.co/D7kVZx9

https://ibb.co/jfX6Hwn

Thanks again for your help @gnosoz ! I want solve it!

gnosoz, 2020-04-25:
as i thought it is basically just sending icmp/udp via wla0 so either it failed to get response (you see all green so I would exclude it) or you have this interface not set properly under mptcp.

check if maybe the interface wlan0 is not set as handover or other operational status.

matfil92, 2020-04-25:
@gnosoz all WAN interface is set as enabled and only one as master
I need to put every interface as master? 
https://ibb.co/m8hJxhd

Ysurac, 2020-04-25:
Only one master is needed (else this will be a war of master).
Can you send the config from System->OpenMPTCProuter and all parameters tab without keys ?

matfil92, 2020-04-25:
@Ysurac  this are System OpenMPTCPRouter
https://ibb.co/pJZPjP4
https://ibb.co/NVXSCqx
https://ibb.co/4d0XvdH
https://ibb.co/SJhSgXJ

Ysurac, 2020-04-25:
All parameters tab, that you can copy/paste in pastebin an remove keys.

gnosoz, 2020-04-25:
postemobile is LTE and you have sqm activated, try disabling it

matfil92, 2020-04-25:
@gnosoz also with SQM disabled same result.

@Ysurac here all parameters tab

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].server='/use-application-dns.net/'
dhcp.@dnsmasq[0].ipset='/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.limit='180'
dhcp.lan.dns='1.1.1.1'
dhcp.lan.dhcp_option='6,1.1.1.1,8.8.8.8'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='A0:C5:89:33:CE:6C'
dhcp.@host[0].name='DESKTOP'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.100.200'
dhcp.@host[1]=host
dhcp.@host[1].mac='0C:08:B4:B3:9A:22'
dhcp.@host[1].name='Sky'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.100.201'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
[... 1417 lines cut ...]
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'eth1' 'eth2' 'tun0' 'usb0' 'wlan0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.hwmode='11g'
wireless.radio0.channel='auto'
wireless.radio0.htmode='HT20'
wireless.wifinet0=wifi-iface
wireless.wifinet0.ssid='mobile'
wireless.wifinet0.device='radio0'
wireless.wifinet0.mode='sta'
wireless.wifinet0.key='PasswordWifi'
wireless.wifinet0.network='wwan'
wireless.wifinet0.encryption='psk2'

Ysurac, 2020-04-25:
I see no error here. You can try in Network->MPTCP to enable "Multipath Debug" and then give me log in State->System log, this should give the info why the path is not used.

Can you also check in Network->MPTCP that IP of the interface is available in MPTCP fullmesh tab.

It's possible that PosteMobile block MPTCP traffic, in this case you can try MPTCP over VPN but this give a slower speed (I need to make more tests of this feature...).

matfil92, 2020-04-25:
@Ysurac  
Yes the IP is in the MPTCP fullmesh tab is present of postemobile 

Here the log with MPTCP Multipath debug enabled 

Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.178209] __mptcp_init4_subsockets: token 0x242b9c11 pi 4 src_addr:192.168.42.141:0 dst_addr:195.X.Y.Z.:65101 ifidx: 180
Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.187404] mptcp_close: Close of meta_sk with tok 0x242b9c11
Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.207836] mptcp_del_sock: Removing subsock tok 0x242b9c11 pi:4 state 7 is_meta? 0
Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.215532] mptcp_del_sock: Removing subsock tok 0x242b9c11 pi:3 state 7 is_meta? 0
Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.223213] mptcp_del_sock: Removing subsock tok 0x242b9c11 pi:1 state 7 is_meta? 0
Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.230901] mptcp_del_sock: Removing subsock tok 0x242b9c11 pi:2 state 7 is_meta? 0
Sat Apr 25 14:25:51 2020 kern.err kernel: [95400.238597] mptcp_sock_destruct destroying meta-sk token 0x242b9c11
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.282399] mptcp_alloc_mpcb: created mpcb with token 0xb3b5aeb1
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.288869] mptcp_add_sock: token 0xb3b5aeb1 pi 1, src_addr:192.168.177.130:29077 dst_addr:195.X.Y.Z.:65101
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.308969] mptcp_add_sock: token 0xb3b5aeb1 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.317132] __mptcp_init4_subsockets: token 0xb3b5aeb1 pi 2 src_addr:192.168.0.99:0 dst_addr:195.X.Y.Z.:65101 ifidx: 9
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.328961] mptcp_add_sock: token 0xb3b5aeb1 pi 3, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.337196] __mptcp_init4_subsockets: token 0xb3b5aeb1 pi 3 src_addr:192.168.8.219:0 dst_addr:195.X.Y.Z.:65101 ifidx: 174
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.349511] mptcp_add_sock: token 0xb3b5aeb1 pi 4, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.357747] __mptcp_init4_subsockets: token 0xb3b5aeb1 pi 4 src_addr:192.168.42.141:0 dst_addr:195.X.Y.Z.:65101 ifidx: 180
Sat Apr 25 14:25:53 2020 kern.err kernel: [95402.373216] mptcp_close: Close of meta_sk with tok 0xb3b5aeb1
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.392892] mptcp_del_sock: Removing subsock tok 0xb3b5aeb1 pi:4 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.393561] mptcp_alloc_mpcb: created mpcb with token 0x1bd4c1aa
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.400623] mptcp_del_sock: Removing subsock tok 0xb3b5aeb1 pi:3 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.400637] mptcp_del_sock: Removing subsock tok 0xb3b5aeb1 pi:2 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.422845] mptcp_add_sock: token 0x1bd4c1aa pi 1, src_addr:192.168.177.130:24189 dst_addr:195.X.Y.Z.:65500
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.442923] mptcp_del_sock: Removing subsock tok 0xb3b5aeb1 pi:1 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.450824] mptcp_sock_destruct destroying meta-sk token 0xb3b5aeb1
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.459433] mptcp_add_sock: token 0x1bd4c1aa pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.467572] __mptcp_init4_subsockets: token 0x1bd4c1aa pi 2 src_addr:192.168.0.99:0 dst_addr:195.X.Y.Z.:65500 ifidx: 9
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.478785] mptcp_add_sock: token 0x1bd4c1aa pi 3, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.486922] __mptcp_init4_subsockets: token 0x1bd4c1aa pi 3 src_addr:192.168.8.219:0 dst_addr:195.X.Y.Z.:65500 ifidx: 174
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.498338] mptcp_add_sock: token 0x1bd4c1aa pi 4, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.506472] __mptcp_init4_subsockets: token 0x1bd4c1aa pi 4 src_addr:192.168.42.141:0 dst_addr:195.X.Y.Z.:65500 ifidx: 180
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.563141] mptcp_del_sock: Removing subsock tok 0x1bd4c1aa pi:2 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.844240] mptcp_close: Close of meta_sk with tok 0x1bd4c1aa
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.867396] mptcp_del_sock: Removing subsock tok 0x1bd4c1aa pi:1 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95402.982726] mptcp_del_sock: Removing subsock tok 0x1bd4c1aa pi:3 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95403.061530] mptcp_del_sock: Removing subsock tok 0x1bd4c1aa pi:4 state 7 is_meta? 0
Sat Apr 25 14:25:54 2020 kern.err kernel: [95403.071080] mptcp_sock_destruct destroying meta-sk token 0x1bd4c1aa
Sat Apr 25 14:25:55 2020 kern.err kernel: [95403.410608] mptcp_alloc_mpcb: created mpcb with token 0xb23f6964
Sat Apr 25 14:25:55 2020 kern.err kernel: [95403.416894] mptcp_add_sock: token 0xb23f6964 pi 1, src_addr:192.168.8.219:20611 dst_addr:130.104.228.140:80
Sat Apr 25 14:25:55 2020 kern.err kernel: [95403.484678] mptcp_add_sock: token 0xb23f6964 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Sat Apr 25 14:25:55 2020 kern.err kernel: [95403.492815] __mptcp_init4_subsockets: token 0xb23f6964 pi 2 src_addr:192.168.0.99:0 dst_addr:130.104.228.140:80 ifidx: 9
[... cut ...]
Sat Apr 25 14:26:22 2020 kern.err kernel: [95431.252723] mptcp_del_sock: Removing subsock tok 0x230d4cd5 pi:3 state 7 is_meta? 0
Sat Apr 25 14:26:22 2020 kern.err kernel: [95431.260412] mptcp_del_sock: Removing subsock tok 0x230d4cd5 pi:1 state 7 is_meta? 0
Sat Apr 25 14:26:22 2020 kern.err kernel: [95431.268078] mptcp_del_sock: Removing subsock tok 0x230d4cd5 pi:2 state 7 is_meta? 0
Sat Apr 25 14:26:22 2020 kern.err kernel: [95431.275772] mptcp_sock_destruct destroying meta-sk token 0x230d4cd5

Ysurac, 2020-04-26:
I don't see problem here.
You can try MPTCP over VPN for this interface in System->OpenMPTCProuter wizard

gnosoz, 2020-04-26:
my suggestion is to monitor the behaviour using munin on the server side and muninlite on the router.

matfil92, 2020-04-26:
@Ysurac if I enable MPTCP over VPN appear this error https://ibb.co/wYddBQb but the sim works correctly because if I'm connect directly to that wifi of postemobile its working. Why appear VPN gateway down?

@gnosoz  Munin not show only the graphics of usage? What you mean?

Ysurac, 2020-04-26:
OK so the problem is not here.
Can you try in System->OpenMPTCProuter and advanced tab to disable TCP Fast Open ?

gnosoz, 2020-04-26:
munin will give you the chance to see an historic of the functioning of all interfaces and system as a whole whilst you concentrate on taking the serious tests to solve the issue

i had a problem with omr-tracker whilst few interfaces where set on vpn for which i reported a small change in the awk command that solved it for me.

the major ally you have are the logs which for sake of good order you should set temporarily for debug level and save in /var/log/syslog

After this you should remove remove physically the interface and reconnect whilst you look for anomalies in the logs

#tail -n100 -f /var/log/syslog

look for the lines related to the interface in question and if it gets too messy just add to above command |grep interface_name to filter only the relevant stuff.

attention that if you set the interface to run under vpn, the system, will give the interface another name and you should interrogate logs for that instead of normal name of the same.

gateway unavailable is given by omr so another check you can do in the logs is to filter them with |grep omr 

this should narrow down the problem

closed 2020-09-14 by github-actions[bot]