Ysurac/openmptcprouter #990: Hello

k0gen, 2020-04-27:
Hello.

Just tried to install on KVM VPS Install on Debian 9 Stretch x86_64 and failed :(

Running:

> `wget -O - https://www.openmptcprouter.com/server/debian9-x86_64.sh | sh`

Ended with error:
>   dh_install
   dh_installdocs
   debian/rules override_dh_installchangelogs
make[1]: Entering directory '/tmp/shadowsocks-libev-3.3.3'
dh_installchangelogs -XChanges
make[1]: Leaving directory '/tmp/shadowsocks-libev-3.3.3'
   dh_installman
   dh_systemd_enable
   dh_installinit
   dh_systemd_start
   dh_perl
   dh_link
   dh_strip_nondeterminism
   dh_compress
   dh_fixperms
   dh_missing
   dh_strip
   dh_makeshlibs
   dh_shlibdeps
dpkg-shlibdeps: error: no dependency information found for /usr/local/lib/libpcre.so.0 (used by debian/shadowsocks-libev/usr/bin/ss-manager)
Hint: check if the library actually comes from a package.
dh_shlibdeps: dpkg-shlibdeps -Tdebian/shadowsocks-libev.substvars debian/shadowsocks-libev/usr/bin/ss-manager debian/shadowsocks-libev/usr/bin/ss-local debian/shadowsocks-libev/usr/bin/ss-redir debian/shadowsocks-libev/usr/bin/ss-tunnel debian/shadowsocks-libev/usr/bin/ss-server returned exit code 255
dh_shlibdeps: Aborting due to earlier error
make: *** [debian/rules:24: binary] Error 25
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2` >

k0gen, 2020-04-27:
Tried to reboot server (booting to Debian 10) and running script again but failed. 
Here is the full (rather big) [install-error.log](https://pastebin.com/7s4ut6sG)

Ended up with rebuilding VPS instance with Ubuntu 18.04 and the update script worked perfectly!

Ysurac, 2020-04-27:
Else it may need "libpcre3-dev" package on your debian.

closed 2020-09-14 by github-actions[bot]