Ysurac/openmptcprouter #993: No MPTCP even when using only 1 wan + frequent browser Connection Refused/Reset

pylaterreur, 2020-04-29:
## Expected Behavior
I would expect MPTCP to "work" with the settings I have, and the TCP connections to always establish. But it seems that it's not the case (no MPTCP and often have Connection Reset in my chrome).

## Current Behavior
http://amiusingmptcp.de/ shows no MPTCP. Also, I often get Connection Reset in my browser when using the setup described in the section below.

## Context (Environment)
My PC NIC is plugged to a PLC, that is plugged on the other side to a switch, on which 3 ports are plugged (1 PLC as said, 1 to the raspberry pi, 1 to the ISP box).
I've configured my PC default route to be 192.168.3.1 (the raspberry pi lan). My raspberry lan and wan1 are actually on the same cable, that goes from raspberry pi 4B to the switch and then to my ISP box (for which I don't want to disable DHCP as other people in my house use this ISP box). Masks for both 192.168.0.x  and 192.168.3.x are 255.255.255.0.

I also have a wwan which is a 4G hotspot (my phone). Enabling or disabling it doesn't change the fact that the MPTCP doesn't seem to work, according to http://amiusingmptcp.de/ .
[image]

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: RPI4B

bnhf, 2020-04-30:
> * OpenMPTCProuter VPS version: N/A

What version of the VPS script are you running? Are you able to verify the VPS is up?

> My raspberry lan and wan1 are actually on the same cable, that goes from raspberry pi 4B to the switch and then to my ISP box (for which I don't want to disable DHCP as other people in my house use this ISP box).

If you want to use DHCP with one of your WANs then it should be on its own interface.  With the RaspberryPi4, this means adding an OpenWRT supported USB3 to Gigabit Ethernet adapter.  Use of Macvlan to sort your LAN and WAN traffic, when mixed on a single switch, requires static IPs all around.

pylaterreur, 2020-04-30:
Thanks for your prompt answer. I removed the VPS script, should I add it back?

bnhf, 2020-04-30:
> I removed the VPS script, should I add it back?

Yes. OMR requires a Virtual Private Server (VPS) to be running in a data center somewhere. "Known to work" VPS providers, VM specifications and supported OS versions are listed in the documentation.

> My raspberry lan and wan1 are actually on the same cable

On the OMR side of the connection, having your LAN and WAN traffic mixed will work, as long as Macvlan is setup (which is by default on the RPi) and you're using static addresses. If you must use DHCP, then you need to have separate physical adapters. Setting up VLANs would probably also work with DHCP, though you'd need to setup a VLAN interface on each device connected to the switch. I use VLAN interfaces myself, but with static addresses.

closed 2020-09-14 by github-actions[bot]