Ysurac/openmptcprouter #994: Test Openmptcprouter

Pokpok-sbo, 2020-04-30:
Hi all,

The Covid crisis forced me to put around thirty employees in telework. I end up with 30 connections.
So I would like to test OpenMPTCProuter.

Here is my list of questions:
Is it possible to install OpenMPTCProuter on an hp elite 8000 sff with a dual core E5700 and 4 GB of RAM?
Can more than one D-link DWM ‑ 156 be used on the different USB ports?
Are there any contraindications that I would not have thought of?

Thank you in advance for your help.

Ysurac, 2020-04-30:
Yes it's possible to install on any x86_64 where Linux can work. A router part and a VPS part are needed.
It's should be possible to use these dongle, but it's only 3G dongle, so not really fast and this may need some manual configuration.

closed 2020-08-12 by Ysurac

Adorfer, 2020-08-13:
Remark upon USB-Network adaptors: Do not expect configurations with more than 1 (one!) USB-NIC (wired, wireless, LTE) to work reliable/stable/performant. 

why? most USB drivers tend to generate IO-load. doing this in a server context may drag your CPU down without you realizing the reasons. sometimes even timing is critical and multiple dongles on the same USB controller (just on different places on the hubs) will cause timeout and packetloss scenarios. 

I do not say that there are boards with reliable USB interfaces and USB dongles doing a good job. But you have to find them during evaluation in order to separate the good ones from the bad ones. 

If you are on any time- and personpower budget, just purchase dedicated LTE-Routers with eth interface like Netgear NighthawkM1. costly off course, but you can even put good X-antennas outside in different directions in order to aim at different cell towers.