Ysurac/openmptcprouter #995: Problem connection vps

kyouran38, 2020-05-01:
## Expected Behavior
Since Yesteserday my RBPI can connect my milkywan server

## Current Behavior

## Context (Environment)
Lost of connection, didn't change anything work as a charm yesterday

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptctp
  - OpenMPTCProuter platform:  RPI3B_x8664

[https://ibb.co/0y3mcCB](url)

Fri May  1 16:11:22 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:11:29 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:11:35 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:11:43 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:11:48 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:11:56 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:01 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:09 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:15 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:23 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:28 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:36 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:41 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:49 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:12:55 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:03 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:08 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:16 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:21 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:29 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:35 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:43 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:48 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:13:56 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:14:01 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:14:09 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:14:14 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
Fri May  1 16:14:23 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)

Ysurac, 2020-05-01:
The server and the API answer, so this looks like a VPS key problem. Check that the API key is correct, if yes try to reboot the VPS and the router.

kyouran38, 2020-05-01:
How can i do that ?

kyouran38, 2020-05-01:
You are a lifesaver everything is working fine now :)

kyouran38, 2020-05-01:
Thank you so much for your help

closed 2020-05-01 by Ysurac

omninet1, 2022-09-25:
sorry to respond to this closed issue but I'm getting the same thing
My setup was never working - I'm in the process of building it
I don't use any cloud for the vps - I'm running it in my data center and it's running on vmware
I used the opemmptcprouter image on both sides - both seem to be working fine
I have local (192.168.100.X) network connectivity to the internet on both sides
The vps site has a firewall infront of it 
I opened a bunch of ports forwarding to the openmptcprouter vm but still no go
I see hits on the inbound policy on my firewall but nothing
any ideas? 
thank you

Ysurac, 2022-09-25:
Please open a new issue and use template, this issue is closed since 2020.