Ysurac/openmptcprouter #996: OMR-Bypass by port broken

Bestle, 2020-05-02:
## Expected Behavior
To be able to bypass specific ports for any protocol combination without issues.

## Current Behavior
If I add a port to OMR-bypass with protocol UDP, the traffic is correctly bypassing OMR and uses a single WAN. The same happens if I add a single rule for a port with protocol TCP. However, if I add a rule with protocol as "ALL" or even two rules, one with TCP and one with UDP, traffic no longer bypasses OMR.

## Steps to Reproduce the Problem

  1. Add bypass rule for a port with protocol "ALL", test if traffic bypasses using pktstat on VPS.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d
  - OpenMPTCProuter VPS version: VPS 0.1015
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-05-02:
What is the result of ```iptables-save``` via SSH on the router ?

Bestle, 2020-05-02:
`root@OpenMPTCProuter:~# iptables-save
# Generated by iptables-save v1.8.3 on Sat May  2 12:26:07 2020
*nat
:PREROUTING ACCEPT [15738:1864895]
:INPUT ACCEPT [14197:1208159]
:OUTPUT ACCEPT [53824:4293600]
:POSTROUTING ACCEPT [8222:686723]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_local_out - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -p tcp -j ss_rules_pre_src
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i eth1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth3 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A OUTPUT -p tcp -j ss_rules_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o eth1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth3 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A ss_rules_dst -m set --match-set omr_dst_bypass_all dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_tun0 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth3 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth2 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth1 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth0 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_lo dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass_all dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p tcp -j REDIRECT --to-ports 1100
-A ss_rules_local_out -m set --match-set omr_dst_bypass_all dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x539 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_tun0 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5396 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth3 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5395 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth2 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5394 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth1 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5393 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth0 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5392 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_lo dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5391 -j RETURN
-A ss_rules_local_out -m mark --mark 0x539 -j RETURN
-A ss_rules_local_out -m mark --mark 0x5396 -j RETURN
[... 283 lines cut ...]
-A zone_wan_dest_ACCEPT -o eth2 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth2 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth3 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth3 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth2 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth3 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth2 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth3 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Sat May  2 12:26:07 2020`

Bestle, 2020-05-02:
root@OpenMPTCProuter:~# iptables-save

# Generated by iptables-save v1.8.3 on Sat May  2 12:28:27 2020
*nat
:PREROUTING ACCEPT [15897:1891853]
:INPUT ACCEPT [14350:1220473]
:OUTPUT ACCEPT [54317:4332254]
:POSTROUTING ACCEPT [8312:693625]
:MINIUPNPD - [0:0]
:MINIUPNPD-POSTROUTING - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_vpn_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_vpn_rule - [0:0]
:prerouting_wan_rule - [0:0]
:ss_rules_dst - [0:0]
:ss_rules_forward - [0:0]
:ss_rules_local_out - [0:0]
:ss_rules_pre_src - [0:0]
:ss_rules_src - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_vpn_postrouting - [0:0]
:zone_vpn_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -p tcp -j ss_rules_pre_src
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i eth0 -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i eth1 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth3 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A OUTPUT -p tcp -j ss_rules_local_out
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o eth0 -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o eth1 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth3 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A ss_rules_dst -m set --match-set omr_dst_bypass_all dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_tun0 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth3 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth2 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth1 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_eth0 dst -j RETURN
-A ss_rules_dst -m set --match-set omr_dst_bypass_lo dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass_all dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_bypass dst -j RETURN
-A ss_rules_dst -m set --match-set ss_rules_dst_forward dst -j ss_rules_forward
-A ss_rules_dst -m comment --comment "dst_default: forward" -j ss_rules_forward
-A ss_rules_forward -p tcp -j REDIRECT --to-ports 1100
-A ss_rules_local_out -m set --match-set omr_dst_bypass_all dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x539 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_tun0 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5396 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth3 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5395 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth2 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5394 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth1 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5393 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_eth0 dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5392 -j RETURN
-A ss_rules_local_out -m set --match-set omr_dst_bypass_lo dst -j RETURN
-A ss_rules_local_out -m mark --mark 0x5391 -j RETURN
-A ss_rules_local_out -m mark --mark 0x539 -j RETURN
[... 285 lines cut ...]
-A zone_wan_dest_ACCEPT -o eth2 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth2 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth3 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth3 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth2 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth3 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth2 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth3 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Sat May  2 12:28:27 2020

Bestle, 2020-05-02:
The specific use case here is to bypass game traffic from COD Warzone. It's free but a hefty download if you want to try to reproduce :)

Bestle, 2020-05-02:
Wait could this be a problem that the port is actually the source port and omr-bypass works with destination? If so, could we get it to work with source too?

Ysurac, 2020-05-02:
We often know destination port but not source port.
I can't test easily COD Warzone, there is no Linux release and I don't have windows.
I will make some tests on port bypass.

Bestle, 2020-05-02:
What would the iptables command be to add a bypass for the source port so that I can test?

Bestle, 2020-05-05:
Managed to use iptables to add a bypass for source port and it works great.

If you could add that to omr-bypass I would be forever grateful.

Bestle, 2020-05-11:
Any chance you could sneak this in for next release?

Ysurac, 2020-05-12:
This will be in next release.

closed 2020-09-15 by github-actions[bot]