Ysurac/openmptcprouter #998: No IPv6 access

Chatissimus, 2020-05-03:
## Expected Behavior
IPv6 connexion available 

## Current Behavior
random issue on IPv6 connexion

## Steps to Reproduce the Problem

after reboot OMR detect IPv6
[image]

and it's possible to get IPv6 address
root@OpenMPTCProuter:~# curl -6 www.google.com
<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="fr"><head><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><metacontent="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop= 

"randomly" OMR displays "No IPv6 access"
[image]

indeed
root@OpenMPTCProuter:~# curl -6 www.google.com
curl: (7) Failed to connect to www.google.com port 80: Connection refuse

even if on VPS it's fine
< OpenMPTCProuter VPS 0.1015 >
Last login: Sun May  3 14:39:04 2020 from 86.216.x.x
cedric@xxxxxxx:~$ curl -6 www.google.com
<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="fr"> 

## Context (Environment)
random issue to access ipv6 website

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-05-03:
What is the VPN used ?
What is the result of ```ip -6 a``` and ```ip -6 r``` via SSH on the router ?

Chatissimus, 2020-05-03:
Hello,

it uses glorytun TCP

```
root@OpenMPTCProuter:~# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 state UP qlen 1000
    inet6 fd8b:242f:4efc::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::dea6:32ff:fe2b:6faf/64 scope link
       valid_lft forever preferred_lft forever
10: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 state UP qlen 50
    inet6 fe80::64ca:96ff:fe72:b5af/64 scope link
       valid_lft forever preferred_lft forever
11: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
    inet6 fe80::3843:1bff:fe11:87d3/64 scope link
       valid_lft forever preferred_lft forever
19: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
    inet6 fe80::e5b:8fff:fe27:9a64/64 scope link
       valid_lft forever preferred_lft forever
20: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 1000
    inet6 fe80::b1b1:e293:accb:28d7/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
23: ifb4wan2: <BROADCAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 32
    inet6 fe80::4dd:94ff:fe98:507e/64 scope link
       valid_lft forever preferred_lft forever
26: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 state UNKNOWN qlen 1000
    inet6 fe80::a00:2/128 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:ff02/64 scope link
       valid_lft forever preferred_lft forever
```

```
root@OpenMPTCProuter:~# ip -6 r
default from fe80::a00:2 dev 6in4-omr6in4 proto static metric 6 pref medium
fd8b:242f:4efc::/64 dev eth0 proto static metric 2 pref medium
unreachable fd8b:242f:4efc::/48 dev lo proto static metric 2147483647 pref medium
fe80::a00:2 dev 6in4-omr6in4 proto static metric 6 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev wan1 proto kernel metric 256 pref medium
fe80::/64 dev wan2 proto kernel metric 256 pref medium
fe80::/64 dev ifb4wan2 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 proto static metric 6 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 metric 1024 pref medium
```

Ysurac, 2020-05-03:
And the result of ```ip a```

Chatissimus, 2020-05-03:
```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:2b:6f:af brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd8b:242f:4efc::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::dea6:32ff:fe2b:6faf/64 scope link
       valid_lft forever preferred_lft forever
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: ifb0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether f2:6c:46:a9:1b:c6 brd ff:ff:ff:ff:ff:ff
5: ifb1: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 12:01:6d:a5:24:20 brd ff:ff:ff:ff:ff:ff
6: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
7: wlan0: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether dc:a6:32:2b:6f:b0 brd ff:ff:ff:ff:ff:ff
10: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc noqueue state UP group default qlen 50
    link/ether 66:ca:96:72:b5:af brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.10/24 brd 192.168.0.255 scope global wan1
       valid_lft forever preferred_lft forever
    inet6 fe80::64ca:96ff:fe72:b5af/64 scope link
       valid_lft forever preferred_lft forever
11: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc htb state UP group default qlen 100
    link/ether 3a:43:1b:11:87:d3 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.10/24 brd 192.168.1.255 scope global wan2
       valid_lft forever preferred_lft forever
    inet6 fe80::3843:1bff:fe11:87d3/64 scope link
       valid_lft forever preferred_lft forever
19: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 100
    link/ether 0c:5b:8f:27:9a:64 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.10/24 brd 192.168.2.255 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::e5b:8fff:fe27:9a64/64 scope link
       valid_lft forever preferred_lft forever
20: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::b1b1:e293:accb:28d7/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
23: ifb4wan2: <BROADCAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc htb state UNKNOWN group default qlen 32
    link/ether 06:dd:94:98:50:7e brd ff:ff:ff:ff:ff:ff
    inet6 fe80::4dd:94ff:fe98:507e/64 scope link
       valid_lft forever preferred_lft forever
26: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.255.2 peer 10.255.255.1
    inet6 fe80::a00:2/128 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:ff02/64 scope link
       valid_lft forever preferred_lft forever

```

Ysurac, 2020-05-03:
I don't see any error here.
I think the problem is related to Shadowsocks.
I would need the result of ```uci show shadowsocks-libev``` on the router (without IP and/or key) and check in ```/etc/shadowsocks-libev/manager.json``` on the VPS that server part listen on ```"[::0]"```

Chatissimus, 2020-05-03:
```
root@OpenMPTCProuter:~# uci show shadowsocks-libev
shadowsocks-libev.hi=ss_redir
shadowsocks-libev.hi.server='sss0'
shadowsocks-libev.hi.local_port='1100'
shadowsocks-libev.hi.mode='tcp_and_udp'
shadowsocks-libev.hi.timeout='1000'
shadowsocks-libev.hi.fast_open='1'
shadowsocks-libev.hi.verbose='0'
shadowsocks-libev.hi.syslog='1'
shadowsocks-libev.hi.reuse_port='1'
shadowsocks-libev.hi.mptcp='1'
shadowsocks-libev.hi.ipv6_first='1'
shadowsocks-libev.hi.no_delay='1'
shadowsocks-libev.hi.ebpf='0'
shadowsocks-libev.hi.local_address='::'
shadowsocks-libev.hi2=ss_redir
shadowsocks-libev.hi2.server='sss0'
shadowsocks-libev.hi2.local_address='0.0.0.0'
shadowsocks-libev.hi2.local_port='1100'
shadowsocks-libev.hi2.mode='tcp_and_udp'
shadowsocks-libev.hi2.timeout='1000'
shadowsocks-libev.hi2.fast_open='1'
shadowsocks-libev.hi2.verbose='0'
shadowsocks-libev.hi2.syslog='1'
shadowsocks-libev.hi2.reuse_port='1'
shadowsocks-libev.hi2.mptcp='1'
shadowsocks-libev.hi2.ipv6_first='1'
shadowsocks-libev.hi2.no_delay='1'
shadowsocks-libev.hi2.ebpf='0'
shadowsocks-libev.ss_rules=ss_rules
shadowsocks-libev.ss_rules.disabled='0'
shadowsocks-libev.ss_rules.src_default='forward'
shadowsocks-libev.ss_rules.dst_default='forward'
shadowsocks-libev.ss_rules.local_default='forward'
shadowsocks-libev.ss_rules.redir_tcp='all'
shadowsocks-libev.sss0=server
shadowsocks-libev.sss0.disabled='0'
shadowsocks-libev.sss0.server_port='65101'
shadowsocks-libev.sss0.method='chacha20-ietf-poly1305'
shadowsocks-libev.sss0.obfs='0'
shadowsocks-libev.sss0.server='45.13.x.x'
shadowsocks-libev.sss0.key='xxxx'
shadowsocks-libev.sss0.obfs_host='45.13.x.x'
shadowsocks-libev.dns=ss_tunnel
shadowsocks-libev.dns.disabled='1'
shadowsocks-libev.dns.mode='tcp_and_udp'
shadowsocks-libev.dns.server='sss0'
shadowsocks-libev.dns.local_port='5353'
shadowsocks-libev.dns.tunnel_address='8.8.8.8:53'
shadowsocks-libev.hi3=ss_redir
shadowsocks-libev.hi3.server='sss0'
shadowsocks-libev.hi3.local_address='0.0.0.0'
shadowsocks-libev.hi3.local_port='1101'
shadowsocks-libev.hi3.mode='tcp_and_udp'
shadowsocks-libev.hi3.timeout='1000'
shadowsocks-libev.hi3.fast_open='1'
shadowsocks-libev.hi3.reuse_port='1'
shadowsocks-libev.hi3.syslog='1'
shadowsocks-libev.hi3.mptcp='1'
shadowsocks-libev.hi3.no_delay='1'
shadowsocks-libev.hi3.verbose='0'
shadowsocks-libev.hi3.ebpf='0'
shadowsocks-libev.hi4=ss_redir
shadowsocks-libev.hi4.server='sss0'
shadowsocks-libev.hi4.local_address='0.0.0.0'
shadowsocks-libev.hi4.local_port='1101'
shadowsocks-libev.hi4.mode='tcp_and_udp'
shadowsocks-libev.hi4.timeout='1000'
shadowsocks-libev.hi4.fast_open='1'
shadowsocks-libev.hi4.reuse_port='1'
shadowsocks-libev.hi4.syslog='1'
shadowsocks-libev.hi4.mptcp='1'
shadowsocks-libev.hi4.no_delay='1'
shadowsocks-libev.hi4.verbose='0'
shadowsocks-libev.hi4.ebpf='0'
shadowsocks-libev.tracker=ss_local
shadowsocks-libev.tracker.server='sss0'
shadowsocks-libev.tracker.local_address='127.0.0.1'
shadowsocks-libev.tracker.local_port='1111'
shadowsocks-libev.tracker.mode='tcp_and_udp'
shadowsocks-libev.tracker.timeout='600'
shadowsocks-libev.tracker.fast_open='1'
shadowsocks-libev.tracker.syslog='0'
shadowsocks-libev.tracker.reuse_port='1'
shadowsocks-libev.tracker.mptcp='1'
shadowsocks-libev.tracker.verbose='0'
shadowsocks-libev.tracker.no_delay='1'
shadowsocks-libev.tracker.ebpf='0'

```

```
cedric@xxxxxx:/etc/shadowsocks-libev$ cat manager.json
{
    "mptcp": true,
    "ipv6_first": true,
    "no_delay": true,
    "ebpf": false,
    "server": [
        "[::0]",
        "0.0.0.0"
    ],
    "method": "chacha20-ietf-poly1305",
    "fast_open": true,
    "timeout": 600,
    "port_key": {
        "65101": "xxx",
    },
    "local_port": 1081,
    "verbose": 0,
    "acl": "/etc/shadowsocks-libev/local.acl",
    "mode": "tcp_and_udp",
    "reuse_port": true,
    "prefer_ipv6": false
}
```

Ysurac, 2020-05-03:
All the configuration seems to be ok.
What do you in State->System log, on the router interface, about ss-redir when there is a connection problem ? Do you have also something about IPv6 when this fail on the VPS via ```dmesg``` command ?

Chatissimus, 2020-05-03:
the only log have between a time where ipv6 is working and not working

```
Sun May  3 17:22:35 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Sun May  3 17:22:36 2020 user.notice omr-tracker: Launching...
Sun May  3 17:22:36 2020 user.notice omr-tracker: Launched
Sun May  3 17:22:36 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
Sun May  3 17:25:16 2020 user.notice post-tracking-post-tracking: Change master interface from wan2 (72 ms) to WAN3 (45 ms)
Sun May  3 17:25:19 2020 user.notice post-tracking-post-tracking: Change master interface from WAN3 (45 ms) to wan2 (19 ms)
Sun May  3 17:26:57 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Sun May  3 17:36:07 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
```

ss-redir is mentioned only at startup
```
Sun May  3 17:10:27 2020 user.info : using tcp fast open
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: enable TCP no-delay
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: resolving hostname to IPv6 address first
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: initializing ciphers... chacha20-ietf-poly1305
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: listening at :::1100
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: tcp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: UDP relay enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: udp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15711]: running from root user
Sun May  3 17:10:27 2020 user.info : using tcp fast open
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: enable TCP no-delay
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: resolving hostname to IPv6 address first
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: initializing ciphers... chacha20-ietf-poly1305
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: listening at 0.0.0.0:1100
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: tcp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: UDP relay enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: udp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15712]: running from root user
Sun May  3 17:10:27 2020 user.info : using tcp fast open
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: enable TCP no-delay
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: initializing ciphers... chacha20-ietf-poly1305
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: using tcp fast open
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: enable TCP no-delay
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: initializing ciphers... chacha20-ietf-poly1305
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: listening at 0.0.0.0:1101
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: tcp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: listening at 127.0.0.1:1111
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: UDP relay enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: tcp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: udp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15713]: running from root user
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: udprelay enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: udp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-local[15710]: running from root user
Sun May  3 17:10:27 2020 user.info : using tcp fast open
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: enable TCP no-delay
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: initializing ciphers... chacha20-ietf-poly1305
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: listening at 0.0.0.0:1101
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: tcp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: UDP relay enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: udp port reuse enabled
Sun May  3 17:10:27 2020 daemon.info /usr/bin/ss-redir[15714]: running from root user
```

i found nothing via dmesg

Ysurac, 2020-05-03:
In System->OpenMPTCProuter and Advanced settings tab, you can try to set the "select master interface" to no change

Chatissimus, 2020-05-03:
ipv6 up again 

log since it was down

```
Sun May  3 17:47:35 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.1.1 dev wan2
Sun May  3 17:47:35 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sun May  3 17:47:35 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Sun May  3 17:47:35 2020 user.notice Shadowsocks: Rules DOWN
Sun May  3 17:47:36 2020 user.notice Shadowsocks: Rules UP
Sun May  3 17:47:37 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun May  3 17:47:37 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun May  3 17:47:38 2020 user.notice omr-bypass: Reload dnsmasq...
Sun May  3 17:47:38 2020 daemon.info dnsmasq[15664]: read /etc/hosts - 4 addresses
Sun May  3 17:47:38 2020 daemon.info dnsmasq[15664]: read /tmp/hosts/odhcpd - 0 addresses
Sun May  3 17:47:38 2020 daemon.info dnsmasq[15664]: read /tmp/hosts/dhcp.cfg01411c - 12 addresses
Sun May  3 17:47:38 2020 daemon.info dnsmasq-dhcp[15664]: read /etc/ethers - 0 addresses
Sun May  3 17:47:38 2020 user.notice omr-bypass: OMR-ByPass is running
Sun May  3 17:47:38 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:38 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:39 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:39 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:39 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:39 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:41 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:41 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:41 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:41 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:44 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:44 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:46 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun May  3 17:47:47 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:47 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:47 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:47 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:50 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:50 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:52 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:52 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:52 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:52 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:52 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:52 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:53 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:54 2020 daemon.err /usr/bin/ss-redir[15711]: [udp] remote_recv_bind: Address not available
Sun May  3 17:47:54 2020 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Sun May  3 17:47:54 2020 user.notice Shadowsocks: Rules DOWN
Sun May  3 17:47:55 2020 user.notice Shadowsocks: Rules UP
Sun May  3 17:47:55 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun May  3 17:47:55 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun May  3 17:47:56 2020 user.notice omr-bypass: Reload dnsmasq...
Sun May  3 17:47:56 2020 daemon.info dnsmasq[15664]: read /etc/hosts - 4 addresses
Sun May  3 17:47:56 2020 daemon.info dnsmasq[15664]: read /tmp/hosts/odhcpd - 0 addresses
Sun May  3 17:47:56 2020 daemon.info dnsmasq[15664]: read /tmp/hosts/dhcp.cfg01411c - 12 addresses
Sun May  3 17:47:56 2020 daemon.info dnsmasq-dhcp[15664]: read /etc/ethers - 0 addresses
Sun May  3 17:47:56 2020 user.notice omr-bypass: OMR-ByPass is running
Sun May  3 18:00:00 2020 cron.err crond[3116]: USER root pid 6733 cmd /etc/init.d/vnstat_backup backup
Sun May  3 18:19:42 2020 user.notice post-tracking-post-tracking: Get status and settings for WAN3...
Sun May  3 18:28:52 2020 user.notice post-tracking-post-tracking: Get status and settings for WAN3... Done
```

Chatissimus, 2020-05-03:
down again, get this in log 
```
Sun May  3 18:48:45 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sun May  3 18:51:27 2020 daemon.err /usr/bin/ss-local[15710]: getpeername: Socket not connected
```

Ysurac, 2020-05-03:
Previous log VPN was down, this time it's shadowsocks, this seems to be related to some connections problems.
You should set wan2 to multipath enable, handover doesn't do anything.

Chatissimus, 2020-05-05:
in fact the multipath setup is like that
wan1 > not used
wan2 > master (ethernet ADSL)
wan3 > enabled (USB 4G) (it was set to enabled since ~17:00 was handover before )

looks fine since monday, maybe just connection issues

closed 2020-08-12 by Ysurac