Ysurac/openmptcprouter #1000: LuCi Shorewall accept from 1 IP only

gnosoz, 2020-05-04:
Setting nat or port forward to accept only from a specified ip results in the port being fully open.

```
 DNAT   net   vpn:@OMR_ADDR PROTO PORT
```
instead as per shorewall manuals it should write the following rule

```
DNAT   net:ORIGIN_IP  vpn:@OMR_ADDR PROTO PORT
```

After this I thought the block would happen in the iptables of the router but also there the port was showing fully open .

This can be a major security concern IMHO especially as some services might offer little to no configuration internally to block unwanted requests like for example SNMP...

Ysurac, 2020-05-04:
Please use the issue template...
Blocking by IP on the VPS side is not yet supported automatically (never requested yet), this can still be set manually.
SNMPv3 support user/pass and encryption.

gnosoz, 2020-05-04:
Thanks Ycarus,
yes I noticed they seemed unsupported hence why I reported it.

I will be setting such rules manually in the vps but may be is worth having
a look at this sma change because in LuCi the option is available therefore
it might gives other users a false sense of security without any
information to the contrary?

closed 2020-05-05 by gnosoz