Ysurac/openmptcprouter #1001: MICRO COUPURES

bosseb, 2020-05-04:
## Expected Behavior
Bonjour j'ai des micros coupures depuis quelques jour, je suis chez MILKYWAN

dans le journal système à 9h58 (pour 11h58)

  - OpenMPTCProuter version:  0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: 

dans le journal système à 9h58 (pour 11h58)

Mon May  4 09:53:01 2020 user.notice post-tracking-post-tracking: Change master interface from wan1 (45 ms) to wan3 (29 ms)
Mon May  4 09:53:06 2020 user.notice post-tracking-post-tracking: Set server vps (45.13.104.175) default route via 192.168.10.1
Mon May  4 09:57:09 2020 daemon.err /usr/bin/ss-redir[587]: send: Connection reset by peer
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.451709] ------------[ cut here ]------------
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.453672] WARNING: CPU: 3 PID: 4314 at cfg80211_vendor_cmd_reply+0x2db1/0x3670 [cfg80211]
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.455628] Modules linked in: ath9k ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option iwldvm ipw cdc_mbim ath9k_hw ath usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx sierra_net sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb iwlmvm rt2x00lib rndis_host qmi_wwan pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mct_u232 mcs7830 mac80211 lzo lz4 keyspan kalmia iwlwifi iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa bas_gigaset ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.463842]  xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT wireguard visor via_velocity via_rhine usbserial usbnet usb_gigaset tulip ts_fsm ts_bm solos_pci sky2 skge sis900 sis190 ser_gigaset rtl8150 r8712u(C) r8169 r8152 r6040 pppox ppp_generic pegasus pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.472080]  nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mmc_spi macvlan lzo_decompress lzo_compress lz4_compress kaweth iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip6table_raw ip_tables hso gigaset forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm bnx2 atl2 atl1e atl1c atl1 asn1_decoder arptable_filter arpt_mangle arp_tables 8390 8139too 8139cp sch_cake sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.480308]  act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs evdev i2c_gpio i2c_dev spi_ks8995 ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig isdn_bsdcomp dss1_divert isdn slhc 3c59x igb e1000
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.488519]  ifb nat46 ip6_udp_tunnel udp_tunnel sit sctp libcrc32c ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo br2684 atm nls_utf8 zram zsmalloc natsemi eeprom_93cx6 sha256_ssse3 sha256_generic sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gcm echainiv des_generic deflate ctr cbc authenc crypto_acompress nls_iso8859_1 nls_cp437 vfat fat sdhci_pltfm xhci_plat_hcd ehci_platform gpio_button_hotplug button_hotplug tg3 b44 ssb ptp realtek broadcom bcm_phy_lib pps_core mii libphy
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.494703] CPU: 3 PID: 4314 Comm: rpcd Tainted: G        WC        4.19.108 #0
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.496747] Hardware name: Intel Corporation NUC7i3BNH/NUC7i3BNB, BIOS BNKBL357.86A.0076.2019.0219.1646 02/19/2019
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.498727] RIP: 0010:cfg80211_vendor_cmd_reply+0x2db1/0x3670 [cfg80211]
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.500691] Code: 0f 84 35 ff ff ff 48 89 df e8 1b 94 bb e0 b8 a6 ff ff ff eb 9b eb ef b8 97 ff ff ff eb 92 48 c7 c7 00 34 b2 a0 e8 18 ad 5c e0 <0f> 0b 48 89 df e8 f5 93 bb e0 b8 ea ff ff ff e9 72 ff ff ff 66 66
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.502717] RSP: 0018:ffffc9000090ba78 EFLAGS: 00010286
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.504752] RAX: 0000000000000024 RBX: ffff88826e7ff2c0 RCX: 0000000000000000
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.506733] RDX: ffff888276b9bc00 RSI: ffff888276b963b8 RDI: ffff888276b963b8
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.508675] RBP: ffffc9000090bab0 R08: 0000000000000000 R09: 00000000000008d3
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.510611] R10: 0000000000000007 R11: 0000000000000000 R12: ffffc9000090bae0
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.512547] R13: 0000000000000000 R14: ffff8882705a1014 R15: ffff888271460300
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.514476] FS:  00007f6857f38d48(0000) GS:ffff888276b80000(0000) knlGS:0000000000000000
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.516427] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.518371] CR2: 00007ffe5aa57c78 CR3: 0000000270412003 CR4: 00000000001606e0
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.520318] Call Trace:
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.522252]  genl_family_rcv_msg+0x1a3/0x350
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.524182]  genl_rcv_msg+0x43/0x90
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.526105]  ? genl_family_rcv_msg+0x350/0x350
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.528034]  netlink_rcv_skb+0x3f/0x110
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.529974]  genl_rcv+0x23/0x40
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.531883]  netlink_unicast+0x16b/0x1e0
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.533805]  netlink_sendmsg+0x1c0/0x3c0
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.535725]  sock_sendmsg+0x15/0x20
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.537639]  ___sys_sendmsg+0x25b/0x280
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.539559]  ? sock_destroy_inode+0x2a/0x30
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.541501]  ? destroy_inode+0x39/0x60
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.543453]  ? evict+0x134/0x190
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.545376]  ? iput+0x14b/0x190
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.547298]  ? _cond_resched+0x14/0x30
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.549244]  ? __dentry_kill+0x11f/0x160
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.551166]  __sys_sendmsg+0x40/0x70
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.553086]  __x64_sys_sendmsg+0x1a/0x20
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.555009]  do_syscall_64+0x57/0x260
Mon May  4 09:58:10 2020 kern.warn kernel: [81506.556927]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[... cut ...]
Mon May  4 09:58:10 2020 kern.warn kernel: [81507.019472] RDX: 0000000000000000 RSI: 00007ffe5aa58a48 RDI: 0000000000000007
Mon May  4 09:58:10 2020 kern.warn kernel: [81507.021459] RBP: 000000000000002e R08: 0000000000000000 R09: 0000000000000000
Mon May  4 09:58:10 2020 kern.warn kernel: [81507.023378] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
Mon May  4 09:58:10 2020 kern.warn kernel: [81507.025287] R13: 00007ffe5aa5efdc R14: 0000000000000001 R15: 0000558b8fd74258
Mon May  4 09:58:10 2020 kern.warn kernel: [81507.027191] ---[ end trace 9e6cdfd12addb59c ]---

Ysurac, 2020-05-04:
Le soucis semble lié à cfg80211. Le wifi est utilisé ?
Je vois aussi un changement de wan, il peut être mieux de mettre dans Système->OpenMPTCProuter et l'onglet paramètres avancés, "Selection de l'interface maître" à "Pas de changements".

bosseb, 2020-05-04:
non je n'utilise pas le wifi de OMR j'utilise un WIFI en mode point d'accès, branché sur mon HUB

bosseb, 2020-05-04:
du coup j'ai désactivé le composant wifi dans le bios du NUC sur lequel OMR est installé , par contre j'ai ce message dans le journal système d'OMR 
Mon May  4 12:12:16 2020 kern.notice kernel: [   64.471152] i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.
Mon May  4 12:12:16 2020 kern.notice kernel: [   64.471371] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915

C'est normal ? Je dois faire quoi ?

Ysurac, 2020-05-04:
Ce n'est pas grave. Juste un firmware manquant pour la gestion de l'alimentation.

bosseb, 2020-05-04:
OK merci ... pour l'instant plus de coupures, je surveille :)

closed 2020-09-14 by github-actions[bot]