Ysurac/openmptcprouter #1004: Unable to put WAN in DHCP mode

DeeJayMX, 2020-05-04:
## Expected Behavior
WAN interface can be set to use DHCP or other in wizard.

## Current Behavior
When I set DHCP in the wizard, Apply configuration and wait.
when I return on the wizard page, the settings return to "Static"
If I change the protocol in the interface page, after apply it comes up to static

## Possible Solution
Maybe it has something to do with VLAN in the interface panel
I have my 3 WAN in VLAN like eth0.11 eth0.12 eth0.13

## Steps to Reproduce the Problem

  1. Just go in the wizard and put them in DHCP mode
  2. Go in interface panel and you will see the settings go from static to DHCP to static

## Context (Environment)
I prefer to have everything plug and play because I will have to change the modems sometimes and I don't want to go in the configuration of the routers everytime I change modem.

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPi4B 4GB

Ysurac, 2020-05-04:
I need your configuration, available in System->OpenMPTCProuter and "Show all settings" tab without keys and/or VPS IP.

DeeJayMX, 2020-05-04:
Here it comes:

`dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].domain='harcourt'
dhcp.@dnsmasq[0].local='/harcourt/'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.leasetime='12h'
dhcp.lan.limit='150'
dhcp.lan.start='100'
dhcp.lan.ra_management='1'
dhcp.lan.dhcp_option='66,"192.168.2.10"' '42,192.168.2.10'
dhcp.lan.dhcpv6='server'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='60:A4:4C:E7:20:DA'
dhcp.@host[0].name='HARCOURT-SERVER'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.2.115'
dhcp.@host[1]=host
dhcp.@host[1].mac='BC:AE:C5:5A:BF:6E'
dhcp.@host[1].name='HARCOURT-MUSIC'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.2.100'
dhcp.@host[2]=host
dhcp.@host[2].mac='08:00:27:5A:7D:4E'
dhcp.@host[2].name='ELASTIX-SRV'
dhcp.@host[2].dns='1'
dhcp.@host[2].ip='192.168.2.10'
dhcp.@host[3]=host
dhcp.@host[3].mac='00:0E:08:c1:e0:90'
dhcp.@host[3].name='SPA3102'
dhcp.@host[3].dns='1'
dhcp.@host[3].ip='192.168.2.11'
dhcp.@host[4]=host
dhcp.@host[4].mac='F8:D1:11:AB:68:2E'
dhcp.@host[4].name='CanonMF4100'
dhcp.@host[4].dns='1'
dhcp.@host[4].ip='192.168.2.201'
dhcp.@host[5]=host
dhcp.@host[5].mac='3C:2A:F4:6A:51:4E'
dhcp.@host[5].name='BrotherRosa'
dhcp.@host[5].dns='1'
dhcp.@host[5].ip='192.168.2.202'
[... 1588 lines cut ...]
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'wan3' 'wan1' 'wan2' 'wan3' 'wan1' 'wan2' 'wan3' 'eth0.11' 'eth0.12'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'`

DeeJayMX, 2020-05-04:
if you need it with everything I can send it to U directly.
Thanks

Ysurac, 2020-05-04:
Interfaces are set as macvlan but you also use vlan. So you have to remove macvlan before using them like that in Network->Macvlan.

DeeJayMX, 2020-05-04:
I never set it !
I didn't go to that page before...

Ysurac, 2020-05-04:
It's a default setting.

DeeJayMX, 2020-05-04:
It's good right now.
I have to add the interface and put them in VLAN mode before go to the wizard.
now it's good.

closed 2020-05-04 by DeeJayMX