Ysurac/openmptcprouter #1006: Forward 5 public ip to ip in lan

DavideCruccolini, 2020-05-04:
Hi everyone, I have been using this fantastic system for some time now and I must say that I am very satisfied.
But now I have a new need, I would like to use my public ip to hijack them directly on some ip of my lan.
But I would like to make sure that the nat was eliminated, in practice I would like the public IP to be intended directly on the LAN machine.
It would be possible?

Thanks

JorgeMoort89, 2020-05-05:
Same problem, maybe with shorewall forward on VPS?

JorgeMoort89, 2020-05-05:
Can I add a DNAT rule on shorewall for a range of ports (1-64999) like basic OMR configuration but with LAN IP? 
Something like this:
DNAT            net             vpn:$PRIVATE_IP   tcp     1-64999
DNAT            net             vpn:$PRIVATE_IP   udp     1-64999

I'll add a static route to access 192.168.100.0 via 10.255.255.2 and PRIVATE_IP in params.vpn

gnosoz, 2020-05-05:
I am interested too and think a DMZ might be the best solution although
firewall rules are readily available.

https://en.m.wikipedia.org/wiki/DMZ_(computing)

technically it should work like this:
  - VPS side you will have the wan interface set with a pool of addresses
you bought from the ISP or that you own.
 - VPS firewall set with a rule or set proper DMZ zone and assign to an
internal machine which in our case is the wonderful openmptcprouter

Now you should be able to set the router with snat / port forward to a set
local machine if address of destination is IP_XXX?

Another consideration is whether you need a proper DMZ or just a forward to
a local IP.
DMZ create separate subsets therefore that machine/s set in the DMZ will
not communicate with the rest of the lan whilst port forward doesn't
segregate.

JorgeMoort89, 2020-05-06:
Maybe the DNAT rule will cover that, do you already tried that?

Adorfer, 2020-05-12:
such a DMZ-"bridge" from the VPS would help a lot. 
(for people who like to run closed appliances "requiring public ip" on consumer broadband without fixed IP or plenty of fibre bandwitdth but just a single IPv4, but need to host multiple appliances)

DavideCruccolini, 2020-05-12:
I'm not very expert, could you tell me step by step how should I do the configuration on the vps side and on the openmptcp side to be able to use multiple public ip's with a single aggregator?

JorgeMoort89, 2020-06-03:
Do you guys have done that? I just can't figure out how to do that

Adorfer, 2020-06-04:
1) no
2) this is why this feature request was issued. otherwise it would be a Pull-Request either for code or docs.

ascomputer, 2020-06-05:
I think it is very important to have the ability to use multiple server-side ip for many reasons ...
one reason is that if openmptcprouter is used for many le clients
dns queries made by a single public ip is likely to have a block and could be seen as spam.
So openmptcorouter which is currently a fantastic bandwidth aggregation system can only be used for home and non-professional use. I hope you can use more than one ip on a dedicated server.

ascomputer, 2020-06-05:
I think it is very important to have the ability to use multiple server-side ip for many reasons ...
one reason is that if openmptcprouter is used for many le clients
dns queries made by a single public ip is likely to have a block and could be seen as spam.
So openmptcorouter which is currently a fantastic bandwidth aggregation system can only be used for home and non-professional use. I hope you can use 2 or more ip on a dedicated server.

Adorfer, 2020-06-05:
> I think it is very important to have the ability to use multiple server-side ip for many reasons ...

I agree, but the reason you are stating is not related/not significant justice for that feature. (In other words: There are far easier ways to navigate around that scenario. But that would a derail of this issue here)

ascomputer, 2020-06-05:
I agree that you and I reply that openmptcprouter is excellent and I personally thank the developers ... I noticed that in shadowsocks there is the possibility of inserting more ip servers ... could it be the solution?
[image]

ascomputer, 2020-06-06:
> La condivisione di un IP per tutti i computer domestici non è un grosso problema, è la stessa quando si dispone di un provider.
> Anche se avere più IP quando si ospita un server a casa può essere utile.
> Le query DNS fatte da un singolo IP pubblico non vengono spesso bloccate (alcuni paesi utilizzano solo pochi IP come uscita) e per lo spam non capisco davvero cosa può essere bloccato qui.
> OpenMPTCProuter è fatto per uso domestico / client di piccole dimensioni. Non è realizzato come router per un datacenter.
> Ad esempio, è possibile ottenere un IP pubblico su LAN con un tunnel GRE, ma se shadowsocks è abilitato si avrà IP VPS come uscita perché è un proxy.

I understand that it is for home use and not for data centers but there are also small wifi providers that with this solution could help them grow.

Ysurac, 2020-06-06:
You should have beginning with this usage, that I can understand.
Now I need to find a cheap VPS/server with cheap IPv4 available to make some tests in France ;)

JorgeMoort89, 2020-06-06:
Hi Ysurac,
if you need some VPS with multiple public IP I can give you them without any problem
I've sent some emails about that some month ago
Let me know

ascomputer, 2020-06-06:
I can also give you a dedicated server with 2 public ip's ... but I'm not in France ... let me know.

Ysurac, 2020-06-08:
@JorgeMoort89 if you have one available that I can use for tests about that yes, this can be interesting.

gnosoz, 2020-06-08:
i have one too and is locate in Paris so is close to your home  ;)

Adorfer, 2020-06-09:
i can offer one at OVH with 2-4 IPv4 addresses. Just let me know the specs (RAM/HDD) needed.

JorgeMoort89, 2020-06-09:
> @JorgeMoort89 if you have one available that I can use for tests about that yes, this can be interesting.

Can I send you an email about that?

Ysurac, 2020-06-09:
Yes, I'm open to any offer. I only need a small server (any x86 with 1024Mo ram and Debian 10) for a month with more than 1 IPv4.

Adorfer, 2020-06-10:
you have mail (vm with 3 IPv4)

Ysurac, 2020-06-10:
@Adorfer Thanks. I will make tests with it.
I can now test when a VPS have 3 interfaces with a different IP on each, I think some have only one interface with all IPs configured on it ? or bridge ?

Adorfer, 2020-06-10:
No problem, i can put the 3 ips on one IF. standy by.

Ysurac, 2020-06-10:
No, it's ok for now, I will try with that first :)

ascomputer, 2020-06-12:
many vps or dedicated servers have an interface with multiple ip's

Ysurac, 2020-06-13:
Thanks to the 2 server with multiples IPs I get, with 2 configurations (it's better for tests).
I will add several things:
* It will be possible to forward a port from any public IP to any LAN IP via firewall interface
* I will add GRE tunnel and new shadowsocks-libev server on VPS part (I will patch shadowsocks manager for that) and ss-redir on router part to allow a client to exit via the public IP it want.

If you see something I missing or another solution, you are free to tell me.

ascomputer, 2020-06-13:
I think these changes are fine and in my opinion after these changes openmptcprouter will be the top ... both for home and professional use ...

DavideCruccolini, 2020-06-13:
Thanks to everyone for collaborating on my request!
If it can be useful I have the possibility to make vps available for tests.
I am waiting for the changes to be implemented.

ascomputer, 2020-07-08:
hello everyone, is there any news for using 2 or more static ip on the server side?

Ysurac, 2020-07-08:
It's available in latest 0.55 beta.

ascomputer, 2020-07-08:
thanks for the reply, where can i download the 0.55 beta version? I do some testing!

Ysurac, 2020-07-08:
The issue is pinned, it's https://github.com/Ysurac/openmptcprouter/issues/959

ascomputer, 2020-07-08:
I installed version 0.55 beta 15 and it works well, however I don't understand how to tell the router that the computer 1925.168.100.193 must go out on the second ip of the vps instead of the first

Ysurac, 2020-07-08:
The rule need to be configured in Services->Shadowsocks-libev and rule tab.

ascomputer, 2020-07-08:
if I try to insert a new rule in redir-rules
gives me this mistake

Save error
Errore durante il salvataggio del form:

RPC call to uci/delete failed with ubus code 4: Resource not found
  at ClassConstructor.handleCallReply (http://192.168.100.1/luci-static/resources/rpc.js?v=git-20.182.56619-8f94331:16:3)

Ysurac, 2020-07-08:
I will fix that, but you shouldn't have to add a new rule, but only to modify rules that are created for available IPs. If you have only one rule, make sure all IPs are configured on the VPS, that you updated VPS script and rebooted the VPS.

ascomputer, 2020-07-08:
the my ifconfig on vps 

dsvpn0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 9000
        inet 10.255.251.1  netmask 255.255.255.255  destination 10.255.251.2
        inet6 64:ff9b::aff:fb01  prefixlen 96  scopeid 0x0<global>
        inet6 fe80::4168:c76b:2dfe:84ea  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6  bytes 480 (480.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 83.xxx.xxx.185  netmask 255.255.254.0  broadcast 83.xxx.xxx.255
        inet6 2a02:29e0:2:6:1:1:2568:cb81  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::216:3eff:fe53:9cf4  prefixlen 64  scopeid 0x20<link>
        ether 00:16:3e:53:9c:f4  txqueuelen 1000  (Ethernet)
        RX packets 311935  bytes 36651114 (34.9 MiB)
        RX errors 0  dropped 813  overruns 0  frame 0
        TX packets 52977  bytes 23769463 (22.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0:0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 83.xxx.xxx.187  netmask 255.255.254.0  broadcast 83.xxx.xxx.255
        ether 00:16:3e:53:9c:f4  txqueuelen 1000  (Ethernet)

gt-tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.255.255.1  netmask 255.255.255.252  destination 10.255.255.1
        inet6 fe80::f7f0:d6dd:df67:65b5  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

gt-udp-tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.255.254.1  netmask 255.255.255.252  destination 10.255.254.1
        inet6 fe80::4f95:c43:67f7:a3a4  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3854  bytes 673324 (657.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3854  bytes 673324 (657.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mlvpn0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1444
        inet 10.255.253.1  netmask 255.255.255.0  destination 10.255.253.1
        inet6 fe80::9c22:3be9:a95:eaf9  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11  bytes 696 (696.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

omr-6in4-user0: flags=209<UP,POINTOPOINT,RUNNING,NOARP>  mtu 1480
        inet6 fe80::aff:fc01  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::a00:1  prefixlen 126  scopeid 0x20<link>
        sit  txqueuelen 1000  (IPv6-in-IPv4)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.255.252.1  netmask 255.255.255.255  destination 10.255.252.2
        inet6 fe80::2a69:ba42:b5a7:c803  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 100  (UNSPEC)
        RX packets 14351  bytes 2600562 (2.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 19229  bytes 14560862 (13.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun1: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.255.250.1  netmask 255.255.255.255  destination 10.255.250.2
        inet6 fe80::2df2:724b:b52b:a0f5  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 100  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4  bytes 288 (288.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

as you can see eth0 holds 2 public ip addresses

as you can see oth0 holds 2 public ip addresses

I updated the vps with the script of version 0.55 beta 15 and I have
restarted the vps.
however if I check with which ip of the vps I am going out I always find the first the of the vps.

Ysurac, 2020-07-08:
You can try to validate the wizard again. You should get GRE tunnel in Network->interfaces, new ss-redir and rules in Services->Shadowsocks.

ascomputer, 2020-07-10:
I did everything you recommended but unfortunately no positive result ... I try to install the vps and router again ...

Ysurac, 2020-07-10:
Do you have a "gre_tunnels" entry in ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` on the VPS ?

ascomputer, 2020-07-10:
this is the content of the admin-config.json file on vps ... i just obfuscated ip and password to publish ;-)

 "port": 65500,
    "users": [
        {
            "admin": {
                "username": "admin",
                "user_password": "151226A82DCE2C315B83B586B8B8F0AAF60AAC3567444D0EA778AD45F23AXXXX",
                "permissions": "admin",
                "disabled": false
            },
            "openmptcprouter": {
                "username": "openmptcprouter",
                "user_password": "642431F5EDC93E5D1B34333E2C99BC3E18BC0B5BD9311F6C49F8201AF14EXXXX",
                "shadowsocks_port": 65101,
                "disabled": false,
                "vpn": "openvpn",
                "lanips": [
                    "192.168.100.1/255.255.255.0"
                ],
                "vpnremoteip": "10.255.252.10",
                "vpnlocalip": "10.255.252.1",
                "ula": "fd72:f0b2:5d9e::/48"
            }
        }
    ],
    "lastchange": 1594385674.198613,
    "allips": [
        "83.136.106.185",
        "83.136.106.187"
    ],
    "ipv4": "83.xxx.xxx.185",
    "hostname": "83.xxx.xxx.185"
}

Ysurac, 2020-07-10:
Are you sure that you use snapshot VPS script ? What do you have in ```/etc/motd``` ? and do you have any error in ```/var/log/daemon.log```.

ascomputer, 2020-07-10:
in the file motd:
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
< OpenMPTCProuter VPS 0.1016-test >

there appears to be no errors in the daemon.log file

ascomputer, 2020-07-15:
I installed the vps again and I put the Beta16 (2020/07/12)
Images: https://download.openmptcprouter.com/release/v0.55beta16/
Latest VPS script required: wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh
in interfaces there is no gre tunnel and in shadowsocks-libev ----> Redir Rules  cannot create a rule to go out with the second ip of the vps ... I await new updates.

Ysurac, 2020-07-15:
If there is no error in daemon.log file I don't know what is the problem. So while I can't reproduce the issue I can't fix it.

Ysurac, 2020-07-15:
I made a little change, can you try to launch again VPS script and reboot ?

ascomputer, 2020-07-16:
I ran wget -O - http: // www again. openmptcprouter.com/server-test/debian10-x86_64.sh | sh on vps as you told me and at the end of the installation it says No working domain detected, not able to generate certificate for v2ray.
You can set VPS_DOMAIN to a working domain if you want a certificate.
Restarting shorewall ...
I restarted the vps this is the content of
{
    "port": 65500,
    "users": [
        {
            "admin": {
                "username": "admin",
                "user_password": "B049238EB6F95F9C444943E8415E40BDEF99DE31AE987A00586C8DC82AE9D20D",
                "permissions": "admin",
                "disabled": false
            },
            "openmptcprouter": {
                "username": "openmptcprouter",
                "user_password": "BE3F7AC089F54280A03D19AE1AE862E1B46FC68E82D0C8387FBDBF2CEE9BD741",
                "shadowsocks_port": 65101,
                "disabled": false,
                "vpn": "openvpn",
                "lanips": [
                    "192.168.100.1/255.255.255.0"
                ]
                "vpnremoteip": "10.255.252.6",
                "vpnlocalip": "10.255.252.1",
                "ula": "fd72: f0b2: 5d9e :: / 48"
            }
        }
    ]
    "allips": [
        "83.xxx.xxx.185"
        "83.xxx.xxx.187"
    ]
    "ipv4": "83.xxx.xxx.185",
    "hostname": "83.xxx.xxx185",
    "lastchange": 1594876764.8529375
}

now I'll post also the content of daemon.log maybe I don't see where the error lies

ascomputer, 2020-07-16:
oops! i found some gateway errors i attach files

[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4929532/daemon.log)

Ysurac, 2020-07-16:
Gateway errors are not related (It's related to OpenVPN, I fixed the error in VPS script).
There is no error about omr-admin. 
I've update again VPS script and added some debug info in the omr-admin python script, can you restart the VPS install script and check again in /var/log/daemon.log if there is no new error ?

ascomputer, 2020-07-16:
ok then I install the vps script and update you

ascomputer, 2020-07-16:
[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4930221/daemon.log)
installed vps script I posted the file

Ysurac, 2020-07-16:
Should be better now, I fixed the script. gre_tunnels will appear in /etc/openmptcprouter-vps-admin/omr-admin-config.json after a VPS script update.

ascomputer, 2020-07-16:
In the file /etc/openmptcprouter-vps-admin/omr-admin-config.json
 ... but there is vpn tunnel down

{
    "port": 65500,
    "users": [
        {
            "admin": {
                "username": "admin",
                "user_password": "B049238EB6F95F9C444943E8415E40BDEF99DE31AE987A00586C8DC82AE9D20D",
                "permissions": "admin",
                "disabled": false
            },
            "openmptcprouter": {
                "username": "openmptcprouter",
                "user_password": "BE3F7AC089F54280A03D19AE1AE862E1B46FC68E82D0C8387FBDBF2CEE9BD741",
                "shadowsocks_port": 65101,
                "disabled": false,
                "vpn": "openvpn",
                "lanips": [
                    "192.168.100.1/255.255.255.0"
                ],
                "vpnremoteip": "10.255.252.6",
                "vpnlocalip": "10.255.252.1",
                "ula": "fd72:f0b2:5d9e::/48",
                "gre_tunnels": {
                    "gre-user0-ip0": {
                        "shadowsocks_port": "65102",
                        "local_ip": "10.255.249.1",
                        "remote_ip": "10.255.249.2",
                        "public_ip": "83.136.106.185"
                    },
                    "gre-user0-ip1": {
                        "shadowsocks_port": "65103",
                        "local_ip": "10.255.249.5",
                        "remote_ip": "10.255.249.6",
                        "public_ip": "83.136.106.187"
                    }
                }
            }
        }
    ],
    "allips": [
        "83.xxx.xxx.185",
        "83.xxx.xxx.187"
    ],
    "ipv4": "83.xxx.xxx.185",
    "hostname": "83.xxx.xxx.185",
    "lastchange": 1594891913.3248146
}

[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4930629/daemon.log)
[image]

Ysurac, 2020-07-16:
You may have to validate wizard again. And why do you use OpenVPN as VPN ? It's the slowest available VPN.

ascomputer, 2020-07-16:
I reinstalled all the vps and the vps script the situation is this
[image]
what do you advise me to do?

Ysurac, 2020-07-16:
You updated the key in the wizard ?

ascomputer, 2020-07-16:
Yes

Ysurac, 2020-07-16:
Try to reboot the router and if no change validate the wizard again.

ascomputer, 2020-07-16:
I restarted the router and checked all the keys ... same problem

[59 more replies]

closed 2021-04-03 by github-actions[bot]