Ysurac/openmptcprouter #999: Ring expansion failed

kevin39, 2020-05-04:
Hi,
I've lot of these error in my system.log what is this about ?
(rpi 4)

`Mon May  4 08:46:56 2020 kern.err kernel: [78078.029578] hub 1-1:1.0: activate --> -12
Mon May  4 08:46:57 2020 kern.err kernel: [78079.244269] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:46:57 2020 kern.err kernel: [78079.249599] hub 1-1:1.0: activate --> -12
Mon May  4 08:46:59 2020 kern.err kernel: [78080.464233] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:46:59 2020 kern.err kernel: [78080.469802] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:00 2020 kern.err kernel: [78081.714228] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:00 2020 kern.err kernel: [78081.719573] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:01 2020 kern.err kernel: [78082.974204] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:01 2020 kern.err kernel: [78082.979525] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:02 2020 kern.err kernel: [78084.224265] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:02 2020 kern.err kernel: [78084.229587] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:04 2020 kern.err kernel: [78085.474258] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:04 2020 kern.err kernel: [78085.479580] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:05 2020 kern.err kernel: [78086.694310] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:05 2020 kern.err kernel: [78086.699641] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:06 2020 kern.err kernel: [78087.924278] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:06 2020 kern.err kernel: [78087.929600] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:07 2020 kern.err kernel: [78089.144249] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:07 2020 kern.err kernel: [78089.149570] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:09 2020 kern.err kernel: [78090.394248] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:09 2020 kern.err kernel: [78090.399633] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:10 2020 kern.err kernel: [78091.644242] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:10 2020 kern.err kernel: [78091.649563] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:11 2020 kern.err kernel: [78092.894302] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:11 2020 kern.err kernel: [78092.899623] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:12 2020 kern.err kernel: [78094.144318] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:12 2020 kern.err kernel: [78094.149641] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:14 2020 kern.err kernel: [78095.394373] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:14 2020 kern.err kernel: [78095.399822] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:15 2020 kern.err kernel: [78096.654292] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:15 2020 kern.err kernel: [78096.659614] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:16 2020 kern.err kernel: [78097.904284] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:16 2020 kern.err kernel: [78097.909605] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:17 2020 kern.err kernel: [78099.154348] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:17 2020 kern.err kernel: [78099.159669] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:19 2020 kern.err kernel: [78100.404342] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:19 2020 kern.err kernel: [78100.409664] hub 1-1:1.0: activate --> -12
Mon May  4 08:47:20 2020 kern.err kernel: [78101.624319] xhci_hcd 0000:01:00.0: Ring expansion failed
Mon May  4 08:47:20 2020 kern.err kernel: [78101.629641] hub 1-1:1.0: activate --> -12`

Ysurac, 2020-05-04:
Please use issue template...
It's a know RPI issue, I hope this will be solved in next release with kernel update.

kevin39, 2020-05-04:
Hi,
Thanks and sorry for the template.
I thought it was not suitable for this problem ...

closed 2020-05-04 by kevin39