Ysurac/openmptcprouter #1007: daemon.err /usr/bin/ss-local[20476]: getpeername: Socket not connected

kyouran38, 2020-05-05:
Hi since today I have a lot of these messages on my log file : 

daemon.err /usr/bin/ss-local[20476]: getpeername: Socket not connected
daemon.err /usr/bin/ss-redir[20480]: remote recv: Operation timed out

I don't know if it's related or not but I've got micro disconnection when I am using Internet on my computer.

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPI3/x86_64

Thanks

regards

Ysurac, 2020-05-05:
What is the VPS provider ?

kyouran38, 2020-05-05:
Milkywan

Ysurac, 2020-05-05:
This can be related to big problems with fibers near DC2 data center an make some problem with the VPS.
Else this can be related to a connection that is not down but many packets lost.

kyouran38, 2020-05-05:
Thank you, so why should I do ?

 

 
 

Sent: Tuesday, May 05, 2020 at 4:05 PM

Ysurac, 2020-05-05:
You can check on the VPS in /var/log/daemon.log the errors you can get.
You can also try connection one by one to check if the problem come from a connection.

kyouran38, 2020-05-06:
I don't if it's normal or not but I don't have any daemon.log in my /var/log directory

 

 
 

Sent: Tuesday, May 05, 2020 at 8:38 PM

Ysurac, 2020-05-06:
If you are using Ubuntu it's normal, else not.

kyouran38, 2020-05-06:
Ah ok, is there another way to check on the server where it can come from ? 

 

Regards

 
 

Sent: Wednesday, May 06, 2020 at 5:06 PM

Ysurac, 2020-05-06:
It's somewhere in /var/log on the the VPS.

kyouran38, 2020-05-06:
On googling my log, I have found this page : https://www.gitmemory.com/issue/Ysurac/openmptcprouter/414/478431437

 

Do you think it can help solving my pb ? 

 

Regards

 
 

Sent: Wednesday, May 06, 2020 at 5:29 PM

kyouran38, 2020-05-06:
I don't know if it's link or not but on my local server when I use the command ss-redir I have got this message 2020-05-06 18:17:53 ERROR: Invalid config path

 
 

Sent: Wednesday, May 06, 2020 at 5:29 PM

closed 2020-09-15 by github-actions[bot]