Ysurac/openmptcprouter #1008: Lots of problem since I try to install luci-wrtbwmon

jeandelamarquise, 2020-05-06:
## Expected Behavior
I wanted to install luci-wrtbwmon with wrtbwmon dependancies

## Current Behavior
luci-wrtbwmon not working
I try to uninstall because i didn't have access to the top menu since the installation

## Possible Solution

## Steps to Reproduce the Problem

  1. install the wrtbwmon and luci-wrtbwmon
  2. rm /tmp/luci-indexcache
  3. no more menu...

## Context (Environment)
I try to upgrade all LuCi components except the luci-i18n-****
Since then, i have the top menu but some problems come in addition like the page with DHCP/DNS with 
_RPCError
RPC call to luci-rpc/getDUIDHints failed with error -32002: Access denied
  at ClassConstructor.handleCallReply (http://192.168.10.1/luci-static/resources/rpc.js?v=git-20.126.39721-9c4f345:12:3)_

The status page show lots of ? :
_Nom d'hôte	?
Modèle	?
Architecture	?
Version du micrologiciel	
Version du noyau	?
Date	?
Temps de service	?
Charge moyenne	?_

## Specifications

  - OpenMPTCProuter version: v0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: RP4

Ysurac, 2020-05-07:
As OpenWrt snapshot package tree is used, you shouldn't update from it, it's not stable.

jeandelamarquise, 2020-05-07:
How can I do to reinstall openmptcprouter properly ?
I would like to have real usage per device monitoring.

jeandelamarquise, 2020-05-08:
Hi all,

I format my sd card and I flash the last version openmptcprouter v0.54.7.
I succeed in Disney+ configuration (netflix and prime video too but i have a backup)

So, my need is to have something to monitor the internet usage live per device (my computer, my phone, etc...) Is there something to do that ?

Thanks !

closed 2020-09-15 by github-actions[bot]