Ysurac/openmptcprouter #1010: Build ZBT-WE3526

JorgeMoort89, 2020-05-07:
Hi Ysurac,
I can't build OMR for ZBT-WE3526, are MIPS CPU compatible?

Thanks

Ysurac, 2020-05-07:
No idea. I don't have any MIPS platform and I don't know any that have a fast enough CPU and enough flash available.

JorgeMoort89, 2020-05-07:
Have I to select only target system, subtarget and profile on menuconfig?
Is steadily failing the build

Ysurac, 2020-05-07:
Yes, some changes may also be needed for lan/wan name of ports.
This may be a package that fail, or something else. But it's not a supported platform and this seems to be a too slow/small platform for me to work in.

JorgeMoort89, 2020-05-21:
I'm successfully building the image, but is too big to fit the 32MB flash on the router (about 35MB). How can I build a lighter image? For example I don't need OpenVPN or MLVPN

Ysurac, 2020-05-21:
You can remove them in make menuconfig, you need to remove openmptcprouter-full first.
I will make a "openmptcprouter-light" when I will have some time for that ;)

JorgeMoort89, 2020-05-22:
Just tested, but removing openmptcprouter-full remove all the other packages, which package are essential to build a working image?
Thanks a lot for the help :)

closed 2020-09-24 by github-actions[bot]