Ysurac/openmptcprouter #1011: OpenMPTCProuter from Remote

ascomputer, 2020-05-07:
hello everyone, I wanted to ask: is it possible to enter the router remotely? example ip vps xxx.xxx.xxx.xxx:8000 and point to the router to enter its webgui.
Thanks to everyone and congratulations to those who developed this magnificent system of aggregating various connections.

gnosoz, 2020-05-08:
1) ssh to router
         - change port
         - key only access
         - strengthen firewall vps and open port manually only when needed

2) dnat vpn to router  destro port 8000 forward to 80
3) service uhttpd start
4) browser => public ip:80
5) service uhttpd stop

that's how I would do it....hope it helps

>

ascomputer, 2020-05-08:
> 1) ssh to router - change port - key only access - strengthen firewall vps and open port manually only when needed 2) dnat vpn to router destro port 8000 forward to 80 3) service uhttpd start 4) browser => public ip:80 5) service uhttpd stop that's how I would do it....hope it helps
> […](#)
thanks for replying ... I don't know how to do it :-(

ascomputer, 2020-05-08:
thanks for replying
I entered the VPN and did this:
1) sudo iptables -A INPUT -p tcp --dport 8000 -j ACCEPT
it's right?

gnosoz, 2020-05-09:
it will require some work to detail all step, let me see what I can do.
Do you have a static address to access the vps? Otherwise it might prove daunting to explain & enable make all this process work.
This would be a great addition for the project documentation.

ascomputer, 2020-05-14:
solved thanks

closed 2020-09-16 by github-actions[bot]

Adorfer, 2020-09-16:
> solved thanks

so why don't you close it? open sub-topic?

FelixWoRe, 2024-06-13:
@ascomputer is there a way you can explain it to me now. I want to also have remote access to the router.
Thx!