Ysurac/openmptcprouter #1012: SRT streaming with bond aggregation

tvcidadeverde, 2020-05-07:
I prepared a Linux machine running or openmptcprouter and adding 3 4G modems.

My objective:
Send a video stream using the SRT protocol.
The SRT server was already installed on my VPS, but I can't access this service either through the internal or external network.

I don't have access to the SRT service or any other on my VPS.

Does anyone have any test scenarios so I can know if my settings are in fact correct ??

Below is my Network overview

[image]

Ysurac, 2020-05-08:
You have opened needed ports on the firewall VPS via ```/etc/shorewall/``` ?

SlipstreamCoding, 2020-05-08:
This is exactly my use case, and it works.
As a test scenario, you need two computers one "caller" (push) on OMR, one "listener" (pull) on your regular Internet. 
Try that way around locally first to troubleshoot, then over the Internet. Punch a hole in your local router for that. 
Then swap rôles (caller/listener). 
Then do the same on OMR/VPS, don't forget to port forward in OMR (that should set the ports for the VPS too) 

@Ysurac SRT is becoming increasingly popular as a means to replace rtmp, rtsp... For streaming video.
https://github.com/Haivision/srt
SRT is basically UDP encapsulated video on a bi-directional communication channel, with FEC (forward error correction) and other techniques that take packet loss and jitter into account. Very good for congested networks. 

If you don't mind this could become a thread dedicated to the protocol? 

I have a similar test setup using 4 USB NCM modems, VPS on Ubuntu hosted by Ikoula. Test stream was SRT@4mb, sometimes caller sometimes listener. 

In my tests I tried Glorytun UDP, TCP, with/without Shadowsocks, with/without vpn layers.
SRT caller -> OMR -> VPS - > SRT listener

Please correct me if my assumptions are wrong. 
For this use case, I quickly determined any kind of obfuscation isn't useful. Shadowsocks only works for TCP, so I disabled it. VPN layers I couldn't quite get to work.

Glorytun UDP was my first theoretical choice. The receiving SRT codec would report a stable 25-30% packet loss.
Glorytun TCP yielded 5-10% packet loss.
OMR bypass 0% if using fiber instead of 4G.
Video playback was flawless in all cases. 

My understanding is OMR should absorb any packet loss, but so should the SRT codec. Maybe there is some sort of double buffering going on, that makes the codec resend data again and again?
Any idea which settings could limit this kind of overhead?

Thanks :-)

vinceducat, 2020-05-08:
@SlipstreamCoding  interresting in this subject is this protocol used by discord.app for video and talk ??

tvcidadeverde, 2020-05-08:
I installed NGINX running internally in my VPS on port 80 to facilitate the tests.

What are the necessary settings in / etc / shorewall / to be able to access the nginx home page ??

On my client host connected to my OpenMPTCProuter, how should I access the service that is running on my VPS ??

I added a rule to the / etc / shorewall / rules file, but without success.

ACCEPT net $ FW tcp 80

SlipstreamCoding, 2020-05-08:
Discord uses WebRTC, like most of all conferencing apps today. It is a standard focused on real-time chat for the masses. Not so good for OMR as latency is increased. 
SRT is mainly used in the content creation industry, it is a protocol for remote vidéo contribution, a simple point to point connexion. Quality is higher at the expense of latency, but the codec let's you tune that.

vinceducat, 2020-05-08:
@SlipstreamCoding  i you should be expert in this type of techs, i have problem with omr and discord. you seems to aware of those protocol, could you help me to solve my problem.. perhaps in private..  ?

SlipstreamCoding, 2020-05-09:
@vinceducat You should open a specific ticket for Discord.

@tvcidadeverde I'm not an expert, but you might want to check out how to route traffic to services internal to your server. OMR-VPS by default routes everything to the internet. You should find other use cases here.
My suggestion would be to have a second server and leave the VPS dedicated to OMR with minimal configuration steps.

closed 2020-05-17 by tvcidadeverde

Llorx, 2020-11-08:
@SlipstreamCoding how are your tests going? What is your best config for this?

video2go, 2021-09-21:
Hey. I am trying to use openMPTCProuter on site connecting to VPS. What are the most resilient settings? Anyone using this?
I use several LTE and one 5G routers to bond. I have trouble getting normal LANs into bond. It works over LTE/5G, but LAN gets dropped when using.

josebroadcast, 2021-12-25:
@SlipstreamCoding I am going to implement a router for srt video transmission, I see that you have knowledge and have come up with the best configuration, is it possible to get in contact with you?