Ysurac/openmptcprouter #1013: Keeping line aggregation with OMR BYPASS

numerobis08, 2020-05-09:
Hello
I used OMR BYPASS in order to avoid to be blacklisted by Netflix and forced my tv to use my WAN2. By doing that I’m loosing benefit of the aggregation. Is there a way to show to Netflix the IP of my WAN2 ( and not the ip if the VPS which is blacklisted ) while keeping aggregation ? 
Thank you in advance

Ysurac, 2020-05-09:
It's not possible.
The only way to keep aggregation with Netflix is to have a VPS provider that is not blacklisted.

kevin39, 2020-05-09:
Not sure but... dispite latency problems, technically this can be possible by connecting the VPS to an external proxy, no ? Like NordVPN or something like that but anyway, this may be quite difficult to have a compatible setup with OMR.

And there will be another problem : Found a VPN provider not blacklisted by Netflix.

Ysurac, 2020-05-09:
Yes, an external VPN can also be used. But it's better for speed to have a VPS not blacklisted by Netflix than an external VPN.

numerobis08, 2020-05-09:
Hi both 
Thanks for the feedback. my VPS is currently hosted by Firstheberg , do you know a provider which is not blacklisted ?

kevin39, 2020-05-09:
FirstHeberg was not blacklisted when I used it few weeks ago.

numerobis08, 2020-05-09:
Also can we force to go through NordVPN only for Netflix ? ( I do have a Nordvpn subscription with dedicated IP)
Thanks

numerobis08, 2020-05-09:
> FirstHeberg was not blacklisted when I used it few weeks ago.

Yes this is why I choose it but the IP that they gave me doesn’t work with Netflix. I’ll looking for milkywan after reading positive comments

Ysurac, 2020-05-09:
Milkywan is not blacklisted.

For NordVPN, by configuring it in OpenVPN section of OpenMPTCProuter and using omr-bypass to redirect netflix to NordVPN OpenVPN interface this should be possible.

numerobis08, 2020-05-09:
> Milkywan is not blacklisted.
> 
> For NordVPN, by configuring it in OpenVPN section of OpenMPTCProuter and using omr-bypass to redirect netflix to NordVPN OpenVPN interface this should be possible.

Thank you both for the quick feedback

numerobis08, 2020-05-10:
One last question. 
I tried to upload the .ovpn file from NordVPN and tried desperately but the VPN doesn't start. 
did i missed something ? 
[image]

Ysurac, 2020-05-10:
Check that it doesn't use "tun0" as interface and check State->System log.

numerobis08, 2020-05-10:
In which section I can check that tun0 is not used ? ( sorry just started 1 week ago with openmptcp)

Ysurac, 2020-05-10:
In VPN->OpenVPN, edit the instance, "Switch to advanced configuration", "Networking" and "dev".

numerobis08, 2020-05-26:
Hello
I cannot find the “switch to advanced configuration” in VPN / OPENVPN is there something to activate ?

numerobis08, 2020-05-26:
By the way many thanks for the support

Ysurac, 2020-05-26:
This seems difficult to give more info. When you edit an instance of OpenVPN it's the first link after the title.

closed 2020-09-28 by github-actions[bot]