Ysurac/openmptcprouter #1016: VPS is freezing every 10 hours

DeeJayMX, 2020-05-10:
## Current Behavior
VPS is freezing every 10 hours

## Possible Solution
a restart of the VPS put everything ON, So I have to restart it every 10 hours.

## Steps to Reproduce the Problem
Install latest VPS and router on Pi4 and just wait couple of hours

## Context (Environment)
No more internet IN and OUT, the router goes in direct output mode

## Specifications

-OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
-OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
-OpenMPTCProuter platform: RPi4B 4GB>

Ysurac, 2020-05-10:
What is the VPS provider and what is the error you get on the VPS ?

DeeJayMX, 2020-05-10:
OVH VPS 2018 SSD 1
I'm trying to reconnect to it.
No issue on the KVM of the VPS, the machine is running...
Which log do you want ?

Ysurac, 2020-05-10:
The result of```dmesg```
via KVM do you have internet access ?
What do you have in ```iptables-save``` ?
Do you have fail2ban running or something like that ?

DeeJayMX, 2020-05-10:
nothing more than the VPS script and IPV6 set before installing script
the daemon log keep looping that every two second forever:

May 10 15:57:57 vps813080 systemd[1]: Reloaded Shorewall IPv4 firewall.
May 10 15:57:57 vps813080 systemd[1]: Reloading Shorewall IPv4 firewall.
May 10 15:57:57 vps813080 shorewall[7340]: Compiling using Shorewall 5.2.3.2...
May 10 15:57:57 vps813080 shorewall[7340]: Processing /etc/shorewall/params ...
May 10 15:57:57 vps813080 shorewall[7340]: Processing /etc/shorewall/shorewall.conf...
May 10 15:57:57 vps813080 shorewall[7340]: Loading Modules...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/zones...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/interfaces...
May 10 15:57:57 vps813080 shorewall[7340]: Determining Hosts in Zones...
May 10 15:57:57 vps813080 shorewall[7340]: Locating Action Files...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/policy...
May 10 15:57:57 vps813080 shorewall[7340]: Adding Anti-smurf Rules
May 10 15:57:57 vps813080 shorewall[7340]: Adding rules for DHCP
May 10 15:57:57 vps813080 shorewall[7340]: Compiling TCP Flags filtering...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling Kernel Route Filtering...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling Martian Logging...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling Accept Source Routing...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/tcinterfaces...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/snat...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling MAC Filtration -- Phase 1...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/rules...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/conntrack...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling MAC Filtration -- Phase 2...
May 10 15:57:57 vps813080 shorewall[7340]: Applying Policies...
May 10 15:57:57 vps813080 shorewall[7340]: Generating Rule Matrix...
May 10 15:57:57 vps813080 shorewall[7340]: Creating iptables-restore input...
May 10 15:57:57 vps813080 shorewall[7340]: Compiling /etc/shorewall/stoppedrules...
May 10 15:57:57 vps813080 shorewall[7340]: Shorewall configuration compiled to /var/lib/shorewall/.reload
May 10 15:57:57 vps813080 shorewall[7340]: Reloading Shorewall....
May 10 15:57:57 vps813080 shorewall[7340]: Initializing...
May 10 15:57:57 vps813080 shorewall[7340]: Setting up Route Filtering...
May 10 15:57:57 vps813080 shorewall[7340]: Setting up Martian Logging...
May 10 15:57:57 vps813080 shorewall[7340]: Setting up Accept Source Routing...
May 10 15:57:57 vps813080 shorewall[7340]: Setting up Traffic Control...
May 10 15:57:57 vps813080 shorewall[7340]:    WARNING: Device gt-tun0 is not in the UP state -- traffic-shaping configuration skipped
May 10 15:57:57 vps813080 shorewall[7340]: Preparing iptables-restore input...
May 10 15:57:57 vps813080 shorewall[7340]: Running /sbin/iptables-restore --wait 60...
May 10 15:57:57 vps813080 shorewall[7340]: IPv4 Forwarding Enabled
May 10 15:57:57 vps813080 shorewall[7340]: done.
May 10 15:57:58 vps813080 systemd[1]: Reloaded Shorewall IPv4 firewall.
May 10 15:57:58 vps813080 systemd[1]: Reloading Shorewall IPv4 firewall.

DeeJayMX, 2020-05-10:
here the result of dmesg:

`[    0.000000] Linux version 4.19.104-mptcp (root@vps-31921.fhnet.fr) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)) #1 SMP Mon Feb 17 11:08:15 UTC 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.19.104-mptcp root=UUID=77e3f255-2ef2-47bc-ad89-7cdbd65f5fbc ro biosdevname=0 net.ifnames=0 console=tty0 console=ttyS0,115200 earlyprintk=ttyS0,115200 consoleblank=0 systemd.show_status=true
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007cfdbfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007cfdc000-0x000000007cffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] bootconsole [earlyser0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: OpenStack Foundation OpenStack Nova, BIOS 2:1.10.2-58953eb7 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: cpu 0, msr 22b5001, primary cpu clock
[    0.000001] kvm-clock: using sched offset of 7689448925 cycles
[    0.001122] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.004482] tsc: Detected 2394.464 MHz processor
[    0.007086] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.007088] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.007093] last_pfn = 0x7cfdc max_arch_pfn = 0x400000000
[    0.008248] MTRR default type: write-back
[    0.008250] MTRR fixed ranges enabled:
[    0.008251]   00000-9FFFF write-back
[    0.008252]   A0000-BFFFF uncachable
[    0.008253]   C0000-FFFFF write-protect
[    0.008254] MTRR variable ranges enabled:
[    0.008256]   0 base 0080000000 mask FF80000000 uncachable
[    0.008257]   1 disabled
[    0.008258]   2 disabled
[    0.008258]   3 disabled
[    0.008259]   4 disabled
[    0.008259]   5 disabled
[    0.008260]   6 disabled
[    0.008260]   7 disabled
[    0.008275] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.015970] found SMP MP-table at [mem 0x000f6a60-0x000f6a6f]
[    0.017585] BRK [0x02401000, 0x02401fff] PGTABLE
[    0.017588] BRK [0x02402000, 0x02402fff] PGTABLE
[    0.017590] BRK [0x02403000, 0x02403fff] PGTABLE
[    0.017634] BRK [0x02404000, 0x02404fff] PGTABLE
[    0.017785] BRK [0x02405000, 0x02405fff] PGTABLE
[    0.017818] RAMDISK: [mem 0x35623000-0x36b08fff]
[    0.018758] ACPI: Early table checksum verification disabled
[    0.019964] ACPI: RSDP 0x00000000000F6870 000014 (v00 BOCHS )
[    0.021104] ACPI: RSDT 0x000000007CFE1504 00002C (v01 BOCHS  BXPCRSDT 00000001 BXPC 00000001)
[    0.022782] ACPI: FACP 0x000000007CFE1418 000074 (v01 BOCHS  BXPCFACP 00000001 BXPC 00000001)
[    0.024515] ACPI: DSDT 0x000000007CFE0040 0013D8 (v01 BOCHS  BXPCDSDT 00000001 BXPC 00000001)
[    0.026153] ACPI: FACS 0x000000007CFE0000 000040
[    0.026985] ACPI: APIC 0x000000007CFE148C 000078 (v01 BOCHS  BXPCAPIC 00000001 BXPC 00000001)
[    0.028329] ACPI: Local APIC address 0xfee00000
[    0.028804] No NUMA configuration found
[    0.029381] Faking a node at [mem 0x0000000000000000-0x000000007cfdbfff]
[    0.030377] NODE_DATA(0) allocated [mem 0x7cfd8000-0x7cfdbfff]
[    0.031800] Zone ranges:
[    0.032332]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.033639]   DMA32    [mem 0x0000000001000000-0x000000007cfdbfff]
[    0.034896]   Normal   empty
[    0.035503] Movable zone start for each node
[    0.036407] Early memory node ranges
[    0.037038]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[... 400 lines cut ...]
[  971.926911] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1016.934484] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1016.937643] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1066.939610] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1111.947374] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1111.950256] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1156.965230] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1156.967160] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1206.975091] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1206.977634] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1251.978007] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1251.985142] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[... cut ...]
[ 1391.995646] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1391.997577] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1442.005877] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1442.007982] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
[ 1487.018712] IPv4: martian source 67.215.246.10 from 192.168.2.199, on dev gt-udp-tun0
[ 1487.026671] IPv4: martian source 82.221.103.244 from 192.168.2.199, on dev gt-udp-tun0
`

DeeJayMX, 2020-05-10:
[DMESG Log.txt](https://github.com/Ysurac/openmptcprouter/files/4605764/DMESG.Log.txt)

DeeJayMX, 2020-05-10:
Just for the information:
Before put OMR in production, I test it on a VM on a QNAP NAS with physical interface directly attached to the OMR VM: no problem.

So I decided to pass it on a "real" router with a Pi4 and VLAN for multiple WAN connection.
from that: no more IPV6 and this strange new behavior.

DeeJayMX, 2020-05-10:
[iptables-save.txt](https://github.com/Ysurac/openmptcprouter/files/4605794/iptables-save.txt)

Ysurac, 2020-05-10:
Looks like a firewall rules that can't be applied. Do you add a rule on router part ?

DeeJayMX, 2020-05-10:
Yes, port forward to LAN server only

Ysurac, 2020-05-10:
What are the rules ?
And what do you have in /etc/shorewall/rules ?

DeeJayMX, 2020-05-10:
[rules.txt](https://github.com/Ysurac/openmptcprouter/files/4605799/rules.txt)

DeeJayMX, 2020-05-10:
[image]

Ysurac, 2020-05-10:
Please do not attach files to issue (I need to download them before read, and I don't like that).

DeeJayMX, 2020-05-10:
> Please do not attach files to issue (I need to download them before read, and I don't like that).

Sorry, the file is a bit long, I don't know how to put long text in a good readable form...

Ysurac, 2020-05-10:
Can you give me router log in State->System log and config in System->OpenMPTCProuter and all settings tab.

DeeJayMX, 2020-05-10:
OMR System log:
```
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI12, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI12
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI12, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI13, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI13
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI13, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI14
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI14, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI15, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI15
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI15, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI16, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI16
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI16, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI17, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI17
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI17, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI18, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI18
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI18, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI19, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI19
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI19, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI20, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI20
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI20, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI21, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI21
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI21, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI22, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI22
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI22, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI23, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI23
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI23, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI24, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI24
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI24, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI25, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI25
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI25, subsystem=tty
Sun May 10 13:55:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Sun May 10 14:55:08 2020 daemon.info dnsmasq-dhcp[15410]: DHCPACK(eth0) 192.168.2.115 60:a4:4c:e7:20:da HARCOURT-SERVER
Sun May 10 14:55:33 2020 daemon.info dnsmasq-dhcp[15410]: DHCPREQUEST(eth0) 192.168.2.111 52:54:00:71:02:4b
Sun May 10 14:55:33 2020 daemon.info dnsmasq-dhcp[15410]: DHCPACK(eth0) 192.168.2.111 52:54:00:71:02:4b KMSServer
Sun May 10 14:56:24 2020 daemon.info dnsmasq-dhcp[15410]: DHCPINFORM(eth0) 192.168.2.115 60:a4:4c:e7:20:da
Sun May 10 14:56:24 2020 daemon.info dnsmasq-dhcp[15410]: DHCPACK(eth0) 192.168.2.115 60:a4:4c:e7:20:da HARCOURT-SERVER
```

DeeJayMX, 2020-05-10:
openmptcprouter/debug:
```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].domain='harcourt.lan'
dhcp.@dnsmasq[0].local='/harcourt.lan/'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.dhcp_option='66,"192.168.2.10"' '42,192.168.2.10'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='60:A4:4C:E7:20:DA'
dhcp.@host[0].name='HARCOURT-SERVER'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.2.115'
dhcp.@host[1]=host
dhcp.@host[1].mac='08:00:27:5A:7D:4E'
dhcp.@host[1].name='ELASTIX-SRV'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.2.10'
dhcp.@host[2]=host
dhcp.@host[2].mac='F8:D1:11:AB:68:2E'
dhcp.@host[2].name='CanonMF4100'
dhcp.@host[2].dns='1'
dhcp.@host[2].ip='192.168.2.201'
dhcp.@host[3]=host
dhcp.@host[3].mac='3C:2A:F4:6A:51:4E'
dhcp.@host[3].name='BrotherRosa'
dhcp.@host[3].dns='1'
dhcp.@host[3].ip='192.168.2.202'
dhcp.@host[4]=host
dhcp.@host[4].mac='3C:2A:F4:AC:45:0C'
dhcp.@host[4].name='BrotherDelphine'
dhcp.@host[4].dns='1'
dhcp.@host[4].ip='192.168.2.203'
dhcp.@host[5]=host
dhcp.@host[5].mac='44:2B:03:8B:BE:20'
dhcp.@host[5].name='WifiAP'
dhcp.@host[5].dns='1'
dhcp.@host[5].ip='192.168.2.204'
dhcp.@host[6]=host
dhcp.@host[6].mac='00:40:8c:e2:01:a8'
dhcp.@host[6].name='IPCam1'
dhcp.@host[6].dns='1'
[... 1554 lines cut ...]
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
```

Ysurac, 2020-05-10:
Can you give me the content of ```/var/log/daemon.log``` ?

DeeJayMX, 2020-05-10:
daemon.log is 10 Megabytes...

DeeJayMX, 2020-05-10:
So I put it here:
[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4606151/daemon.log)
but if you want I paste it here...

DeeJayMX, 2020-05-10:
Right now the network OMR part of the VPS just stopped...

Ysurac, 2020-05-10:
How do you see it's stopped ? in OMR status page or VPS doesn't answer to ping/SSH ?

DeeJayMX, 2020-05-10:
here the end of daemon.log:
```
May 10 19:03:54 vps813080 systemd[1]: Reloaded Shorewall IPv4 firewall.
May 10 19:03:54 vps813080 systemd[1]: Reloading Shorewall IPv4 firewall.
May 10 19:03:54 vps813080 shorewall[23476]: Compiling using Shorewall 5.2.3.2...
May 10 19:03:55 vps813080 shorewall[23476]: Processing /etc/shorewall/params ...
May 10 19:03:55 vps813080 shorewall[23476]: Processing /etc/shorewall/shorewall.conf...
May 10 19:03:55 vps813080 shorewall[23476]: Loading Modules...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/zones...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/interfaces...
May 10 19:03:55 vps813080 shorewall[23476]: Determining Hosts in Zones...
May 10 19:03:55 vps813080 shorewall[23476]: Locating Action Files...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/policy...
May 10 19:03:55 vps813080 shorewall[23476]: Adding Anti-smurf Rules
May 10 19:03:55 vps813080 shorewall[23476]: Adding rules for DHCP
May 10 19:03:55 vps813080 shorewall[23476]: Compiling TCP Flags filtering...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling Kernel Route Filtering...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling Martian Logging...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling Accept Source Routing...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/tcinterfaces...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/snat...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling MAC Filtration -- Phase 1...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/rules...
May 10 19:03:55 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/conntrack...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling MAC Filtration -- Phase 2...
May 10 19:03:55 vps813080 shorewall[23476]: Applying Policies...
May 10 19:03:55 vps813080 shorewall[23476]: Generating Rule Matrix...
May 10 19:03:55 vps813080 shorewall[23476]: Creating iptables-restore input...
May 10 19:03:55 vps813080 shorewall[23476]: Compiling /etc/shorewall/stoppedrules...
May 10 19:03:55 vps813080 shorewall[23476]: Shorewall configuration compiled to /var/lib/shorewall/.reload
May 10 19:03:55 vps813080 shorewall[23476]: Reloading Shorewall....
May 10 19:03:55 vps813080 shorewall[23476]: Initializing...
May 10 19:03:55 vps813080 shorewall[23476]: Setting up Route Filtering...
May 10 19:03:55 vps813080 shorewall[23476]: Setting up Martian Logging...
May 10 19:03:55 vps813080 shorewall[23476]: Setting up Accept Source Routing...
May 10 19:03:55 vps813080 shorewall[23476]: Setting up Traffic Control...
May 10 19:03:55 vps813080 shorewall[23476]:    WARNING: Device gt-tun0 is not in the UP state -- traffic-shaping configuration skipped
May 10 19:03:55 vps813080 shorewall[23476]: Preparing iptables-restore input...
May 10 19:03:55 vps813080 shorewall[23476]: Running /sbin/iptables-restore --wait 60...
May 10 19:03:55 vps813080 shorewall[23476]: IPv4 Forwarding Enabled
May 10 19:03:55 vps813080 shorewall[23476]: done.
May 10 19:03:55 vps813080 systemd[1]: Reloaded Shorewall IPv4 firewall.
May 10 19:03:55 vps813080 systemd[1]: Reloading Shorewall IPv4 firewall.
May 10 19:03:55 vps813080 shorewall[23748]: Compiling using Shorewall 5.2.3.2...
May 10 19:03:55 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
May 10 19:03:55 vps813080 shorewall[23748]: Processing /etc/shorewall/params ...
May 10 19:03:55 vps813080 shorewall[23748]: Processing /etc/shorewall/shorewall.conf...
May 10 19:03:55 vps813080 shorewall[23748]: Loading Modules...
May 10 19:03:55 vps813080 shorewall[23748]: Compiling /etc/shorewall/zones...
May 10 19:03:55 vps813080 shorewall[23748]: Compiling /etc/shorewall/interfaces...
May 10 19:03:55 vps813080 shorewall[23748]: Determining Hosts in Zones...
May 10 19:03:55 vps813080 shorewall[23748]: Locating Action Files...
May 10 19:03:55 vps813080 shorewall[23748]: Compiling /etc/shorewall/policy...
May 10 19:03:56 vps813080 shorewall[23748]: Adding Anti-smurf Rules
May 10 19:03:56 vps813080 shorewall[23748]: Adding rules for DHCP
May 10 19:03:56 vps813080 shorewall[23748]: Compiling TCP Flags filtering...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling Kernel Route Filtering...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling Martian Logging...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling Accept Source Routing...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling /etc/shorewall/tcinterfaces...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling /etc/shorewall/snat...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling MAC Filtration -- Phase 1...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling /etc/shorewall/rules...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling /etc/shorewall/conntrack...
May 10 19:03:56 vps813080 shorewall[23748]: Compiling MAC Filtration -- Phase 2...
May 10 19:03:56 vps813080 shorewall[23748]: Applying Policies...
May 10 19:03:56 vps813080 omr-admin.py[769]: DEBUG:    Get status...
May 10 19:03:56 vps813080 shorewall[23748]: Generating Rule Matrix...
May 10 19:03:56 vps813080 shorewall[23748]: Creating iptables-restore input...
[... 73 lines cut ...]
May 10 19:04:08 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
[... cut ...]
May 10 19:04:32 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
May 10 19:04:33 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
May 10 19:04:34 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
May 10 19:04:35 vps813080 ss-server[712]: getpeername: Transport endpoint is not connected
May 10 19:04:36 vps813080 omr-admin.py[769]: DEBUG:    Get status...
May 10 19:04:36 vps813080 omr-admin.py[769]: DEBUG:    Get status: done
```

DeeJayMX, 2020-05-10:
I'm accessing the web server on https routed on the vps to OMR.
but it comes up after some minutes...

Ysurac, 2020-05-10:
The error "getpeername: Transport endpoint is not connected" should happen on the VPS when website can't be reached.
Can you check using top/htop/... cpu/memory usage of the VPS ?

DeeJayMX, 2020-05-10:
nothing alarming:
```
top - 19:33:25 up  3:49,  2 users,  load average: 0.04, 0.01, 0.00
Tasks:  88 total,   2 running,  85 sleeping,   0 stopped,   1 zombie
%Cpu(s):  0.7 us,  0.7 sy,  0.0 ni, 96.0 id,  0.0 wa,  0.0 hi,  2.7 si,  0.0 st
MiB Mem :   1950.1 total,   1405.6 free,    339.7 used,    204.8 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   1448.8 avail Mem
```

Ysurac, 2020-05-10:
what is the result of a ```ping 8.8.8.8``` from the VPS when this fail ?
What is also the status of each wan in OMR status page when this fail ?

DeeJayMX, 2020-05-11:
So it's really freezing right now, no more SSH on VPS

DeeJayMX, 2020-05-11:
on the KVM when I'm trying to ping 8.8.8.8 > Network is unreachable

DeeJayMX, 2020-05-11:
on eth0 no more inet, I mean no more IPv4

DeeJayMX, 2020-05-11:
OMR status page:

[image]

Ysurac, 2020-05-11:
If no more IPv4 on eth0 of the VPS, the issue is really on the VPS side.
I think IPv4 is set via DHCP on the VPS ? You can try to set it as static IP ( https://wiki.debian.org/NetworkConfiguration )

Ysurac, 2020-05-11:
Who is the VPS provider ?

DeeJayMX, 2020-05-11:
OVH VPS 2018 SSD 1

DeeJayMX, 2020-05-11:
So I force a DHCP release / renew and everything comes up !
is it possible that the DHCP client of the VPS is trying to get an IP on a bad interface and so it release an adress and can't have a new one ?

Ysurac, 2020-05-11:
Not really.
You can try something like ```grep dhclient /var/log/syslog``` to check if there is an error.

DeeJayMX, 2020-05-11:
no error on grep dhclient /var/log/syslog
but it's just displaying my last attempt at 16:58

DeeJayMX, 2020-05-11:
I put the IP in static mode in the VPS configuration, we will see...

DeeJayMX, 2020-05-14:
Seems to be good now.
is it normal that on VPS log messages Shorewall is keeping reloaded everysecond ?

closed 2020-09-17 by github-actions[bot]