Ysurac/openmptcprouter #1017: install issue on Debian 10

jhackeur, 2020-05-10:
## Expected Behavior

Unable to install OpenMPTCPserver on Debian 10
## Current Behavior

When I ran the installation server script, I got this issue:
" E: Package 'tracebox' has no installation candidate
E: Package 'omr-iperf3' has no installation candidate "
## Possible Solution

Package not available on repository ?
## Steps to Reproduce the Problem

  1. Install Debian 10 via Netinstall on ESX virtual host
  2. install via wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh
  3. get the error message

## Context (Environment)

Unable to install new OpenMPTCPserver  instance on Debian 10
Complete log here:
[https://lafibre.info/routeur/openmptcprouteur/?action=dlattach;attach=83482](url)

## Specifications

  - OpenMPTCProuter version: no relevant here
  - OpenMPTCProuter VPS version: version installed via this script: wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh
  - OpenMPTCProuter platform: Debian 10 hosted in ESXI with OneProvider

gnosoz, 2020-05-10:
following as I had the same results when I tried on both physical machine
and virtual machine.

Ysurac, 2020-05-10:
I'm not able to reproduce the issue.
Packages are available in the repo.
What is the result of ```apt-get install omr-iperf3``` and ```apt-cache search omr``` ?

jhackeur, 2020-05-10:
Here the result:
_root@openMPTCPserver:/home/administrateur# apt-get install omr-iperf3
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
Aucune version du paquet omr-iperf3 n'est disponible, mais il existe dans la base
de données. Cela signifie en général que le paquet est manquant, qu'il est devenu obsolète
ou qu'il n'est disponible que sur une autre source

E: Le paquet « omr-iperf3 » n'a pas de version susceptible d'être installée
root@openMPTCPserver:/home/administrateur# apt-cache search omr
freewnn-common - Fichiers partagés pour les paquets FreeWnn
freewnn-jserver - Système de saisie en japonais
libwnn-dev - Header files and static libraries for Wnn (FreeWnn jserver)
libwnn0 - FreeWnn library for Wnn (FreeWnn jserver)
sdaps - scripts for data acquisition with paper-based surveys
libwnn6-1 - Wnn6 client library
libwnn6-dev - Header files and static library for Wnn6 client library
omr-vps - OpenMPTCProuter VPS meta package
omr-vps-admin - OpenMPTCProuter VPS admin rest api
omr-vps-tools - OpenMPTCProuter VPS tools
root@openMPTCPserver:/home/administrateur#_

Ysurac, 2020-05-10:
Strange... What is the result of ```dpkg --print-architecture``` ?

jhackeur, 2020-05-10:
root@openMPTCPserver:/home/administrateur# dpkg --print-architecture
i386
root@openMPTCPserver:/home/administrateur#

Ysurac, 2020-05-10:
This is the problem. Only x86_64 is supported.

gnosoz, 2020-05-10:
I have amd64 and same risults, I am retrying as we speak and will let you
know where is giving the errors

Ysurac, 2020-05-10:
I added a check in next release of VPS script.

closed 2020-05-10 by Ysurac

gnosoz, 2020-05-10:
arch amd64 here and show a different set of errors regardless of deb version as I tried a few for good measure.
The script was working perfectly apart  from a small error in the middle related to the locale settings which shouldn't bother perl too much

```
+ scripts/feeds install -a -d y -f -p openmptcprouter
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:

```

Then it suddenly went tits up looking for dependencies which are already installed in the system
```
+ cp .config.keep .config
+ echo Done
Done
+ [ ! -f ../../config-x86_64 ]
+ echo Building openmptcprouter for the target x86_64
Building openmptcprouter for the target x86_64
+ make defconfig
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
#
# configuration written to .config
#
+ make IGNORE_ERRORS=m -j1
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-wlan-gen2', which does not exist
time: target/linux/prereq#0.07#0.03#0.11
 make[1] world
 make[2] tools/compile
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/tar compile
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/root/openmptcprouter/x86_64/source/include/toplevel.mk:220: world] Error 1

```

I checked which of these packages I have or don't and here is the result:
```
 # apt list libpam* libgnutls* *ldap* libidn2* *mt6625l* |grep installed
 
 libgnutls30/stable,stable,now 3.6.7-4+deb10u3 amd64 [installed]
 libidn2-0/stable,stable,now 2.0.5-1+deb10u1 amd64 [installed]
 libldap-2.4-2/stable,stable,now 2.4.47+dfsg-3+deb10u2 amd64 [installed,automatic]
 libldap-common/stable,stable,now 2.4.47+dfsg-3+deb10u2 all [installed,automatic]
 libpam-modules-bin/stable,now 1.3.1-5 amd64 [installed]
 libpam-modules/stable,now 1.3.1-5 amd64 [installed]
 libpam-runtime/stable,now 1.3.1-5 all [installed]
 libpam-systemd/stable,now 241-7~deb10u4 amd64 [installed]
 libpam0g/stable,now 1.3.1-5 amd64 [installed]
 
```

Ysurac, 2020-05-10:
This is not VPS script, it's not related to this issue.

gnosoz, 2020-05-10:
very sorry, yes I thought I was writing on the router.
 I shall delete these comments

jhackeur, 2020-05-11:
> 
> 
> This is the problem. Only x86_64 is supported.

Indeed, working fine with x64 ! Thx!