Ysurac/openmptcprouter #1018: Debian install from source of the OpenMPTCrouter

gnosoz, 2020-05-10:
## Expected Behavior
router installation from scratch cloning repo onto a fresh debian 10 minimal 

## Current Behavior
Errors in arch amd64 build.sh install regardless of deb version as I tried 10 & 11 installs for check.
The script was working perfectly apart from a small error in the middle related to the locale settings which shouldn't bother perl at all but I report it nonetheless

+ scripts/feeds install -a -d y -f -p openmptcprouter
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:

Then it suddenly went tits up looking for dependencies which are already installed in the system

```
+ cp .config.keep .config
+ echo Done
Done
+ [ ! -f ../../config-x86_64 ]
+ echo Building openmptcprouter for the target x86_64
Building openmptcprouter for the target x86_64
+ make defconfig
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist

configuration written to .config

+ make IGNORE_ERRORS=m -j1
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-wlan-gen2', which does not exist
time: target/linux/prereq#0.07#0.03#0.11
 make[1] world
 make[2] tools/compile
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/tar compile
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/root/openmptcprouter/x86_64/source/include/toplevel.mk:220: world] Error 1
```

I checked which of these packages I have or don't and here is the result:

```
apt list libpam* libgnutls* *ldap* libidn2* *mt6625l* |grep installed
 
 libgnutls30/stable,stable,now 3.6.7-4+deb10u3 amd64 [installed]
 libidn2-0/stable,stable,now 2.0.5-1+deb10u1 amd64 [installed]
 libldap-2.4-2/stable,stable,now 2.4.47+dfsg-3+deb10u2 amd64 [installed,automatic]
 libldap-common/stable,stable,now 2.4.47+dfsg-3+deb10u2 all [installed,automatic]
 libpam-modules-bin/stable,now 1.3.1-5 amd64 [installed]
 libpam-modules/stable,now 1.3.1-5 amd64 [installed]
 libpam-runtime/stable,now 1.3.1-5 all [installed]
 libpam-systemd/stable,now 241-7~deb10u4 amd64 [installed]
 libpam0g/stable,now 1.3.1-5 amd64 [installed]
```
 

## Possible Solution

I tried then having a look at toplelev.mk and world but the error even with sh +x dowsn't show exactly in which lines the problem is happening.

I can try install all available packages to check if that resolves but to be honest I am not sure that would help...
What are your thoughts on this matter?

## Steps to Reproduce the Problem
  1. fresh debian 10 minimal (ssh + common)
  2. apt install build-essential git unzip ncurses-dev libz-dev libssl-dev openssl-1.0-dev
  python python3-dev python3.5 libelf-dev subversion gettext gawk wget curl rsync perl
   ==> openssl will give error as is already installed as version 1.1
  3. clone git router
  4. sh -x build.sh -j1

## Context (Environment)
I am trying to install freshly on debian as I am having issues with installing muninlite on the prepackaged version. This will help me monitor all systems with standard munin-node whilst I perform several performance tests which I would like to share here in the hope they will help your project version 1.0 :)

## Specifications
  - OpenMPTCProuter version: master from git clone
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: amd64 on VirtualBox, if successfull I will dd and flash on the router which is still amd64

Ysurac, 2020-05-10:
It doesn't look for dependencies, it compile them for the build toolchain.
Don't forget that custom images are not supported ;)
What is the error if you run ```make -j1 V=s``` in ```x86_64/source``` ?

gnosoz, 2020-05-10:
ok so the result is the following
```
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist

```
After above msg, which we are already aware of, it keeps on going fine with all checks until it stops showing  following details:
```
checking for mkfifo... yes
checking whether mkfifo rejects trailing slashes... yes
checking whether mknod can create fifo without root privileges... configure: error: in `/root/openmptcprouter/x86_64/source/build_dir/host/tar-1.32':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [Makefile:32: /root/openmptcprouter/x86_64/source/build_dir/host/tar-1.32/.configured] Error 1
make[3]: Leaving directory '/root/openmptcprouter/x86_64/source/tools/tar'
time: tools/tar/compile#13.00#3.44#19.89
make[2]: *** [toolexi s/Makefile:163: tools/tar/compile] Error 2
make[2]: Leaving directory '/root/openmptcprouter/x86_64/source'
make[1]: *** [tools/Makefile:159: /root/openmptcprouter/x86_64/source/staging_dir/host/stamp/.tools_compile_yyyyynnyyynyyyyynyynnyyyynyyyyyyyyyyyyyyyyynynnyyynnyyy] Error 2
make[1]: Leaving directory '/root/openmptcprouter/x86_64/source'
make: *** [/root/openmptcprouter/x86_64/source/include/toplevel.mk:220: world] Error 2

```

Ysurac, 2020-05-10:
So you have the answer : don't run this as root.

gnosoz, 2020-05-10:
lol I figured  my stupidity as soon as I saw the errors but wanted to share them nonetheless heheh
I honestly believed j1 would allow it to continue undisturbed but apparently that is not the case :)
Thank you, I will try it shortly and confirm that was just me being silly!

closed 2020-09-14 by github-actions[bot]