Ysurac/openmptcprouter #1019: Wireguard server  on openmptcp vps

brazzo97, 2020-05-10:
Hi i am trying to host a wireguard server on the same vps that i use for openmptcp
i tryed to open the port on shorewall but i cant get it to work , can you give me some help thanks 

  - OpenMPTCProuter version: beta3
  - OpenMPTCProuter VPS version: beta3
  - OpenMPTCProuter platform: <x64>

xabix, 2020-05-11:
Hello

This may answer to your question:
https://github.com/Ysurac/openmptcprouter/issues/884

brazzo97, 2020-05-11:
i am trying to open the port 51820 on the vps to tunnel my mobile phone traffic trough the vps to get adblocking on the go , i am not trying to use wireguard to aggregate. but i am not able to open any port on the vps

Ysurac, 2020-05-11:
To open port for IPv4 it's in ```/etc/shorewall/rules```, you can copy the rule used for SSH on port 65222. After changes you need to reload shorewall: ```systemctl reload shorewall```

brazzo97, 2020-05-11:
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the$
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
ACCEPT          net             $FW             udp     51820   # i did that but still cant connect 
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#     

i triyed but i still cant connect and port seems closed

Ysurac, 2020-05-11:
With this UDP port 51820 is open, at least if you restarted shorewall.

brazzo97, 2020-05-11:
i restarted shorewall and even the vps but i cant reach wireguard from the client

sirius5481, 2020-05-11:
Hi , got the same issue : ssh port is open according to this [test](https://www.yougetsignal.com/tools/open-ports/) but applying the same rule with a an other port didn't do the trick.
Must be an other setting but i'm still looking for it

Ysurac, 2020-05-11:
A port is see as open only if it is open in firewall and a service is really listening to it.

sirius5481, 2020-05-11:
> it

Yep it's working , my mistake :) so happy thank you :+1:

Ysurac, 2020-05-11:
No, rules are not deleted by updates or by the VPS API.

closed 2020-09-14 by github-actions[bot]