Ysurac/openmptcprouter #1020: can't get public ip address from shadowsocks

jhackeur, 2020-05-11:
## Expected Behavior

Unable to connect VPN between router and server
## Current Behavior

Direct output access show can't get public ip from shadowsocks
https://lafibre.info/routeur/openmptcprouteur/?action=dlattach;attach=83529
## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

OpenMPTCProuter running on local ESXI, server running on ESX hosted in Onprovider behind firewall

## Specifications

  - OpenMPTCProuter version: 4.19.108
  - OpenMPTCProuter VPS version: version available in the install script
  - OpenMPTCProuter platform: ESX - Debian10 - x64

Ysurac, 2020-05-11:
Reboot the server, check that kernel on the server use mptcp with ```uname -a ```
Check that all ports available in the doc are open and that you use the correct key in the wizard.
(And you should really put real version in the specification)

jhackeur, 2020-05-11:
Hi Ysurac
thanks for this prompt reply!

There is no port restriction between the both public IP, and all this ports has been natted to server side:
    SSH port is 65222 (TCP)
    Shadowsocks port is 65101 (TCP & UDP)
    Glorytun port is 65001 (TCP & UDP)
    OMR JSON admin is 65500 (TCP)
    OpenVPN port is 65301 (TCP)
    MLVPN ports are 65201-65208 (UDP)
    Iperf3 on port 65400 (TCP & UDP)
    DSVPN port is 65401 (TCP)

regardind kernel version:
root@OpenMPTCPserver-Debian:/home/administrateur# uname -a
Linux OpenMPTCPserver-Debian 4.19.104-mptcp #1 SMP Mon Feb 17 11:08:15 UTC 2020 x86_64 GNU/Linux

OpenMPTCProuter VPS version - how can I check that ?

Ysurac, 2020-05-11:
OpenMPTCProuter VPS version is available in ```/etc/motd```.

jhackeur, 2020-05-11:
Ok then: OpenMPTCProuter VPS 0.1015

jhackeur, 2020-05-13:
I don't know what is the issue, I proceed with another installation and working fine yet - closed

closed 2020-05-13 by jhackeur