Ysurac/openmptcprouter #1021: Can't get aggregation...

dettofatto, 2020-05-11:
## Expected Behavior
If I test my connection with omr bypass i've correct speed for each. But when is aggregated I don't have the "right" speed, but is slower.
On the first wan I have a b535  huawei, with WIND LTE (about 120/30)  with speed test, on the second wan I have b715, always Wind lte (about 210/45)
 

## Current Behavior
When they are aggregated I have about 130/30

## Possible Solution
 I don't know

## Steps to Reproduce the Problem

  1. Raspberry pi4, with 2 lans (1 integrated, 1 on usb 3.1 192.168.1.155) I've have the same issue if use only integrated lan
  2. Wan 1 192.168.2.1 b715 huawei LTE
  3. Wan 2 192.168.3.1 b535 huawei LTE

Ysurac, 2020-05-11:
To test speed, there is ```omr-test-speed``` available via SSH on the router. Without argument this test aggregated speed, with wan1 argument this test speed on wan1 real interface.

You can try others Multipath TCP scheduler in Network->MPTCP and other Congestion control.

Check also memory/cpu usage on the RPI and on the VPS: communication between RPI and VPS are encrypted, this can use lot of CPU.

dettofatto, 2020-05-11:
I'll try...! I'm testing speed with omr-test-speed but is almost the same.
Can I test with only 1 gb?
It's strange, because when I used an usb 4g lte adapter (alcatel ik40) on the same pi4, I had a really aggregated speed.
In any case I had made a new configuration.

Ysurac, 2020-05-11:
ctrl+c to stop a speed test after 1-2 minutes.

dettofatto, 2020-05-11:
I've done... seems that the issue is on huawei b715, that has more speed directly (bypass), but is more slow during aggregation. I can't understand why. The lte provider is the same and in any case the b715 is about double faster than my other b535, that haven't any external antenna.

Ysurac, 2020-05-11:
If it's same provider and use same provider antenna, it's possible that you reach max bandwidth on a band.

dettofatto, 2020-05-11:
It is a good question... But I can't understand why I have this issue only when I test with omr-test-speed wan2 or aggregated, and not when I use bypass...
And I don't have this issue when I use theese aggregated connections with usb adapter. Or if I use simultaneously theese modem with different pc.

Ysurac, 2020-05-11:
```omr-test-speed wan2``` should give same speed as a bypass, this use omr-bypass to download directly from the interface.
Check that your switch doesn't limit RPI interface ?

dettofatto, 2020-05-11:
the wan2 and wan1 are on the same interface...
I'm trying switching vps, in the next hours I'll try other switch...
Thank you for your job.

dettofatto, 2020-05-11:
Thi is my situation...
[image]

Ysurac, 2020-05-12:
You tried others Multipath TCP scheduler in Network->MPTCP and other Congestion control ?

dettofatto, 2020-05-12:
I'm trying, but you suggest to reboot the router, when I change scheduler/congestion control?
Thank you.

Ysurac, 2020-05-12:
Yes, to be sure that reset all connections. You can begin with BLEST scheduler and Cubic congestion control.

dettofatto, 2020-05-13:
Ok... I've solved.... The issue was the ethernet cable that was very long... I don't know why it was ok, during the single step, anyway with 2 modems connected I didn't have any aggregation. When I used 2 ethernet cables, I had a regular aggregation

closed 2020-05-13 by dettofatto

Ysurac, 2020-05-13:
ok thanks for the feedback and for the donation. Now I know it's something to check when there is an issue :)