Ysurac/openmptcprouter #1022: IPv6 Subnet Routing

weltmeyer, 2020-05-12:
## Expected Behavior
Routing IPv6 Subnet to Homenet

## Current Behavior
IPv6 is natted

## Possible Solution
dunno

## Steps to Reproduce the Problem
I tried so far:
My Server IPv6:
XXXX:1234:5678::1/64
My Subnet: 
XXXX:1234:5678::/48
Subnet for Routing between vps and router:
XXXX:1234:5678:1::/64
VPS omr6-in4@user0=
XXXX:1234:5678:1::1/64
Router omr6in4
XXXX:1234:5678:1::2/64
(Hardcoded/Edited som values on both sides as the default always makes it back to fe80...addr)

Router LAN Subnet:
XXXX:1234:5678:2::/64
Router LAN IP:
XXXX:1234:5678:2::1/64
Clients get IPs from this Subnet.
All working, IPV6 works, but:
The Address, the Remote Side sees is always the ipv6 of the VPS(XXXX:1234:5678::1)
instead the ones of the Clients

## Context (Environment)
Route the Ipv6 packets via omr6-in4,
add routable ipv6 addresses to omr6-in4
dont nat

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-05-12:
This will be available in next release, at least when /48 available on VPS, and a /56 will be available on router.

weltmeyer, 2020-05-12:
> This will be available in next release, at least when /48 available on VPS, and a /56 will be available on router.

That sounds good.
Also, what could be nice, is to have different interfaces in v4 and v6.

i modified the /shorewall6/params.net to be eth1 and added some parts in omrr-admin.py
to get the correct interface for ipv6

the subnets should be configurable ;)

btw.:
cuzrrently i am using wireguard between a second router at home and the vps to tunnel my ipv6 subnets through omr.
this makes it using only one of my uplinks. :(

weltmeyer, 2020-05-12:
[image]

Adorfer, 2020-05-13:
a /48 on a VPS is hard to find, even online.net does seem to have reduced it to /64. 
My take: 
1) /64 on the router enough for SLAAC
2) some VPS providers offer to order multiple /64 blocks, which are by luck not direct neighbours.

tunneling a /48 via gre from sixxs would be the last resort on the vps...

Ysurac, 2020-05-13:
@weltmeyer Interface for IPv6 only support can be added. IPv6 is provided via a tunnel ? I don't have any case on VPS/servers where I need to use another interface.
For the remote address, when shadowsocks is enabled I don't think it's possible to output another address than the server IP.

@Adorfer Milkywan provide a /48

weltmeyer, 2020-05-13:
I have a second interface with for routing of my IPv6 PI Space :)

It would be nic if we can just add a clear routing function, no mather what subnet size.

You could have one IPv6-Address  for VPS,
one Subnet for Routing between VPS and OMR (the Addresses on the 6in4 interface) and another /64 for the clients(Or let it be whatevetr size we get)

My Setup is even further:

VPS A(Default):
eth0 ipv4/ipv6
eth1 ipv6 subnet router
eth2 INTERNAL

VPS B(OMR-VPS)
eth0 ipv4/ipv6
eth1 INTERNAL

the internal routes a part of my ipv6 for the omr /home usage

OMR@Home:
eth0 IPv6: HOME_ROUTING/64
WAN1
WAN2

OPNSENSE: The Real Router @home, using OMR as the UPStream Gateway and route some more of my /64 Networks via OMR(Currently not working because of the NAT, using Wireguard here currently)

W

Ysurac, 2020-08-12:
OpenMPTCProuter v0.55 is released. This should be fixed in release.
Fell free to re-open bug if needed.

closed 2020-08-12 by Ysurac