Ysurac/openmptcprouter #1024: Install Router on a cubietruck ?

moritan, 2020-05-14:
Hello,

Has anyone try to install the router on a cubietruck card or some thing like that ?
I run armbian on it and want to try openmptcprouter.

It 's based on an Allwinner A20 SOC.

I supposed that i will need to recompiled the source but i read something about the kernel that need to be patch.
What do i need to check before starting to validate that my card will work ?

Thanks for your help.

http://linux-sunxi.org/Cubietruck
https://www.armbian.com/cubietruck/

Ysurac, 2020-05-14:
This seems to be supported by OpenWrt, so should be easy to compile OpenMPTCProuter for it following the doc: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform.
But the card seems really slow, not sure you can get any good result with it.

moritan, 2020-05-14:
ok, i will give a try. I don't have choice, it's the only one free that i have for the moment.

Ysurac, 2020-05-14:
You can also try with virtualbox, vmware, hyper-v, kvm,... on your computer. It's free if you have a computer that is less than 10 years old :)

closed 2020-09-17 by github-actions[bot]