Ysurac/openmptcprouter #1026: [Banana PI R2] errors on LAN ports (any)

vobo70, 2020-05-15:
Hello,
I should get high speed on 1GbE LAN ports (or WAN) but I have very low speed,
probably due to errors:

`maciek@bananapipro1:~$ sudo iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.21 port 48602 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  15.4 MBytes   129 Mbits/sec    4    276 KBytes
[  5]   1.00-2.00   sec  15.7 MBytes   131 Mbits/sec    0    314 KBytes
[  5]   2.00-3.00   sec  15.4 MBytes   129 Mbits/sec    0    348 KBytes
[  5]   3.00-4.00   sec  17.2 MBytes   144 Mbits/sec   29    273 KBytes
[  5]   4.00-5.00   sec  20.9 MBytes   175 Mbits/sec    0    325 KBytes
[  5]   5.00-6.00   sec  22.2 MBytes   186 Mbits/sec   13    260 KBytes
[  5]   6.00-7.00   sec  16.4 MBytes   138 Mbits/sec    0    307 KBytes
[  5]   7.00-8.00   sec  22.4 MBytes   188 Mbits/sec    0    359 KBytes
[  5]   8.00-9.00   sec  21.0 MBytes   176 Mbits/sec    0    399 KBytes
[  5]   9.00-10.00  sec  20.8 MBytes   174 Mbits/sec    0    437 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   187 MBytes   157 Mbits/sec   46             sender
[  5]   0.00-10.02  sec   187 MBytes   156 Mbits/sec                  receiver`

I know it's beta3 I will try to upgrade to beta4

My Banana PI R2 is working on this image:
openmptcprouter-v0.55beta3
my platform is Banana PI R2
My LAN port acts as WAN
and my WAN1-WAN4 ports are acts as BR-LAN
Mptcp is disabled

Other devices in my network, between them I get very fast speed (they're connected through 1GbE switch)

vobo70, 2020-05-15:
I have got this answer on Banana PI forum:
You can check if you have trgmii active on gmac 1, gmac2 is not available on default (and mine 5.4)…and look if trgmii-patches are in:

https://patchwork.kernel.org/project/linux-mediatek/list/?series=266293

Ysurac, 2020-05-15:
beta release are not supported...
You should use dedicated issue: https://github.com/Ysurac/openmptcprouter/issues/959

vobo70, 2020-05-15:
OK. Thank you

closed 2020-05-15 by vobo70

Ysurac, 2020-05-15:
I'm testing patches for next beta.