Ysurac/openmptcprouter #1027: Port forwarding doesn't work

Koubiacz, 2020-05-16:
Hi,

I want to redirect port 1194 UDP from the VPS to a other rapsberryPI that runs VPN

I read the "Port forwarding" documentation in the Wiki.

I can't contact my raspberry where OpenVPN is located, 
I checked on the VPS that port 1194 is not open, I tried to reboot it, it didn't change anything.

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0 (64Bits)
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1015
  - OpenMPTCProuter platform: RPI3B+ 
  - OpenMPTCProuter: Dedicated Server Kimsufi (OVH), Debian 10.4

Thx

Ysurac, 2020-05-17:
How did you check that port 1194 is not open ? You checked in ```/etc/shorewall/rules``` on the VPS ?

Koubiacz, 2020-05-17:
I'm using "lsof -i :1194" and "netstat -pln" on the VPS and the ports aren't open

In the /etc/shorewall/rules I have 
`DNAT            net             vpn:$OMR_ADDR   udp     1194  `

Ysurac, 2020-05-17:
lsof or netstat only show used ports, not if they are open or close by the firewall.
The rule in shorewall is good, you can check using ```iptables-save | grep 1194``` that it's applied on firewall and port is open and redirected. In this case the issue is not on VPS part.

Koubiacz, 2020-05-17:
Result of iptables-save | grep 1194
```
-A net-vpn -d 10.255.255.2/32 -p udp -m udp --dport 1194 -j ACCEPT
-A net_dnat -p udp -m udp --dport 1194 -j DNAT --to-destination 10.255.255.2
```

I also tried a telnet on port 1194 from an outside machine. I haven't been able to open a connection.

Koubiacz, 2020-05-19:
Hi,

I was able to get the forwarding to work using TCP on a port other than 1194.

closed 2020-09-22 by github-actions[bot]