Ysurac/openmptcprouter #1029: WiFi ISP requires webpage authentication

aFrosty0, 2020-05-17:
## Expected Behavior
Connect OpenMPTCProuter to a Wifi network the usual way, as one of the WAN connections.
Use OMR-Bypass to set a device (windows laptop) source LAN IP bypass to wlan0 interface
Go to any webpage on device to trigger the Wifi login page, and login.
Remove OMR-Bypass rule.

## Current Behavior
Can't get the Wifi login page to load.
Reported as "Gatweway DOWN" on status page

## Possible Solution
I can successfully use the Wifi network if I use an Android phone to connect and authenticate onto the wifi, then USB tether to the OpenMPTCProuter. This is not ideal as I had planned to use the same android phone for LTE connection.

## Steps to Reproduce the Problem
(Described in first section)

## Context (Environment)
I assume anyone who's ISP requires an additional login through the browser will experience this problem.
Maybe there is already a better solution to this problem I have not found?

Many thanks

## Specifications
  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: Unsure where to find this information. Installed today (17/05/2020 using "debian10-x86_64.sh"
  - OpenMPTCProuter platform: RPI4 & GCP instance

Ysurac, 2020-05-17:
Using a connection with a captive portal seem difficult. I never used any captive portal, so I have no real solutions for that.

aFrosty0, 2020-05-17:
Some testing I've done may provide some insight into a possible solution or workaround using the existing utilities in OpenMPTCProuter:

It works fine through NAT. For example if I use a ddwrt router to connect to the wifi as the WAN interface, then connect a laptop to the LAN side, the captive portal is successful behind the NAT, and need only be done once, on only one device, then any device on the LAN can use the connection.

It can also be tricked with a MAC address change, so it must use the device MAC address to identify users. Although this will be troublesome every time the wifi requires login (every 24 hours in my case).
Also in the NAT example, I have tested to demonstrate the wifi ISP is recording the MAC address of the ddwrt router, and not the device behind the NAT.

For now I have a workaround with a 2nd Raspberry Pi in between, do the captive portal, and bypass rules so I can VNC to the Raspberry Pi every day to relogin to the captive portal!

closed 2020-09-20 by github-actions[bot]