Ysurac/openmptcprouter #1030: Streaming issues

bigcachetx, 2020-05-19:
## Expected Behavior
2 Wan bond and speed up to 100Mbps / 87Mbps (fast.com)
The youtube connect speed should close these 2 numbers

## Current Behavior
Youtube connection speed is only 10Mbps or slower
When I use only 1 WAN, work perfect, up to it's full speed.

## Possible Solution
I don't know
## Steps to Reproduce the Problem
  1.2 WAN bond and open a 4K video on youtube, in stats for nerd, check the "connection speed"
  2.Unplug 1 WAN and try to open this 4K video, check the "connection speed"
  

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  v0.54.7
  - OpenMPTCProuter VPS version: 0.1015.4.19.104-mptcp
  - OpenMPTCProuter platform: X86-64

Adorfer, 2020-05-19:
suggestion: can you top/htop on router and vps? (is the load staying at reasonable avalues or does one of the systems "hit the roof" when in dualWAN mode?

Ysurac, 2020-05-19:
All is green in the status page ?
The VPN is enabled ?