Ysurac/openmptcprouter #1032: Slow upload only  for surveillance cameras through port forward

dettofatto, 2020-05-20:
## Expected Behavior
I've some surveillance cameras (about 10), I can see them with port forward through my bounding 4g (one master, one backup)  with my public ip

## Current Behavior
The upload speed when watch the (also whatching only one) surveillance camera is terrible slow! 
If I use reverse tunneling with ssh , I don't have any issue! ( I'm using reverse tunneling, and now I'm studing your brilliant sw)
With reverse tunneling the upload speed of the camera  is about >250 kb/s, with vps max 80 kb/s, so is very slow. Seems a replay....

## Possible Solution
Boh.....

## Steps to Reproduce the Problem

  1. Enable port forward for surveillance camera
  2. Enable reverse tunneling for the same surveillance camera
  3. Test the difference speed/Image

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: debian 
Version 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: pi4

Ysurac, 2020-05-20:
Port forward use the VPN, by default Glorytun TCP. You can try another VPN in the wizard, advanced settings checkbox.

dettofatto, 2020-05-20:
So, for you, the issue for the low upload speed of only my cameras  is Glorytun TCP?

Ysurac, 2020-05-20:
Maybe.

dettofatto, 2020-05-20:
I've tested... it's about the same...
Any other suggestions?
I've seen that glorytun udp is more responsive. It is placebo, or not?

dettofatto, 2020-09-17:
No more news! It is always the same...

kevin39, 2020-09-19:
Because it do not seems to be an OMR issue. On my side, my Xiaomi Cameras are working fine without upload slowness.

A feedback from my experien: But be careful with IPV6. I've been having slow upload on many services (WhatsApp, Facebook, etc) until I realized that one of my ISP routers was forcing DHCP ipv6 enabled. So some of my devices were uploading using IPV6 (and so, bypassing OMR). I blocked IPV6 on my ISP's router and since few month everything is working fine. Maybe it will help you.

closed 2021-01-04 by github-actions[bot]