Ysurac/openmptcprouter #1034: Problem with starting Router on PC x86_64 and Raspberry3B+ and Virtualbox

anjszulc, 2020-05-21:
## Expected Behavior
The router should start up but stop at the command
8021q: adding VLAN 0 to HW filter on device wan2

## Current Behavior
hang on 8021q: adding VLAN 0 to HW filter on device wan2 

## Possible Solution

## Steps to Reproduce the Problem

 I try different images for x86 (DELL Optiplex with two ethernet cards
Rasp 3B+
Virtualbox 

 Image for VirtualBox
[image]

Rasp3B+ screen
[image]

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 Ubuntu / 0.1015 Debian as backup
  - OpenMPTCProuter platform: RPI3/x86_64/VBOX x86_64

Ysurac, 2020-05-21:
Press "Enter" to display the login. Interface is available by default at http://192.168.100.1/

anjszulc, 2020-05-21:
ohh...shit... stupid me...

Ysurac, 2020-05-21:
No problem, you are not the first. I know it can be strange when we know other Linux distribution :)

closed 2020-05-21 by Ysurac