Ysurac/openmptcprouter #1035: Traduction

Quenty-tolosan, 2020-05-22:
Bonjour,

j'ai vu des captures du logiciels en français et en anglais.
Je souhaiterais savoir s'il est possible de proposer une autre langue.
Je n'ai pas bien compris comment fonctionnait le système.

Merci !

Ysurac, 2020-05-22:
Pour le moment en traduisant les .po dans les répertoire luci-app-*
Mais je vais prochainement ajouter un outils de traduction sur un serveur.

Ysurac, 2020-08-12:
Un outils de tradition est maintenant disponible sur https://weblate.openmptcprouter.com/

closed 2020-08-12 by Ysurac

Quenty-tolosan, 2020-08-12:
Merci ! C'est possible d'ajouter l'occitan dans la liste des langues du projet ?

Ysurac, 2020-08-12:
Oui pas de soucis, il suffit de créer un compte dans weblate et de démarrer une nouvelle traduction.

Quenty-tolosan, 2020-08-12:
Non je n'ai pas le bouton pour démarrer une nouvelle traduction =/

Ysurac, 2020-08-12:
Tu dois l'avoir ici normalement : https://weblate.openmptcprouter.com/projects/omr/luciapplicationsopenmptcprouter/ sous la liste des langues.

Quenty-tolosan, 2020-08-23:
J'ai envoyé une demande pour traduire OpenWrt en occitan :)
Je vais voir ce que mes mémoires de traductions peuvent me faire gagner comme temps :D

Quenty-tolosan, 2020-10-13:
Salut @Ysurac sais-tu dire quand l'occitan sera intégré ?

Ysurac, 2020-10-13:
J'essaye de bosser là dessus. C'est plus complexe du fait que l'Occitan n'est pas présent dans OpenWrt.
Je pense qu'au minimum il y aura le fichier base.po à traduire en plus, mais faut déjà que j'arrive à faire afficher des trucs en Occitan ;)

Quenty-tolosan, 2020-10-14:
j'avais essayé d'ajouter l'occitan sur leur interface de traduction sans succès. Si tu as des contacts je suis preneur