Ysurac/openmptcprouter #1036: Third WAN not working

fooke123, 2020-05-22:
## Expected Behavior
Three WANs bond to speed up internet access.

## Current Behavior
Only the power of two WANs is used.

## Context (Environment)
I started a couple of weeks ago using OpenMPTCPRouter, everything was working fine. I used a DSL (FritzBoxG8) and a 4G (LTE) connection. 
Now i try to add another DSL Internet access (FritzBoxG5) to speed up my internet.
But there was no traffic through the new line. 
MPTCP Support Check for this interface said:
1: * 
2: * 
...
63: * 
64: * 
Result:
nil
I'm wondering about this, because the first WAN (FritzBoxG8) is the same Internet provider, an there everything is working. So i think i made a wrong configuration somewhere. But i'm a beginner and for me, everything was looking fine. Maybe someone could give me a hind, where to look and how to analyze this probleme. 

[image]

[image]

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: ESXi VM

Ysurac, 2020-05-22:
Do you have something about MPTCP in State->System log ?

fooke123, 2020-05-22:
My Current Syslog looks like this:
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS0, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS0
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS0, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty30, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty30
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty30, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty59, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty59
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty59, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty20, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty20
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty20, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty4, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty4
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty4, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty49, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty49
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty49, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty10, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty10
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty10, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS9, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS9
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS9, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty39, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty39
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty39, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty29, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty29
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty29, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty57, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty57
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty57, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty19, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty19
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty19, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty2, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty2
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty2, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty47, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty47
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty47, subsystem=tty
Fri May 22 17:14:57 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Fri May 22 17:26:12 2020 user.notice post-tracking-post-tracking: Get status and settings for LTE...
Fri May 22 17:35:22 2020 user.notice post-tracking-post-tracking: Get status and settings for LTE... Done
Fri May 22 17:37:01 2020 daemon.info dnsmasq-dhcp[17179]: DHCPREQUEST(eth0) 192.168.178.121 e4:f0:42:69:c7:fd
Fri May 22 17:37:01 2020 daemon.info dnsmasq-dhcp[17179]: DHCPACK(eth0) 192.168.178.121 e4:f0:42:69:c7:fd Google-Home-Mini
Fri May 22 17:39:14 2020 daemon.err /usr/bin/ss-redir[18232]: server recv: Host is unreachable

Ysurac, 2020-05-22:
what is the result, from the router via SSH of ```omr-test-speed eth3```,```omr-test-speed eth2```, ```omr-test-speed eth1``` and ```omr-test-speed``` (ctrl+c to stop after 1-2 minutes) ?

fooke123, 2020-05-22:
[image]

Ysurac, 2020-05-22:
And what is eth2 ? if it's the new connection then there is a problem. The OVH test site is not reachable with this interface.
I also see that the server can't be reached with ping, ping is filtered by provider or a firewall block ping on Server end ? What is the result of ```ping -B -I eth1 8.8.8.8``` and ```ping -B -I eth2 8.8.8.8``` ?

fooke123, 2020-05-22:
root@OpenMPTCProuter:~# ping -B -I eth1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.8.3 eth1: 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=53 time=26.0 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=53 time=25.6 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=53 time=25.4 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=53 time=26.6 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=53 time=42.2 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=53 time=25.7 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=53 time=25.2 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=53 time=51.8 ms
^C
--- 8.8.8.8 ping statistics ---
8 packets transmitted, 8 received, 0% packet loss, time 7008ms
rtt min/avg/max/mdev = 25.234/31.077/51.839/9.518 ms
root@OpenMPTCProuter:~# ping -B -I eth2 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.5.3 eth2: 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
26 packets transmitted, 0 received, 100% packet loss, time 26039ms

root@OpenMPTCProuter:~# ping -B -I eth2 192.168.5.1
PING 192.168.5.1 (192.168.5.1) from 192.168.5.3 eth2: 56(84) bytes of data.
64 bytes from 192.168.5.1: icmp_seq=1 ttl=64 time=7.29 ms
64 bytes from 192.168.5.1: icmp_seq=2 ttl=64 time=2.26 ms
64 bytes from 192.168.5.1: icmp_seq=3 ttl=64 time=6.66 ms
64 bytes from 192.168.5.1: icmp_seq=4 ttl=64 time=2.40 ms
64 bytes from 192.168.5.1: icmp_seq=5 ttl=64 time=34.8 ms
64 bytes from 192.168.5.1: icmp_seq=6 ttl=64 time=2.42 ms
64 bytes from 192.168.5.1: icmp_seq=7 ttl=64 time=1.83 ms
64 bytes from 192.168.5.1: icmp_seq=8 ttl=64 time=1.50 ms
64 bytes from 192.168.5.1: icmp_seq=9 ttl=64 time=9.88 ms
64 bytes from 192.168.5.1: icmp_seq=10 ttl=64 time=1.41 ms
64 bytes from 192.168.5.1: icmp_seq=11 ttl=64 time=12.7 ms
^C
--- 192.168.5.1 ping statistics ---
11 packets transmitted, 11 received, 0% packet loss, time 10012ms
rtt min/avg/max/mdev = 1.413/7.563/34.812/9.362 ms

eth2 is the new interface. 
I see that the internet connection through eth2 is not working. Is it a problem with the new internet-router or do i have to configure a static-rout or something like this for this way?

Ysurac, 2020-05-22:
No, route are added automatically.
What is the result of ```ip r``` ?
Are you sure of your config of eth2 interface on ESXI ?

fooke123, 2020-05-22:
root@OpenMPTCProuter:~# ip r
default via 10.255.255.1 dev tun0
default via 192.168.8.1 dev eth1 metric 3
default via 192.168.5.1 dev eth2 metric 6
default via 192.168.1.1 dev eth3 metric 7
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
82.165.125.244 via 192.168.8.1 dev eth1 metric 1
82.165.125.244 via 192.168.8.1 dev eth1 metric 3
82.165.125.244 via 192.168.1.1 dev eth3 metric 7
127.0.0.0/8 dev lo proto static scope link metric 1
192.168.1.0/24 dev eth3 scope link metric 7
192.168.5.0/24 dev eth2 scope link metric 6
192.168.8.0/24 dev eth1 scope link metric 3
192.168.16.0/24 dev eth4 proto static scope link metric 8
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd8a:8503:7b94::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_syn_retries='1'
network.globals.mptcp_debug='0'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='00:0c:29:64:16:bb'
network.lan.metric='2'
network.lan.ipaddr='192.168.178.1'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.ip4table='wan'
network.wan1.masterintf='eth0'
network.wan1.ipv6='0'
network.wan1.delegate='0'
network.wan1.ifname='eth1'
network.wan1.defaultroute='0'
network.wan1.peerdns='0'
network.wan1.hostname='mptcprouter'
network.wan1.proto='static'
network.wan1.macaddr='00:0c:29:64:16:c5'
network.wan1.ipaddr='192.168.8.3'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.8.1'
network.wan1.metric='3'
network.wan1.type='macvlan'
network.wan1.label='FritzBoxG8'
network.wan1.multipath='master'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='00:0c:29:64:16:c5'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth0'
network.wan2_dev.macaddr='00:0c:29:64:16:cf'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='1000'
network.omrvpn.ipv6='0'
network.omrvpn.proto='none'
network.omrvpn.metric='4'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.gateway='fe80::a00:1'
network.omr6in4.ip6addr='fe80::a00:2/128'
network.omr6in4.auto='0'
network.omr6in4.metric='5'
network.omr6in4.ipaddr='10.255.251.2'
network.omr6in4.peeraddr='10.255.251.1'
network.wan2=interface
network.wan2.masterintf='eth0'
network.wan2.ipv6='0'
network.wan2.defaultroute='0'
network.wan2.peerdns='0'
network.wan2.ifname='eth2'
network.wan2.delegate='0'
network.wan2.label='FritzBoxG5'
network.wan2.metric='6'
network.wan2.proto='static'
network.wan2.macaddr='00:0c:29:64:16:cf'
network.wan2.netmask='255.255.255.0'
network.wan2.ipaddr='192.168.5.3'
network.wan2.gateway='192.168.5.1'
network.wan2.type='macvlan'
network.wan2.multipath='on'
network.LTE=interface
network.LTE.proto='static'
network.LTE.ifname='eth3'
network.LTE.delegate='0'
network.LTE.ipaddr='192.168.1.3'
network.LTE.netmask='255.255.255.0'
network.LTE.gateway='192.168.1.1'
network.LTE.metric='7'
network.LTE.defaultroute='0'
network.LTE.peerdns='0'
network.LTE.ipv6='0'
network.LTE.macaddr='00:0c:29:64:16:d9'
network.LTE.label='LTE'
network.LTE.multipath='on'
network.LAN_G6=interface
network.LAN_G6.ifname='eth4'
network.LAN_G6.proto='static'
network.LAN_G6.netmask='255.255.255.0'
network.LAN_G6.delegate='0'
network.LAN_G6.metric='8'
network.LAN_G6.defaultroute='0'
network.LAN_G6.peerdns='0'
network.LAN_G6.ipv6='0'
network.LAN_G6.macaddr='00:0c:29:64:16:e3'
network.LAN_G6.ipaddr='192.168.16.1'
network.LAN_G6.multipath='off'

About ESXI - i'm thinking i configured it right, because the new Internetrouter (192.168.5.1) is reachable by ping.

Ysurac, 2020-05-22:
Some interfaces are defined as macvlan even if it's not the case, you remove them in network->macvlan and create them again.

fooke123, 2020-05-22:
I'm not sure how i configure it right. I delete both entrys and create the missing interfaces new. But there is no change. Ping from eth2 to 8.8.8.8 or to vps is still not possible.

fooke123, 2020-05-22:
I think the problem is somewhere in my esxi and vlan configuration. I've to figure it out - tomorrow. Thank you so far. I'll keep you up to date!

Adorfer, 2020-05-22:
> my esxi and vlan configuration

please enable "promisc eth/mac change/mac spoofing" for the VM in ESXI. Just for a test.

fooke123, 2020-05-25:
Hello,
i spend a couple of hours looking after the problem, but now i got the answer. 
First of all i have to say thank you! 
Ysurac, your software is working great! And i'm pleasantly suprised about the fast and helpful support - also for a newby like me. 
Now the technical details: 
I'm using the second WAN-Router by a Ubiquiti directional radio connection. 
After activating WDS (Transparent Bridge Mode) in both Radiomodules everything is working fine.

Adorfer, 2020-05-25:
If i understand correctly: WAN3 was not working because there was no working TCP/IP on the interface from external?

fooke123, 2020-05-25:
Hi Adorfer,
TCP/IP in basic was working, a ping to the WAN-Router connected ton WAN3 was working, but a further Ping to somewhere in the www (behind the WAN-Router) was not working.  After activating WDS (Transparent Bridge Mode) in both Ubiquiti-Modules, it is working and also the ping is working.

OMPTCP ---UbiquitiNanoM5 - - - - - - - UbiquitiNanoM5----WAN-Router---www

closed 2020-09-27 by github-actions[bot]