Ysurac/openmptcprouter #1038: OpenMCTCProuter does not survive to packages update suggested by the system itself

AndreaLeidi, 2020-05-24:
Thanking in advance the developer and those who want to reply to this message, I would like to point out the problem that performing the updates as suggested in the LuCI board SOFTWARE section causes a bit (too) often an explosion in the fragile balance between the official OpenWRT packages and those written and mantained for OpenMPTCProuter.

Briefly, after a full packages update as listed in the LuCI SOFTWARE section (naively perhaps, but not mistakenly: I think updating is a good practice that should always be encouraged, expecially for devices 24/7 online) I have a 4 ports x86-64 linux production system with running on kernel 4.19.108, without shadowsocks connectivity to the VPS and inaccessible in many LuCI sections:

- Advanced Reboot
- DSVPN
- MLVPN
- GloryTUN
- OMR Tracker
- Mail Settings
- OMR Bypass
- NGINX High Availability
- MacVLAN
- SQM QoS
- OMR DSCP
- Quota

because of:

"Insufficient permissions to read UCI configuration."

I would humbly like to suggest to:

1. arrange future OpenMPTCPRouter images without the LuCI package management (software) section, as updating them leads (more or less predictably) to a loss of system functionality

or

2. register those packages in the official OpenWRT repositories, this way advising the editors of their existence in this corner of the universe and giving  (system and us) a chance of survival to future updates

or

3. clearly alert users who download OpenMCTPC Router that the software comes AS IS , must not be updated with other methods than by downloading the most recent images from the site.

## Steps to Reproduce the Problem

  1. Spend some time to build a working Linux DSL bonding device based on OpenMCTPC Router
  2. Update the system with the .ipk packages provided by the LuCI SOFTWARE section
  3. KA-BOOM!

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.140.43567-aac1a8d
  - OpenMPTCProuter VPS version: fastpipe.io Ubuntu 18.04, server script debian9-x86_64.sh KERNEL_VERSION="5.4.0"
  - OpenMPTCProuter platform: Intel quad-core x86_64, kernel 4.19.108, 8GB Ram, 64GB HDD ( root partition correctly expanded to fill the disk with gparted)

Ysurac, 2020-05-24:
It was already reported in an issue: updating all packages doesn't work. I'm based on OpenWRT snapshot and this was updated with ACL and JS support, incompatible with previous releases.

ACL support will be in next release (that is in beta now), and I will rent a server to compile own version of packages (so update will be possible via the interface) and keep them stable for current release (and I hope to be able to use OpenWRT stable release when out with 5.4 kernel, even if I will keep an OpenMPTCProuter version based on snapshot).

OpenMPTCProuter is still in a release < 1.0, so still not considered as stable. And all servers/hardware are bought with donations and own money, so this take time to get a production ready router OS and environment for package/download/mirrors/....

(And for kernel version, stable VPS script is still in 4.19)

Adorfer, 2020-05-25:
Perhaps we can add a hint in the docu "do not update packages via LUCI!"?

Ysurac, 2020-05-25:
It's already added in the doc: https://github.com/Ysurac/openmptcprouter/wiki/Router-install#update

AndreaLeidi, 2020-05-25:
Thank you again for your interest and help. Restoring the original image through the firmware upgrade page then uploading a previous conf backup brought the features back to life.
In my case this was not enough, because I use OpenMPTCP on a linux PC. Some docker containers ran in this environment and, obviously, they have been nuked by the restore.
Anyway I was able to solve in an acceptable time, so imho OpenMPTCP remains a good choice even if it is still in beta. Just for my peace of mind, I will try to backup the whole thing with rsync.

closed 2020-05-25 by AndreaLeidi