Ysurac/openmptcprouter #1039: Comment faire fonctionner ipv6

xillibit, 2020-05-25:
Le VPS est chez Scaleway avec une ipv6 affectée. Sur la machine ou est installé OMR j'ai deux interfaces LAN, une pour la box internet et la box 4g et une pour les équipements du LAN. J'ai activé pour chaque LAN :

[image]

[image]

J'ai activé aussi ipv6 dans l’assistant de configuration mais l'état m'affiche ça : 

[image]

Mais pas d'ipv6 : 

[image]

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-05-26:
Le VPN doit être fonctionnel pour que le tunnel IPv6 fonctionne.
Il y a certainement plus d'infos dans Etat->Journal Système.

xillibit, 2020-05-26:
Il y a ça qui revient :  

`Tue May 26 16:16:31 2020 daemon.err odhcpd[5575]: Failed to send to ff02::1%lan@eth0 (Address not available)
Tue May 26 16:16:33 2020 daemon.err omr-tracker-ss[3029]: Failed to establish connection
Tue May 26 16:16:38 2020 daemon.err omr-tracker-ss[3029]: Failed to establish connection
Tue May 26 16:16:44 2020 daemon.err omr-tracker-ss[3029]: Failed to establish connection
Tue May 26 16:16:47 2020 daemon.warn odhcpd[5575]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Tue May 26 16:16:47 2020 daemon.err odhcpd[5575]: Failed to send to ff02::1%LAN2@eth1 (Address not available)
Tue May 26 16:16:47 2020 daemon.warn odhcpd[5575]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Tue May 26 16:16:47 2020 daemon.err odhcpd[5575]: Failed to send to ff02::1%lan@eth0 (Address not available)`

Ysurac, 2020-05-26:
C'est lié à IPv6 (la case "Toujours annoncer le routeur par défaut" sûrement non cochée), mais rien à propos du VPN ici.
Si le VPN a été laissé par défaut, il s'agit de glorytun-tcp.

xillibit, 2020-05-26:
J'ai laissé le VPN pas défaut

Dés que j'active les options IPv6 sous le LAN2 (celui ou se trouvent les machines du réseau local) le VPN passe en down :

[image]

Ysurac, 2020-05-27:
Etrange.
Je peux avoir la configuration ? la partie "network" dans Système->OpenMPTCProuter et onglet "Voir tous les paramètres" ou via SSH sur le routeur : ```uci show network```

xillibit, 2020-05-27:
J'ai copié la configuration dans un fichier texte en enlevant les clés et  IP publics

[config_omr.txt](https://github.com/Ysurac/openmptcprouter/files/4689922/config_omr.txt)

Ysurac, 2020-08-12:
OpenMPTCProuter v0.55 is released. This should be fixed in release.
Fell free to re-open bug if needed.

closed 2020-08-12 by Ysurac