Ysurac/openmptcprouter #1040: plusieurs adresse ip wan

ctrlx83, 2020-05-26:
Bonjour à tous,

est il possible de connecter plusieurs VPS distant afin de pouvoir disposer de plusieurs adresses ip internet ?

merci à vous
Clément

Ysurac, 2020-05-26:
Il est possible d'avoir plusieurs VPS, mais l'intérêt est uniquement le failover. Donc quand un VPS répond plus, ça passe au suivant. Pas possible de pouvoir sortir/entrée via l'IP de son choix.

ctrlx83, 2020-05-26:
par contre si j'ai besoin de 3 adresses ip wan (par exemple), je prends 3 vps avec 3 pi, et ensuite à moi de régler les gateways en interne ?

Ysurac, 2020-05-26:
Non. Ca ne fonctionnera pas. Pour la redirection des ports, le VPN est utilisé, et un seul VPN est possible. Sinon c'est de la configuration manuel et de l'ajout de VPN, indépendant d'OpenMPTCProuter.
Certains ont utilisés plusieurs IPs sur un VPS et un tunnel GRE, mais je n'ai pas eu de retour sur la configuration et je n'ai pas testé moi même.

ctrlx83, 2020-05-26:
ok merci pour les infos

ctrlx83, 2020-05-29:
Bonjour,
je viens de mettre en place 2 machines virtuelles OMR raccordée chacune sur 1 vps scaleway différent. Le fonctionnement est bon, je dispose de 2 adresses ip wan supplémentaires

kevin39, 2020-05-29:
En poussant la logique de ton schéma on pourrait faire en sorte que tes 2 VPS renvoient vers une même DMZ qui fait aussi de gateway chez toi (ex. PFSense) et pour la sortie WAN il s'agit de simple règle du FW et tu pourrais router sur 1 ou l'autre de tes OMR

ctrlx83, 2020-05-29:
Voici mon schéma d'installation : 
- 1 liaison VDSL orange
ip livebox : 192.168.253.254

- les systèmes openmptcp et les serveurs sont virtualisés sur Hyper-V
- il y a 2 cartes réseau par système openmptcp : eth0 (wan, vswitch externe) et eth1 (lan, vswitch interne)

- système openmptcp router #1 
ip wan : 51.15.xx.xx
ip lan : 192.168.254.101
gw : 192.168.253.254 (livebox)
NAT TCP 80+443 > 192.168.254.151

- système openmptcp router #2 
ip wan : 163.172.xx.xx
ip lan : 192.168.254.102
192.168.253.254 (livebox)
NAT TCP 80+443 > 192.168.254.152

- serveur #1 
utilise les ports 80 et 443 TCP
ip 192.168.254.151/24 connecté sur vswitch interne
gw 192.168.254.101

- serveur #2 
utilise les ports 80 et 443 TCP
ip 192.168.254.152/24 connecté sur vswitch interne
gw 192.168.254.102

**Les serveurs #1 et #2 ont besoin d'avoir une adresse ip wan qui leur est propre. Raison pour laquelle je doit avoir plusieurs adresses IP WAN.**

J'aurai pu faire arriver la laision en DMZ sur un PFSense ou un USG mais je suis resté en direct sur l'OMR.

kevin39, 2020-05-29:
Intéressant :)

Pour info, si cela t'intéresse, tu peux redonder les VPS si tu as besoin d'une haute disponibilité, ça marche du feu de dieu. J'ai 2 VPS configurés, OMR switch de l'un à l'autre tout seul. Ca manque de fluidité mais ça marche tout seul...

ctrlx83, 2020-05-29:
@kevin39 tu passes par un LB scaleway ?

kevin39, 2020-05-29:
Non c'est natif OMR le LB.
Si tu configures 2 VPS dans l'assistant, le LB sera actif.

ctrlx83, 2020-05-29:
ok merci de l'info
je n'ai jamais testé cette fonctionnalité

closed 2020-10-01 by github-actions[bot]