Ysurac/openmptcprouter #1042: Wed May 27 09:41:14 2020 daemon.err /usr/bin/ss-local[15191]: getpeername: Socket not connected Wed May 27 09:41:14 2020 daemon.err /usr/bin/ss-redir[15192]: remote recv: Operation timed out

kyouran38, 2020-05-27:
## Expected Behavior
Stable connection

## Current Behavior
Repeatedly short small disconnection

My connection is very unstable, I made a fresh vps / RBPI install, however I still have this disconnection, in my journal system log I have repeatedly thoses messages :

Wed May 27 09:41:14 2020 daemon.err /usr/bin/ss-local[15191]: getpeername: Socket not connected
Wed May 27 09:41:14 2020 daemon.err /usr/bin/ss-redir[15192]: remote recv: Operation timed out 

I am really desperate, I dont know where I can search to solve this problem. Any help will be appreciated please.

I have to 4G modem/router / one is 4g Orange Connection the second is Nrjmobile

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

kyouran38, 2020-05-27:
I connected to my vps : I made a systemctl status shadowsocks-libev.service

I had a ● shadowsocks-libev.service - Shadowsocks-libev Default Server Service
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:shadowsocks-libev(8)
then I did systemctl start shadowsocks-libev.service
● shadowsocks-libev.service - Shadowsocks-libev Default Server Service
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-05-27 12:01:18 CEST; 2s ago
     Docs: man:shadowsocks-libev(8)
 Main PID: 9419 (ss-server)
   Memory: 432.0K
   CGroup: /system.slice/shadowsocks-libev.service
           └─9419 /usr/bin/ss-server -c /etc/shadowsocks-libev/config.json -u
And apparently this is much more stable, why shadowsocks could not connect by itslef ?

kyouran38, 2020-05-27:
It was too good to be true,

The problem just did repeat again :(

Ysurac, 2020-05-27:
On the VPS you launched the wrong service, ```shadowsocks-libev-manager``` is used.
What is the latency of each connection ? (available in OpenMPTCProuter status page).
What is the VPS provider ?

kyouran38, 2020-05-27:
Wan 1 : Latency 52 MS
Wan 2 : Latency 36 Ms

My provider is Mylkiwan

Ysurac, 2020-05-27:
All is green in OpenMPTCProuter status page ?

kyouran38, 2020-05-27:
yes except : Your IP was not leased by this router

Ysurac, 2020-05-27:
What do you have in ```/var/log/daemon.log``` on the VPS when you have a disconnection ?

kyouran38, 2020-05-27:
This 

 

May 27 12:06:36 pierrickl ss-server[493]: getpeername: Transport endpoint is not connected
May 27 12:06:39 pierrickl ss-server[493]: getpeername: Transport endpoint is not connected
May 27 12:06:42 pierrickl ss-server[493]: getpeername: Transport endpoint is not connected
May 27 12:06:50 pierrickl ss-server[493]: server recv: Connection reset by peer
May 27 12:06:50 pierrickl ss-server[493]: server recv: Connection reset by peer
May 27 12:06:50 pierrickl ss-server[493]: server recv: Connection reset by peer
May 27 12:11:40 pierrickl ss-server[493]: remote recv: Connection reset by peer
May 27 12:11:41 pierrickl ss-server[493]: remote recv: Connection reset by peer
May 27 12:16:11 pierrickl ss-server[493]: remote recv: Connection reset by peer
May 27 12:17:19 pierrickl ss-server[493]: server recv: Connection reset by peer
 

 
 

Sent: Wednesday, May 27, 2020 at 1:12 PM

Ysurac, 2020-05-27:
Can you check with a ```ping -B -I wan1 <ipofthevps>``` (same for wan2) via SSH from the router, that you don't loose too much packets ?
Can you check also on the router, in State->System logs, what you have about omr-tracker ?
Routers use RJ45 or USB ?

kyouran38, 2020-05-28:
Hi,

 

I forget to tell you that I have also another modem on Free on my Wan1.

 

Here are the result of the ping :

 

root@OpenMPTCProuter:~# ping -B -I wan1 45.13.XXX.X
PING 45.13.XXX.X7 (45.13.XXX.X) from 192.168.10.2 wan1: 56(84) bytes of data.
64 bytes from 45.13.104.57: icmp_seq=1 ttl=51 time=52.9 ms
64 bytes from 45.13.104.57: icmp_seq=2 ttl=51 time=180 ms
64 bytes from 45.13.104.57: icmp_seq=3 ttl=51 time=46.6 ms
64 bytes from 45.13.104.57: icmp_seq=4 ttl=51 time=89.9 ms
64 bytes from 45.13.104.57: icmp_seq=5 ttl=51 time=49.9 ms
64 bytes from 45.13.104.57: icmp_seq=6 ttl=51 time=80.8 ms
64 bytes from 45.13.104.57: icmp_seq=7 ttl=51 time=108 ms
64 bytes from 45.13.104.57: icmp_seq=8 ttl=51 time=40.8 ms
64 bytes from 45.13.104.57: icmp_seq=9 ttl=51 time=78.2 ms
64 bytes from 45.13.104.57: icmp_seq=10 ttl=51 time=78.3 ms
64 bytes from 45.13.104.57: icmp_seq=11 ttl=51 time=36.7 ms
64 bytes from 45.13.104.57: icmp_seq=12 ttl=51 time=36.0 ms
64 bytes from 45.13.104.57: icmp_seq=13 ttl=51 time=124 ms
64 bytes from 45.13.104.57: icmp_seq=14 ttl=51 time=35.3 ms
64 bytes from 45.13.104.57: icmp_seq=15 ttl=51 time=74.4 ms
64 bytes from 45.13.104.57: icmp_seq=16 ttl=51 time=73.5 ms
64 bytes from 45.13.104.57: icmp_seq=17 ttl=51 time=303 ms
64 bytes from 45.13.104.57: icmp_seq=18 ttl=51 time=72.6 ms
64 bytes from 45.13.104.57: icmp_seq=19 ttl=51 time=71.9 ms

----------------------------------------------------------------

PING 45.13.XXX.X (45.13.XXX.X) from 192.168.11.2 wan2: 56(84) bytes of data.
64 bytes from 45.13.104.57: icmp_seq=1 ttl=53 time=83.9 ms
64 bytes from 45.13.104.57: icmp_seq=2 ttl=53 time=33.9 ms
64 bytes from 45.13.104.57: icmp_seq=3 ttl=53 time=31.1 ms
64 bytes from 45.13.104.57: icmp_seq=4 ttl=53 time=35.2 ms
64 bytes from 45.13.104.57: icmp_seq=5 ttl=53 time=33.7 ms
64 bytes from 45.13.104.57: icmp_seq=6 ttl=53 time=34.2 ms
64 bytes from 45.13.104.57: icmp_seq=7 ttl=53 time=33.4 ms
64 bytes from 45.13.104.57: icmp_seq=8 ttl=53 time=35.5 ms
64 bytes from 45.13.104.57: icmp_seq=9 ttl=53 time=35.7 ms
64 bytes from 45.13.104.57: icmp_seq=10 ttl=53 time=41.9 ms
64 bytes from 45.13.104.57: icmp_seq=11 ttl=53 time=37.4 ms
64 bytes from 45.13.104.57: icmp_seq=12 ttl=53 time=35.5 ms
64 bytes from 45.13.104.57: icmp_seq=13 ttl=53 time=34.9 ms
64 bytes from 45.13.104.57: icmp_seq=14 ttl=53 time=31.8 ms
64 bytes from 45.13.104.57: icmp_seq=15 ttl=53 time=33.3 ms
64 bytes from 45.13.104.57: icmp_seq=16 ttl=53 time=31.4 ms
64 bytes from 45.13.104.57: icmp_seq=17 ttl=53 time=31.8 ms
64 bytes from 45.13.104.57: icmp_seq=18 ttl=53 time=34.9 ms

-------------------------------------------------------------------

root@OpenMPTCProuter:~# ping -B -I wan3 45.13.XXX.X
PING 45.13.XXX.X (45.13.XXX.X) from 192.168.12.2 wan3: 56(84) bytes of data.
64 bytes from 45.13.104.57: icmp_seq=1 ttl=47 time=46.5 ms
64 bytes from 45.13.104.57: icmp_seq=2 ttl=47 time=44.0 ms
64 bytes from 45.13.104.57: icmp_seq=3 ttl=47 time=42.9 ms
64 bytes from 45.13.104.57: icmp_seq=4 ttl=47 time=63.9 ms
64 bytes from 45.13.104.57: icmp_seq=5 ttl=47 time=64.9 ms
64 bytes from 45.13.104.57: icmp_seq=6 ttl=47 time=59.9 ms
64 bytes from 45.13.104.57: icmp_seq=7 ttl=47 time=58.9 ms
64 bytes from 45.13.104.57: icmp_seq=8 ttl=47 time=39.1 ms
64 bytes from 45.13.104.57: icmp_seq=9 ttl=47 time=64.9 ms
64 bytes from 45.13.104.57: icmp_seq=10 ttl=47 time=50.9 ms
[... 632 lines cut ...]
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu May 28 07:35:08 2020 user.notice OMR-VPS: OpenVPN restart...
Thu May 28 07:35:08 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Thu May 28 07:35:08 2020 user.notice SQM: Using generic sqm_start_default function.
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Thu May 28 07:35:08 2020 user.notice OMR-VPS: MLVPN restart...
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu May 28 07:35:08 2020 daemon.warn [7027]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Thu May 28 07:35:08 2020 user.notice OMR-VPS: DSVPN restart...
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu May 28 07:35:08 2020 daemon.warn [7027]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Thu May 28 07:35:08 2020 user.notice SQM: simple.qos was started on tun0 successfully
Thu May 28 07:35:08 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu May 28 07:35:08 2020 user.notice firewall: Reloading firewall due to ifup of

Ysurac, 2020-05-28:
For a big part of the log there is a date/time problem. Date/time must be accurate on VPS and router.

kyouran38, 2020-05-28:
I corrected it, now the time are the same on the vps and the router

 
 

Sent: Thursday, May 28, 2020 at 10:00 AM

kyouran38, 2020-05-28:
On the server I have this information :

 

root@pierrickl:/var/log# date
Thu 28 May 2020 10:15:39 AM CEST
 

Is it correct for a French located server ?

 

 

 
 

Sent: Thursday, May 28, 2020 at 10:00 AM

Ysurac, 2020-05-28:
CEST is current french timezone, so UTC+2. When a timezone is defined it's ok even if one use UTC and the other CEST.

olivierb2, 2020-06-01:
I don't know if free is a problem, but I have the same issue (two free ADSL + 1 Free Mobile 4G + Scaleway VPS). Sometime it's working pretty fine, sometime it's really a shame.

I have hundred of `server recv: Connection reset by peer` in the logs, I'm not sure what is causing this. If I disable ShadowSocks, seems much stable but of course I loose aggregation.

Ysurac, 2020-06-01:
"server recv: Connection reset by peer" means the client connected to the router close the connection actively.
Do you use p2p ?
You can also check by testing each connections (disabling mptcp on one, the other,...)

olivierb2, 2020-06-01:
Nope, I never use any kind of P2P software. I guess my 4G put sometime my connection very unstable, it's more stable without this but my internet is of course much slower then.

kyouran38, 2020-06-01:
Same here, I don't do any p2p, how to disable mptcp on interface ? 

 

 

 

Sent: Monday, June 01, 2020 at 10:13 AM

Ysurac, 2020-06-01:
In Network->MPTCP.

Ysurac, 2020-06-01:
@olivierb2 difficult to have something really good if connection loose too many packets.
Increasing the value of "Multipath TCP SYN retries" in Network->MPTCP can help.
Setting cubic as congestion control and BLEST as Multipath TCP scheduler may also help in some case.

olivierb2, 2020-06-01:
Thanks ! I just update all those value for testing, I always wait a while until I can say if it's better or not :-)

olivierb2, 2020-06-08:
I guess this can be closed as I don't have anymore issue with 0.55beta7 release.

closed 2020-10-12 by github-actions[bot]

kyouran38, 2021-08-23:
Hi I am reopening this issue cos I have this log all the time : 

Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14425]: remote recv: Host is unreachable
Mon Aug 23 12:53:14 2021 daemon.err /usr/bin/ss-redir[14426]: remote recv: Host is unreachable

My version of  openmptcprouter v0.58.5 r0+16336-b36068d35d