Ysurac/openmptcprouter #1043: v2ray not working

bournehlm, 2020-05-28:
## Current Behavior
When enable v2ray+http from the router, the vps shadowsocks can't startup correctly.
Here is the vps server configration and logs.

cat /etc/shadowsocks-libev/config.json 
{
    "server": "::0",
    "server_port": 65101,
    "local_port": 1081,
    "mode": "tcp_and_udp",
    "key": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
    "timeout": 600,
    "method": "chacha20",
    "verbose": 0,
    "ipv6_first": true,
    "prefer_ipv6": false,
    "fast_open": true,
    "no_delay": true,
    "reuse_port": true,
    "mptcp": true,
    "ebpf": false,
    "acl": "/etc/shadowsocks-libev/local.acl",
    "plugin": "/usr/local/bin/v2ray-plugin",
    "plugin_opts": "server"
}

systemctl status shadowsocks-libev-server@config
● shadowsocks-libev-server@config.service - Shadowsocks-Libev Custom Server Service for config
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev-server@.service; indirect; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-05-28 16:04:02 CST; 4s ago
     Docs: man:ss-server(1)
  Process: 26219 ExecStart=/usr/bin/ss-server -c /etc/shadowsocks-libev/config.json --no-delay (code=exited, status=255)
 Main PID: 26219 (code=exited, status=255)

May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 INFO: plugin "/usr/local/bin/v2ray-plugin" enabled
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 INFO: UDP relay enabled
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 INFO: enable TCP no-delay
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 INFO: initializing ciphers... chacha20
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 INFO: tcp server listening at [::1]:32964
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 INFO: tcp port reuse enabled
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 ERROR: bind: Cannot assign requested address
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ ss-server[26219]:  2020-05-28 16:04:02 ERROR: failed to bind address
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ systemd[1]: shadowsocks-libev-server@config.service: Main process exited, code=exited, status=255/n/a
May 28 16:04:02 iZbp11ypsiojldq9m8tl6cZ systemd[1]: shadowsocks-libev-server@config.service: Failed with result 'exit-code'.

## Steps to Reproduce the Problem

  1. Router enable v2ray+http.
  2. Check the VPS, there is no ss and v2ray process.

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: BPI-R2

Ysurac, 2020-05-28:
Too old release. I only support latest release (for BPI-R2 support will be back in next release, available as beta).
Many changes was done since 0.52.3.

In your log I can see that it can't bind port, so it seems that port is already used, maybe by an already running shadowsocks with another startup script.

bournehlm, 2020-05-30:
Yes, it's the bind address problem, the v2ray plugin can't bind ipv6 address, must remove the ipv6 address from the ss configration. Thanks.

BTW, if i use 0.52.3 router, which highest vps version can i use? Or only can use 0.1001.

closed 2020-10-02 by github-actions[bot]