Ysurac/openmptcprouter #1046: openmptcprouter wireless only (raspberry pi)

bopnrgi, 2020-05-31:
Bonjour
_[avant propos : désolé de poster ici, ce n'est pas le bon endroit car il ne s'agit pas d'un issue je n'ai pas trouvé de forum officiel, mais s'il existe je me ferai une joie de le rejoindre]_

Je souhaite utiliser un raspberry pi 4 - 4Go avec OpenMPTCPRouter **exclusivement en wifi**.
Je détaille ce que je souhaite faire sur le schéma ci-dessous.
[image]

La qualité de ma connexion internet est très faible (petit bourg de campagne), mais un hotspot orange (opérateur français) est disponible. Je souhaiterais doubler ma vitesse de connexion avec OpenMPTCPRouter, en combinant le hotspot public et ma connexion internet "4G". 

Mais comment connecter WAN1, WAN2 puis réaliser un accès point (LAN) avec la même carte radio... je coince. une aide est la bienvenue !

Ensuite, pour WAN2 et WAN3 je pense qu'il y a une petite triche à réaliser car les adresses IP sont fournies par orange avec le même masque sous-réseau. Une idée ? Est-il possible de faire une translation d'IP ou du NAT ? à votre avis. 

PS : l’authentification au près du réseau orange est déjà faite

bopnrgi, 2020-05-31:
Hello
_[foreword: sorry to post here, this is not the right place because it is not an issue I did not find an official forum, but if it exists I will be happy to join him]_

I want to use a raspberry pi 4 - 4Go ​​with OpenMPTCPRouter ** exclusively in wifi **.
I detail what I want to do on the diagram below.
[image]

The quality of my internet connection is very low (small country town), but an orange hotspot (French operator) is available. I would like to double my connection speed with OpenMPTCPRouter, by combining the public hotspot and my "4G" internet connection.

But how to connect WAN1, WAN2 then realize a point access (LAN) with the same radio card ... I get stuck. help is welcome!

Then, for WAN2 and WAN3 I think there is a little cheat to achieve because the IP addresses are provided by orange with the same subnet mask. An idea ? Is it possible to translate IP or NAT? in your opinion.

PS: authentication near the orange network is already done
I manage to join my VPS (everything is green in the status) if I do a SCAN> Join the network. So my VPS is correctly configured and the VPN goes up correctly.

Ysurac, 2020-05-31:
Je n'ai jamais été un grand fan du wifi, donc je peux pas vraiment aider sur ce point.
Il faut savoir que le wifi interne du RPI est très limité.

Sinon pour le soucis de multiple adresses, l'utilisation de VLAN est possible par exemple.
Niveau forum, il y a https://lafibre.info/routeur/openmptcprouteur/

bopnrgi, 2020-05-31:
Merci pour la réponse rapide :)
je vais voir sur le forum, mais si quelqu'un passe ici avec un début de réponse ou une piste, je prends !

closed 2020-10-04 by github-actions[bot]