Ysurac/openmptcprouter #1049: Yet another IPV6 problem

vinceducat, 2020-06-01:
## Expected Behavior
ipv6 acces to OMR

## Current Behavior
error in the status view no ipv6 acces

## Possible Solution
firewall problem / not good configuration

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)
change my omr from raspi 3 0.54.8 to pc x86_64 0.55

and the probleme appear

## Specifications

  - OpenMPTCProuter version: 0.55beta7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform:  x86_64)

vinceducat, 2020-06-01:
root@OpenMPTCProuter:/etc/init.d# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP,80000> mtu 1500 state UP qlen 1000
    inet6 fda0:79d4:a378::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fea4:9a27/64 scope link
       valid_lft forever preferred_lft forever
6: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
    inet6 fe80::e4e6:eaff:fe75:7bef/64 scope link
       valid_lft forever preferred_lft forever
7: wan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
    inet6 fe80::d485:f6ff:fe9f:bc08/64 scope link
       valid_lft forever preferred_lft forever
27: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 state UNKNOWN qlen 1000
    inet6 fe80::a00:2/128 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:ff02/64 scope link
       valid_lft forever preferred_lft forever

Ysurac, 2020-06-01:
Do not open new issue for beta please. Use the dedicated topic.
(And snapshot VPS script is needed, if 0.1015 is used this can't work).

closed 2020-06-01 by Ysurac