Ysurac/openmptcprouter #1051: Linksys WRT3200ACM not working after installing firmware

LeprovostNoam, 2020-06-02:
## Expected Behavior
The router will normaly restart with the OpenMPTCRouter firmware

## Current Behavior
The router restart, power and internet led are ON, but no wireless and when i connect to my computer by Etherent, i've no gateway IP detected, and 192.168.100.1 is not online

## Steps to Reproduce the Problem

  1. Upload the OpenMPTCRouter firmware
  2. wait for reboot

## Context (Environment)
I'm trying to install OpenMPTCRouter firmware on my linksys router

## Specifications

  - OpenMPTCProuter version: v0.54.7
  - OpenMPTCProuter platform: WRT3200ACM

Ysurac, 2020-06-02:
Your computer use DHCP ? if yes, did you try to set it as static on 192.168.100.x network ?
You put lan cable on "internet" port of the router ?

LeprovostNoam, 2020-06-02:
The lan was not put on "internet", so it's solved

closed 2020-06-02 by LeprovostNoam