Ysurac/openmptcprouter #1053: Openmptcp dns pi-hole

yaya68, 2020-06-03:
Bonjour ,

Désoler pour le post précédent, alors je suis sur une pi 4 4go avec la version de openmptcp v0.54.7 r0+12509-04a21c26a0 et vps avec ovh debian 9 . 
Tous fonctionne à merveille,mais j’aimerais de l’aide pour configurer le dns de openmptcp pour mettre l’adresse IP de ma pi 3 ou j’ai pi-hole installer !
Merci d’avance

Ysurac, 2020-06-03:
Alors ce n'est toujours pas le template qui est le texte que tu as effacé lors de l'ouverture de l'issue.

Dans réseau->DHCP & DNS en remplaçant le 127.0.0.1#5353 par ton ip#port. Si la valeur est refusée à cet endroit (je ne sais plus si le soucis est corrigé dans la 0.54.7), supprime le 127.0.0.1#5353 et indique le serveur dans "serveur local" à la place de /lan/ toujours sous la forme ip#port

yaya68, 2020-06-03:
Ok merci d’avoir répondu et désoler pour le template ! Effectivement la valeur es refuser dans l’endroit transmission dns ou y’a le 127.0.0.1#5353
Je vais essayer de le supprimer et le mettre dans serveur local je vais tester ça 
Merci encore 
Et vraiment top openmptcp 👍